espressif/esp-idf #4877: ESP32 not restarting after Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

FrGrQuim, 2020-03-03:
## Environment

- Development Kit:     none
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
    // v4.0-rc
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2.0
- Operating System:     Linux
- Power Supply:        external 5V

## Problem Description

After a "Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)" the esp doesn't restart even if the  Panic handler behaviour (in sdkconfig) =  Print registers and reboot

### Expected Behavior
The esp32 reboot after all Guru Meditation Error

### Actual Behavior
The esp is blocked on Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1.

## Debug Logs

periodic_task  	232705839		5%
music_play_task	899168351		22%
Tmr Svc        	5346569		<1%
IDLE1          	3546688256		88%
IDLE0          	2870422359		71%
i2s_read_task  	1064949000		26%
band_tmp_data_t	308455069		7%
tiT            	43113710		1%
main           	585483		<1%
ipc1           	214275		<1%
ipc0           	92902		<1%
uart_event_task	50		<1%
sys_evt        	9512388		<1%
mqtt_log_task  	742794		<1%
httpd          	1617		<1%
wifi           	280604090		7%
hciT           	1565494113		39%
btuT           	1346213055		33%
BTC_TASK       	538554129		13%
esp_timer      	84063093		2%
btController   	4093607157		102%
dl_task        	103		<1%
update_manager 	30		<1%
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

After restart:
Core 1 register dump:
PC      : 0x40090c56  PS      : 0x00060034  A0      : 0x80099d79  A1      : 0x3f8379f0  
A2      : 0x3f837078  A3      : 0x3f40bf38  A4      : 0x00000014  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x3f40bf38  A9      : 0x3f837078  
A10     : 0x00000001  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x3f40bb5e  
A14     : 0x3ff47024  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400932a9  LEND    : 0x400932b9  LCOUNT  : 0xfffffff8

bitmakk, 2020-03-03:
You have to set it in the menuconfig

FrGrQuim, 2020-03-04:
if you talk about "Panic handler behaviour = Print registers and reboot" it is already done in menuconfig.
The esp restart automatically for all type of panic error except this one.

Dazza0, 2020-03-09:
@FrGrQuim 

I ran the following code and could not recreate your issue.

```c
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void cpu1task(void *arg)
{
    printf("cpu1task running\n");
    portDISABLE_INTERRUPTS();
    while(1) {
        ;    //Diasble interrupts and loop to trigger INT WDT
    }
}

void app_main()
{
    printf("hello world\n");
    xTaskCreatePinnedToCore(cpu1task, "cpu1", 2048, NULL, 2, NULL, 1);
}
```

Could you please provide decode the backtraces printed by the panic handler, and also provide the sdkconfig of your application

KonstantinKondrashov, 2020-03-10:
Hi @FrGrQuim!
If you are not able to see the backtrace and it locks up in the panic handler on `ELF file SHA256:` then please look at https://github.com/espressif/esp-idf/issues/1853#issuecomment-597065189. Maybe that patch can help in your case.
Thanks.

Alvin1Zhang, 2020-05-26:
@FrGrQuim Thanks for reporting, would you please help check if any updates for the issue? Thanks.

AxelLin, 2022-02-18:
@FrGrQuim 
I just test with v4.3 branch but I cannot reproduce this issue.
v4.0 is EOL, could you try with newer esp-idf releases?
BTW, if you can reproduce it please also provide your sdkconfig.

ESP-Marius, 2024-08-02:
Closing this one due to lack of activity. Feel free to reopen if there is still any issues here.

closed 2024-08-02 by ESP-Marius