espressif/esp-idf #4879: Unable to make HTTPS requests - esp-tls: mbedtls_ssl_handshake returned -0x1c

vparuthi, 2020-03-04:
I was wondering if you could help me with an issue I am having regarding making https requests. This is for an esp32 and my code seems to crash whenever I make an https request with esp_http_client but works fine for http requests. I am using the Amazon-FreeRTOS esp-idf which means I'm using release/v3.3. 

Below is my task that runs once a wifi connection is established:

```
void send_http_request(void *arg){
    ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
    esp_http_client_config_t http_config = {
            .url = "https://www.howsmyssl.com",
            .event_handler = http_event_handle,
            .method = HTTP_METHOD_GET,
            .cert_pem = "-----BEGIN CERTIFICATE-----\n ..."
    };

    esp_http_client_handle_t client = esp_http_client_init(&http_config);

    ESP_LOGI(WIFI_TASK_NAME, "Watermark: %i, free heap size: %i, %i", uxTaskGetStackHighWaterMark(NULL),
            esp_get_free_heap_size(),
             esp_get_minimum_free_heap_size());
    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        ESP_LOGI(WIFI_TASK_NAME, "Status = %d, content_length = %d",
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));
    }else{
        ESP_LOGI(WIFI_TASK_NAME, "send_http_request(): %i, %s", err, esp_err_to_name(err));
    }

    esp_http_client_cleanup(client);
    for(;;){}
}
```

The code seems to crash on the following line: 
```
    esp_err_t err = esp_http_client_perform(client);
```

with the following console output:

```
I (918) WIFI_TASK: Event = 0004
I (918) WIFI_TASK: SYSTEM_EVENT_STA_CONNECTED
I (3328) event: sta ip: 192.168.7.104, mask: 255.255.255.0, gw: 192.168.7.1
I (3328) WIFI_TASK: Event = 0007
I (3328) WIFI_TASK: SYSTEM_EVENT_STA_GOT_IP:192.168.7.104
I (3338) WIFI_TASK: Watermark: 2520, free heap size: 229772, 219344
E (3448) esp-tls: mbedtls_ssl_handshake returned -0x1c
I (3448) esp-tls: Certificate verified.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40127124  PS      : 0x00060a30  A0      : 0x80127a8d  A1      : 0x3ffb95a0  
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127124: [Errno 2] No such file or directory
A2      : 0xc6351bff  A3      : 0x00000190  A4      : 0x3ffb9510  A5      : 0x0000000c  
A6      : 0x3ffb9260  A7      : 0x00000003  A8      : 0x80123660  A9      : 0x3ffb9590  
A10     : 0xc6351bff  A11     : 0x00000000  A12     : 0x00000190  A13     : 0x3ffba36c  
A14     : 0x3ffb91d0  A15     : 0x00000019  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xc6351c07  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4000c46c: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4000c477: [Errno 2] No such file or directory

ELF file SHA256: e1b2cfd4353c7e3aaee1938c143dae2d2a224974d3f4b14c207552fa68671d98

Backtrace: 0x40127124:0x3ffb95a0 0x40127a8a:0x3ffb95c0 0x4015473c:0x3ffb95e0 0x40154f7b:0x3ffb9600 0x40154fbf:0x3ffb9620 0x401542bb:0x3ffb9640 0x40157489:0x3ffb9660 0x4014c274:0x3ffb9680 0x4014c5df:0x3ffb96a0 0x400d6f75:0x3ffb96c0
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127124: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127a8a: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4015473c: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40154f7b: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40154fbf: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x401542bb: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40157489: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4014c274: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4014c5df: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x400d6f75: [Errno 2] No such file or directory

Rebooting...
ets Jun  8 2016 00:22:57
```

I generated my certificate using the following command: 
```
openssl s_client -showcerts -connect https://httpbin.org/deflate </dev/null 2>/dev/null|openssl x509 -outform PEM >mycertfile.pem
```
and pasted into the http_config initialization.

**Things to Note**
1. The following line in the console log:
```
E (3448) esp-tls: mbedtls_ssl_handshake returned -0x1c
I (3448) esp-tls: Certificate verified.
```
and I think -0x1c represents the following errors

```
#define MBEDTLS_ERR_MPI_DIVISION_BY_ZERO                  -0x000C  /**< The input argument for division is zero, which is not allowed. */
#define MBEDTLS_ERR_MPI_ALLOC_FAILED                      -0x0010  /**< Memory allocation failed. */
```
2. However I do print out the amount of space left on the line above and there does seem to be enough

3. I'm not sure why the following line keeps printing since the file does exist
```
/Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x400d6f75: [Errno 2] No such file or directory
```

If anyone could help that would be great! Thanks! 

**sdkconfig**

```
#
# mbedTLS
#
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC=
CONFIG_MBEDTLS_CUSTOM_MEM_ALLOC=
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=8192
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096
CONFIG_MBEDTLS_DEBUG=
CONFIG_MBEDTLS_ECP_RESTARTABLE=y
CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=
CONFIG_MBEDTLS_HARDWARE_SHA=
CONFIG_MBEDTLS_HAVE_TIME=y
CONFIG_MBEDTLS_HAVE_TIME_DATE=
CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT=
CONFIG_MBEDTLS_TLS_SERVER_ONLY=
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
CONFIG_MBEDTLS_TLS_DISABLED=
CONFIG_MBEDTLS_TLS_CLIENT=y
CONFIG_MBEDTLS_TLS_ENABLED=y

#
# TLS Key Exchange Methods
#
CONFIG_MBEDTLS_PSK_MODES=
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_DHE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_RSA=y
CONFIG_MBEDTLS_SSL_RENEGOTIATION=y
CONFIG_MBEDTLS_SSL_PROTO_SSL3=
CONFIG_MBEDTLS_SSL_PROTO_TLS1=y
CONFIG_MBEDTLS_SSL_PROTO_TLS1_1=y
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
CONFIG_MBEDTLS_SSL_PROTO_DTLS=
CONFIG_MBEDTLS_SSL_ALPN=y
CONFIG_MBEDTLS_SSL_SESSION_TICKETS=y

#
# Symmetric Ciphers
#
CONFIG_MBEDTLS_AES_C=y
CONFIG_MBEDTLS_CAMELLIA_C=
CONFIG_MBEDTLS_DES_C=
CONFIG_MBEDTLS_RC4_DISABLED=y
CONFIG_MBEDTLS_RC4_ENABLED_NO_DEFAULT=
CONFIG_MBEDTLS_RC4_ENABLED=
CONFIG_MBEDTLS_BLOWFISH_C=
CONFIG_MBEDTLS_XTEA_C=
CONFIG_MBEDTLS_CCM_C=y
CONFIG_MBEDTLS_GCM_C=y
CONFIG_MBEDTLS_RIPEMD160_C=

#
# Certificates
#
CONFIG_MBEDTLS_PEM_PARSE_C=y
CONFIG_MBEDTLS_PEM_WRITE_C=y
CONFIG_MBEDTLS_X509_CRL_PARSE_C=y
CONFIG_MBEDTLS_X509_CSR_PARSE_C=y
CONFIG_MBEDTLS_ECP_C=y
CONFIG_MBEDTLS_ECDH_C=y
CONFIG_MBEDTLS_ECDSA_C=y
CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
CONFIG_MBEDTLS_ECP_NIST_OPTIM=y
```

Alvin1Zhang, 2020-03-04:
@vparuthi Thanks for reporting. We will look into.

shubhamkulkarni97, 2020-03-16:
@vparuthi Please try building your application on [this branch](https://github.com/shubhamkulkarni97/amazon-freertos/tree/feature/add_esp_http_client).

shubhamkulkarni97, 2020-03-23:
@vparuthi Is your issue resolved?

AdityaHPatwardhan, 2020-03-23:
Actually , This issue should be closed, as It was related to AFR build process and not `esp-idf` directly.

Alvin1Zhang, 2020-03-23:
@vparuthi Thanks for reporting and thank @shubhamkulkarni97 @AdityaHPatwardhan for the support. Will close now.

closed 2020-03-23 by Alvin1Zhang