espressif/esp-idf #4882: lock_acquire_generic- assert failed!

umer-ilyas, 2020-03-04:
## Environment

- Development Kit: custom
- Kit version
- Module or chip used: ESP32-WROOM-32D
- IDF version v4.0-174-gf9cb434ee
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
(Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply: Battery

## Problem Description
micro controller rebooted with following log 
mqtt was connected at the time 

```
F:/ESP-IDF-v4.0/esp-idf/components/newlib/locks.c:136 (lock_acquire_generic)- assert failed!
abort() was called at PC 0x40083cd7 on core 1
0x40083cd7: lock_acquire_generic at F:/ESP-IDF-v4.0/esp-idf/components/newlib/locks.c:136 (discriminator 1)

ELF file SHA256: c98f60cbdceedad5ed0735976833f71bc8de9b172f716ca62ce4036621391655

Backtrace: 0x40091c45:0x3ffd7280 0x40091fc9:0x3ffd72a0 0x40083cd7:0x3ffd72c0 0x40083e09:0x3ffd72f0 0x400d8a87:0x3ffd7310 0x400d74c9:0x3ffd7330 0x4000bd83:0x3ffd7350 0x4000117d:0x3ffd7370 0x400592fe:0x3ffd7390 0x4005937a:0x3ffd73b0 0x40058bbf:0x3ffd73d0 0x4018ad4f:0x3ffd7400 0x40190303:0x3ffd7430 0x40190471:0x3ffd7740 0x400833bd:0x3ffd7770 0x400f09c3:0x3ffd77c0 0x40095705:0x3ffd7800
0x40091c45: invoke_abort at F:/ESP-IDF-v4.0/esp-idf/components/esp32/panic.c:155
0x40091fc9: abort at F:/ESP-IDF-v4.0/esp-idf/components/esp32/panic.c:172
0x40083cd7: lock_acquire_generic at F:/ESP-IDF-v4.0/esp-idf/components/newlib/locks.c:136 (discriminator 1)
0x40083e09: _lock_acquire_recursive at F:/ESP-IDF-v4.0/esp-idf/components/newlib/locks.c:171
0x400d8a87: uart_write at F:/ESP-IDF-v4.0/esp-idf/components/vfs/vfs_uart.c:198 (discriminator 1)
0x400d74c9: get_local_fd at F:/ESP-IDF-v4.0/esp-idf/components/vfs/vfs.c:277
 (inlined by) esp_vfs_lseek at F:/ESP-IDF-v4.0/esp-idf/components/vfs/vfs.c:427
0x4018ad4f: __sprint_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
0x40190303: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)
0x40190471: vprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x400833bd: esp_log_write at F:/ESP-IDF-v4.0/esp-idf/components/log/log.c:217
0x400f09c3: esp_mqtt_task at F:/ESP-IDF-v4.0/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1280 (discriminator 9)
0x40095705: vPortTaskWrapper at F:/ESP-IDF-v4.0/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7248
ho 0 tail 12 room 4
load:0x40078000,len:14944
load:0x40080400,len:4940
entry 0x40080704
[0;32mI (69) boot: Chip Revision: 1[0m
[0;32mI (34) boot: ESP-IDF v4.0-174-gf9cb434ee-dirty 2nd stage bootloader[0m
[0;32mI (34) boot: compile time 12:13:14[0m
[0;32mI (34) boot: Enabling RNG early entropy source...[0m
[0;32mI (41) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (46) boot: SPI Speed      : 80MHz[0m
[0;32mI (50) boot: SPI Mode       : QIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (91) boot:  3 ota_0            OTA app          00 10 00010000 00180000[0m
[0;32mI (99) boot:  4 ota_1            OTA app          00 11 00190000 00180000[0m
[0;32mI (106) boot: End of partition table[0m
[0;32mI (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x34a68 (215656) map[0m
[0;32mI (178) esp_image: segment 1: paddr=0x00044a90 vaddr=0x3ffb0000 size=0x04e34 ( 20020) load[0m
[0;32mI (185) esp_image: segment 2: paddr=0x000498cc vaddr=0x40080000 size=0x00400 (  1024) load[0m
0x40080000: _WindowOverflow4 at F:/ESP-IDF-v4.0/esp-idf/components/freertos/xtensa_vectors.S:1778
[0;32mI (185) esp_image: segment 3: paddr=0x00049cd4 vaddr=0x40080400 size=0x0633c ( 25404) load[0m
[0;32mI (202) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xcd140 (840000) map[0m
0x400d0018: _stext at ??:?
[0;32mI (431) esp_image: segment 5: paddr=0x0011d160 vaddr=0x4008673c size=0x12114 ( 74004) load[0m
0x4008673c: bb_wdt_int_enable at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2740
[0;32mI (455) esp_image: segment 6: paddr=0x0012f27c vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (456) esp_image: segment 7: paddr=0x0012f2e8 vaddr=0x50000000 size=0x00004 (     4) load[0m
[0;32mI (478) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (478) boot: Disabling RNG early entropy source...[0m
[0;32mI (478) cpu_start: Pro cpu up.[0m
[0;32mI (482) cpu_start: Application information:[0m
[0;32mI (487) cpu_start: Project name:     solar[0m
[0;32mI (491) cpu_start: App version:      v0.0.22-72-gddcc0b1-dirty[0m
[0;32mI (498) cpu_start: Compile time:     Mar  3 2020 14:33:33[0m
[0;32mI (504) cpu_start: ELF file SHA256:  c98f60cbdceedad5...[0m
[0;32mI (510) cpu_start: ESP-IDF:          v4.0-174-gf9cb434ee-dirty[0m
[0;32mI (517) cpu_start: Starting app cpu, entry point is 0x40081508[0m
```

mqtt Commit: 8a1e1a5a9ffce89313aaa209375e1def893a4c73 [8a1e1a5]

igrr, 2020-03-04:
@umer-ilyas Could you please share the code used to reproduce this issue?

Alvin1Zhang, 2020-05-26:
@umer-ilyas Thanks for reporting, would you please help share if any updates or code for reproducing the issue? Thanks.

kennycoder, 2020-06-04:
Same story...
Just trying to build aws freertos.
Using esp-idf (3.x) and running
```cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build```
and then 
```cmake --build build```

Build fails on newlib components due to xPortCanYield being undefined.

closed 2021-02-26 by Alvin1Zhang