espressif/esp-idf #4884: MAX_REASONABLE_REQUESTS in btc_profile_queue.c  in ESP-IDF v4.0 causes build to fail

mattkefford, 2020-03-04:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v4.1
- Module or chip used: ESP32-WROVER
- IDF version: v4.0 (v4.0-dirty after the edit)
- Build System: Not sure. CMake I think
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: VisualGDB within Visual Studio
- Using an IDE?: Visual Studio
- Power Supply: USB

## Problem Description
Previously I have been working using the v3.3.1 Legacy GNU Make Build System. We already have a fairly large code project that builds fine on the legacy build environment.
Following a joint decision to upgrade due to WLAN functionality, I just had to migrate to v4.0 of the ESP-IDF.

I installed the new IDF by following the Getting Started guide. This instructs you to download the tools installer and as part of that, installs a version of IDF.
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/windows-setup.html

When prompted, I chose the first option, v4.0 (release version) which was going to download to a new folder on the desktop called esp-idf-5. I changed this to a folder on my C drive called ESP-IDF-5. I am not sure why it is 5 but I just went with it.

After all was installed, the init.bat ran to set up the paths.

I had issues with "no rule to make xxx" which was solved by doing Build>Clean in Visual Studio and re-running the export.bat file. I also made sure that the SDKConfig file was pointing to the correct one.

Now the build was going fine except for one error causing the build to fail:
C:/ESP-IDF-5/components/bt/host/bluedroid/btc/core/btc_profile_queue.c:34:21: error: 'MAX_REASONABLE_REQUESTS' defined but not used [-Werror=unused-const-variable=]
 static const size_t MAX_REASONABLE_REQUESTS = 10;

There are also 48 warnings which are all "variable X set but not used" or "unused variable X".

I changed MAX_REASONABLE_REQUESTS from a static const size_t to a #define and the error went away, but now when I compile I have v4.0-dirty listed as the IDF version used.

I wonder if this is just a case of tweaking the settings for the build rules? Because when I looked at some of these warnings, the variables are actually being used so I have no idea why the build thinks they aren't being used. Most of them are asserts.

### Expected Behavior
Get release version of ESP-IDF, build, build should complete with no warnings or errors.

### Actual Behavior
Got release version of ESP-IDF, built and got 48 warnings and 1 error which caused build to fail.

### Steps to reproduce
1. Install ESP-IDF v4.0.
2. Set up paths.
3. Attempt build of an existing code project.
4. Build fails.

// If possible, attach a picture of your setup/wiring here.
[image]

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
CC Release/bt/host/bluedroid/btc/core/btc_ble_storage.o
CC Release/lwip/lwip/src/netif/lowpan6_ble.o
CC Release/wpa_supplicant/src/wps/wps_enrollee.o
CC Release/wpa_supplicant/src/wps/wps_attr_build.o
CC Release/wpa_supplicant/src/wps/wps_registrar.o
CC Release/wpa_supplicant/src/wps/wps_common.o
CC Release/bt/host/bluedroid/btc/core/btc_storage.o
CC Release/bt/host/bluedroid/btc/core/btc_dev.o
CC Release/lwip/lwip/src/netif/lowpan6.o
C:/ESP-IDF-5/components/bt/host/bluedroid/btc/core/btc_profile_queue.c:34:21: error: 'MAX_REASONABLE_REQUESTS' defined but not used [-Werror=unused-const-variable=]
 static const size_t MAX_REASONABLE_REQUESTS = 10;
                     ^~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/ESP-IDF-5/make/component_wrapper.mk:292: host/bluedroid/btc/core/btc_profile_queue.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC Release/lwip/lwip/src/netif/zepif.o
CC Release/lwip/lwip/src/netif/bridgeif_fdb.o
CC Release/wpa_supplicant/src/wps/wps.o
CC Release/wpa_supplicant/src/wps/wps_dev_attr.o
CC Release/wpa_supplicant/src/wps/wps_attr_parse.o
CC Release/wpa_supplicant/src/wps/wps_validate.o
CC Release/wpa_supplicant/src/wps/wps_attr_process.o
CC Release/lwip/lwip/src/netif/ethernet.o
CC Release/lwip/lwip/src/netif/slipif.o
CC Release/lwip/port/esp32/vfs_lwip.o
CC Release/lwip/port/esp32/freertos/sys_arch.o
CC Release/lwip/port/esp32/netif/ethernetif.o
CC Release/lwip/port/esp32/netif/dhcp_state.o
CC Release/lwip/port/esp32/netif/nettestif.o
CC Release/lwip/port/esp32/netif/wlanif.o
CC Release/lwip/port/esp32/debug/lwip_debug.o
AR Release/wpa_supplicant/libwpa_supplicant.a
make: *** [/c/ESP-IDF-5/make/project.mk:590: component-bt-build] Error 2
make: *** Waiting for unfinished jobs....
AR Release/lwip/liblwip.a
-------------------------------------------------------------
Command exited with code 2
Executable: C:\SysGCC\esp32\usr\bin\bash.exe
Arguments: --login -c "export IDF_PATH=/c/ESP-IDF-5 && export BATCH_BUILD=1 && export PATH=/mingw32/bin:/usr/bin:$PATH && export -n PYTHONHOME && export LANG=en_US && cd /C/bartec_code/ESP32_v2 && make -j12 BUILD_DIR_BASE='/C/bartec_code/ESP32_v2/Release' SDKCONFIG='/C/bartec_code/ESP32_v2/sdkconfig-release' V=0"
Directory: 
Command-line action failed

========== Project Rebuild Summary ==========
    BWC  rebuilt in 05:03
========== Rebuild: 0 Succeeded, 1 Failed ==========

## Other items if possible

- [ ] sdkconfig file (I had to add the .txt extension, in my project it has no extension and just shows a type "File".
[sdkconfig-release.txt](https://github.com/espressif/esp-idf/files/4286964/sdkconfig-release.txt)

mattkefford, 2020-03-04:
In addition to this, if I build under release config then I don't see that error or warnings but I do get the warning:
/c/ESP-IDF-5/make/project.mk:626: warning: undefined variable 'IDF_SKIP_CHECK_SUBMODULES'

GYC-Espressif, 2020-03-18:
The issue is that when CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE option is enabled in sdkconfig, the variable MAX_REASONABLE_REQUESTS becomes unused. This leads to a compilation error.
I suggest you to enable the CONFIG_COMPILER_DISABLE_GCC8_WARNINGS option to silence the warnings, or enable CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT option to keep the asserts (without affecting the binary size). I would probably suggest doing the latter in any case, as assertions can catch errors which would otherwise be hard to debug.

Alvin1Zhang, 2020-05-25:
@mattkefford Thanks for reporting, would you please help share if any updates for the issue? Thanks.

mattkefford, 2020-07-03:
Thanks for looking into that! Unfortunately due to the coronavirus I lost my job where I was working with the ESP32, but I am sure this will come up for other people having the same issue. Where I worked we had a policy of 0 errors 0 warnings for all releases so it was impossible to go any further whilst those warnings were present.

closed 2020-07-03 by mattkefford