espressif/esp-idf #4885: ESP IDF MQTT SSL

SERIDJ, 2020-03-04:
hi,

i want to do the same thing, so  without certificat its work  well :D  , and when i try to use ssl connection, add 3 certificat in my raspberry broker (ca.cert server.crt and server.key)  and enable  require_certificate, that   not work :( 

in esp32 code   if i add or not certificat in  **esp_mqtt_client_config_t** i have the same result in terminal  : 

I (692579) MQTT_CLIENT: Error MQTT Connected
I (692579) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (702589) MQTTS_EXAMPLE: Other event id:7
I (702589) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (702599) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=104
E (702599) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1

that mean its connected with broker but cant send and receive data 
so eclipes or esp32 code  doesn't use those certificate.

i dont know what i put in  **.cert_len**  ( If the certificate is not null-terminated then cert_len should also be set )

if i try .cert_len = 2048 i have in terminal : 

Explicit cert-/key-len is not available in IDF version v4.0

what is len available in IDF V4.0 ????????

[image]

ESP32
ECLIPSE
ESP IDF V4.0

david-cermak, 2020-03-05:
Hi @SERIDJ 

esp-mqtt on v4.0 could only process PEM formatted certificates (which are text based and zero terminated), so please use the PEM format and set the `.cert_len = 0` or leave it undefined.
It seems that the ESP could not connect to the broker on mqtt level. To find out more please enable debug log level in menuconfig and check the mosquitto log on your raspberry.

SERIDJ, 2020-03-05:
hi @david-cermak 

thaks for ur replay,

i dont thins esp coulad not connect to the broker bcs when i stop sever in raspberry  with **sudo service mosquitto stop**  eclipse change error in terminal and show : 

[image]

that mean esp32 can detect a correct server with host and port and cant send data ... i will chek with menuconfig

SERIDJ, 2020-03-05:
for cerficat i use openssl
ca:
openssl genrsa -out ca.key 2048
openssl req -new -x509 -days 1826 -key ca.key -out ca.crt

server :  
openssl genrsa -out server.key 2048
openssl req -new -out server.csr -key server.key
openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 360

client : 
client:
openssl genrsa -out client.key 2048
openssl req -new -out client.csr -key client.key
openssl x509 -req -in client.csr -CA ca.crt -CAkey ca.key  -CAcreateserial -out client.crt -days 360

[image]

david-cermak, 2020-03-06:
No need to null-terminate the certificate, it's added to the binary as a string so this already contains the terminator (after the `END CERTIFICATE ---` marker).
This looks okay from the log, otherwise you'd see an error in parsing the certificate.

What the logs from the broker and debug logs from the ESP say?

SERIDJ, 2020-03-06:
@david-cermak  yes yes that generat ca.crt.S, client.crt.S and client.key.S

 
i add  in config file raspberry those command : 

log_type all
log_type error
log_type warning
log_type notice
log_type information
connection_messages true
log_timestamp true

but i dont have anny logs when esp32 is connect or not

when i stop broker pi that change logs in eclipse that mean esp is connected to broker and cant send data 

[image]

SERIDJ, 2020-03-06:
i dont have anny logs from brokrer pi, i use : **sudo mosquitto -p 1883 -v 
 **
[image]

SERIDJ, 2020-03-06:
for ca cert i modified Component.mk and CMakeList.txt is that all and correct 

[image]

[image]

SERIDJ, 2020-03-06:
the probleme is here 
[image]

ESP-Marius, 2020-03-09:
@SERIDJ Seems to me like you haven't set up MQTT to use SSL. Try adding `.transport = MQTT_TRANSPORT_OVER_SSL,` to your `mqtt_cfg` init struct.

SERIDJ, 2020-03-09:
hi @ESP-Marius 

yes yes that true but i  have another problem i try to fix it ...

[image]

SERIDJ, 2020-03-09:
@ESP-Marius  yes yes that  true, but i have another problem that work without ca.crt so when i add ca.crt i have error : 

[image]

iam sure its the same ca.cert in eclipse and raspberry broker

SERIDJ, 2020-03-09:
my second question how i can hide those certificat in esp32, i can put them in flash encryption or smth more secure ?

ESP-Marius, 2020-03-12:
1. MbedTLS (the TLS library we use) expects the hostname to match the common name, which it doesnt seem to do in your case (hostname would be the ip you are connecting to,  CN=MASTERCA)

2. MbedTLS do not support TLSv1.3
3. Mosquitto supports it, but there is no functionality in ESP-IDF for it.
4. Yes, you could use flash encryption to hide them.

SERIDJ, 2020-03-13:
@ESP-Marius thanks for ur replay, 
for the 1st question i dont understand well : 

i know MbedTLS   expects the hostname to match the common name 👍  👍 👍  that what i did 

[image]

ESP-Marius, 2020-03-16:
In the `mqtt_cfg` you specify the host with an ip address, and this will be used as the hostname for the TLS connection. So mbedtls compares the ip with the common name and find that they do not match.

SERIDJ, 2020-03-27:
hi  @david-cermak @ESP-Marius I hope you are well. 

so i want to encrypt my certificat and keys in esp32 what is the best thing encrypt all flash or encrypt one partition table or only key.

i try to enable encryption flash using idf.py menuconfig but i have error **: esptool write_flash : error : argument <address><filename> : detected at adress : 0x8000 for file : partition_table/partition-table.bin**  that mean when i enabled encryption flash  bootloader app binary bootloader.bin exceed the default bootloader size limit (28672 bytes) and when i try to Set partition table offset to a higher value than 0x8000 i need to update  **partition table** too , i dont know how i can do that from menuconfig of manualy in file csv in file ESP-IDF .... 

[image]

Alvin1Zhang, 2020-04-01:
@SERIDJ Thanks for reporting, for how questions and third party questions/issues, please help raise a post on our forum at https://esp32.com. Thanks.

closed 2020-04-01 by Alvin1Zhang

SERIDJ, 2020-04-01:
@Alvin1Zhang  oky thanks

Majd-tameem, 2022-02-15:
Hi every body,
I had faced this problem and I solved it by add (s) to the broker URI 
char *uri_mqtt_Raspberry_server = "mqtts://10.10.3.216:8883";
exactly mqtts not mqtt
then maybe it will ask to check CN in certificates.
and every think works well.