espressif/esp-idf #4886: Wi-Fi power management stops when disconnected

balix-microej, 2020-03-04:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ x ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER]
- IDF version: v3.3.1-209-gcc34d5a5b
- Build System:         [Make]
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32].
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description

The ESP32 module is powered by a battery. Power management is enabled to prevent power issues when connecting to Wi-Fi access points.

The power management seems to be deactivated in case the access point on which the ESP32 is connected is shut down. This will quickly drain the battery power unless the Wi-Fi is disabled. In the worst case, the battery level or the Wi-Fi disconnection are not detected in time and the brownout detector will be triggered.

### Expected Behavior

The Wi-Fi driver stays in power management mode MIN_MODEM.

### Actual Behavior

Power management is deactivated when the Wi-Fi is down.

### Steps to reproduce

1. Run the power_save example and wait until the ESP32 is connected to the access point
2. Shut down the access point
3. Power management stops and ESP32 current consumption raise

### Code to reproduce this issue

Reproducible with the power_save example of the esp-idf: https://github.com/espressif/esp-idf/tree/release/v3.3/examples/wifi/power_save

Only the Wi-Fi SSID and password have been changed in the configuration.

## Debug Logs

```
I (3259) wifi: AP's beacon interval = 102400 us, DTIM period = 1
I (6749) event: sta ip: 192.168.0.89, mask: 255.255.255.0, gw: 192.168.0.254
I (6749) power_save: SYSTEM_EVENT_STA_GOT_IP
I (6749) power_save: got ip:192.168.0.89

I (28329) wifi: bcn_timout,ap_probe_send_start
I (30829) wifi: ap_probe_send over, resett wifi status to disassoc
I (30829) wifi: state: run -> init (c800)
I (30829) wifi: pm stop, total sleep time: 17168969 us / 27447850 us

I (30839) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (30839) power_save: SYSTEM_EVENT_STA_DISCONNECTED
```

liuzfesp, 2020-03-05:
Hi @balix-microej this is our internal known issue, we will optimize it in future, but the exact fix plan hasn't been determined yet.

balix-microej, 2020-03-05:
OK thanks for the answer.

Meanwhile is there anything you can recommend as a workaround or some advice to prevent this issue ?

AxelLin, 2020-08-03:
> Hi @balix-microej this is our internal known issue, we will optimize it in future, but the exact fix plan hasn't been determined yet.

Hi @liuzfesp 
Is there any update regarding this issue?

sagb2015, 2020-08-05:
@balix-microej As a workaround, you can use `esp_wifi_stop` upon disconnection/max number of connection failures.  Retry after some time can happen after `esp_wifi_start`

AxelLin, 2020-09-13:
> @balix-microej As a workaround, you can use `esp_wifi_stop` upon disconnection/max number of connection failures. Retry after some time can happen after `esp_wifi_start`

But when will the real fix available? (This issue has been reported for 6 months)

liuzfesp, 2020-09-14:
Hi @AxelLin, I think we require at least another 2~3 months for final fix of this issue according to our current plan.

chungleepeople, 2021-08-11:
Hi @liuzfesp  I faced same problem. Did you deliver version? Thank a lot.

TianaESP, 2021-08-11:
Hi @chungleepeople, can you please try with the latest v4.3 or Master, and enable **ESP_WIFI_STA_DISCONNECTED_PM_ENABLE**? thanks

negativekelvin, 2021-08-12:
https://github.com/espressif/esp-idf/blob/606557b48bd195e400c2e3928d8d7c7d4ed9b050/components/esp_wifi/include/esp_wifi.h#L1241-L1243

@TianaESP Is the interval in milliseconds or is the recommended interval not possible?

AxelLin, 2021-08-12:
@TianaESP 
What does this comment mean? https://github.com/espressif/esp-idf/blob/606557b48bd195e400c2e3928d8d7c7d4ed9b050/components/esp_wifi/include/esp_wifi.h#L1240

Does it mean calling  esp_now_set_wake_window() ? (I don't quite understand the relationship of esp_now with ESP_WIFI_STA_DISCONNECTED_PM_ENABLE)

Espressif-liuuuu, 2021-08-12:
Hi @negativekelvin , thats another feature for connectionless modules (e.g espnow) to wake up periodically at disconnected, and its in milliseconds. If you just want to enter power-save mode at disconnected, just leave it alone.

Hi @AxelLin , they are two separated features. Basically, we could enter power-save mode at disconnected if we enable ESP_WIFI_STA_DISCONNECTED_PM_ENABLE. And, if we want to do periodically wake at disconnected for espnow, we could use `esp_now_set_window` and `esp_wifi_set_connectionless_wake_interval` in combination to make the behaviour of power-save.

AxelLin, 2022-03-01:
> Hi @chungleepeople, can you please try with the latest v4.3 or Master, and enable **ESP_WIFI_STA_DISCONNECTED_PM_ENABLE**? thanks

@TianaESP 
ESP_WIFI_STA_DISCONNECTED_PM_ENABLE is only available in v4.3+.
Just notice this issue is still open, so I'm wondering if you will backport the fix to older releases?

AxelLin, 2022-03-06:
> Hi @chungleepeople, can you please try with the latest v4.3 or Master, and enable **ESP_WIFI_STA_DISCONNECTED_PM_ENABLE**? thanks

@TianaESP @liuzfesp @Espressif-liuuuu 

I'm wondering why introducing ESP_WIFI_STA_DISCONNECTED_PM_ENABLE config?
What is the benefit of setting CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=n?
What is the downside of setting CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y?
Just curious why not just make it enabled by default so people don't need to set the setting.

Espressif-liuuuu, 2022-03-08:
> > Hi @chungleepeople, can you please try with the latest v4.3 or Master, and enable **ESP_WIFI_STA_DISCONNECTED_PM_ENABLE**? thanks
> 
> @TianaESP @liuzfesp @Espressif-liuuuu
> 
> I'm wondering why introducing ESP_WIFI_STA_DISCONNECTED_PM_ENABLE config? What is the benefit of setting CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=n? What is the downside of setting CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y? Just curious why not just make it enabled by default so people don't need to set the setting.
> 
> PS. I asked this because usually you don't know if some customers will use the esp32 devices by battery powered in some use cases. In the end there is no choice but to always set CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y to avoid this issue. So I'd like to know if any bad side-effect with CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y. If there is no bad side-effect, why not just make it always enabled (and remove this config option)?

Hi @AxelLin , we would discuss recently to see if we would backport it to older version.  

And also thanks for your suggestion. Yes, you`re right. At that moment, we just tried to avoid uncertain side-affect to modem-sleep, so we disabled it in default to make it still stable for most users. We would remove it in the future.

AxelLin, 2022-04-26:
> Hi @negativekelvin , thats another feature for connectionless modules (e.g espnow) to wake up periodically at disconnected, and its in milliseconds. If you just want to enter power-save mode at disconnected, just leave it alone.

The **micriosecond** is obviously a typo that needs fix:
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L1233

AxelLin, 2022-08-10:
> And also thanks for your suggestion. Yes, you`re right. At that moment, we just tried to avoid uncertain side-affect to modem-sleep, so we disabled it by default to make it still stable for most users. We would remove that configuration and enable it by default in the future.

@Espressif-liuuuu 

I still find the CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE in v5.1-dev.
So I'm wondering if CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE will be removed or not.
If not, it should at least change to **default y** to get more testing.

Espressif-liuuuu, 2022-08-22:
> @Espressif-liuuuu
>
> I still find the CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE in v5.1-dev. So I'm wondering if CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE will be removed or not. If not, it should [...]

Yes, we are keep testing internally. Considering to the different current behaviour, especially those with ESP-NOW, we would like to change it with a update on document together later:)

hannodv, 2022-08-26:
The following is still an issue:
In the code, the following is documented:
```
@attention 4. A sensible interval which is not too small is recommended (e.g. 100ms)
  *
  * @param      interval  how much micriosecond would the chip wake up, from 1 to 65535.
```
However, as @negativekelvin pointed out already, the recommended value of 100ms would equal 100 000 microseconds, which is impossible to use, since the param can only be set between 1 and 65535.

AxelLin, 2022-08-26:
> The following is still an issue: In the code, the following is documented:
>
> ```
> @attention 4. A sensible interval which is not too small is recommended (e.g. 100ms)
>   *
>   * @param      interval  how [...]

I think it was a typo, fixed in master: (The unit of interval param should be millisecond)
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L1236

WangShuoran, 2022-12-19:
@AxelLin I have any little question of the `esp_wifi_set_connectionless_wake_interval()`
1. I want to confirm the unit of interval param is millisecond, does this right?
2. As the document of master, I don't find `esp_wifi_set_connectionless_wake_interval()`, does this remove in IDFv5?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/network/esp_wifi.html

Espressif-liuuuu, 2022-12-27:
hi @WangShuoran ,
the unit is millisecond for sure, and the API is renamed to `esp_wifi_connectionless_module_set_wake_interval`, you could refer it in document https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/network/esp_wifi.html#_CPPv448esp_wifi_connectionless_module_set_wake_interval8uint16_t

WangShuoran, 2022-12-28:
@Espressif-liuuuu Thank you for quick relay.

And I have another question, If I set `esp_wifi_connectionless_module_set_wake_interval()` and `esp_now_set_wake_window()`, does the window at above API be made one, or the window will controlled by the superposition of both?

Espressif-liuuuu, 2022-12-28:
> @Espressif-liuuuu Thank you for quick relay.
> 
> And I have another question, If I set `esp_wifi_connectionless_module_set_wake_interval()` and `esp_now_set_wake_window()`, does the window at above API be made one, or the window will controlled by the superposition of both?

hi, its made one and the meaning could be refered here
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html#connectionless-modules-rx

jack0c, 2024-02-07:
Since EOL of v4.3(v4.4+ versions all include this change) and default value of CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE set to 1 in v5.1, we can close this issue.

closed 2024-02-07 by jack0c