espressif/esp-idf #4889: Self signed SSL certificate for https server with OPENSSL when using static IP

Zeni241, 2020-03-05:
Following example _esp-idf\examples\protocols\https_server_  I am running HTTPS server in AP mode **with static IP 192.168.1.222**. 
The only lines I changed from example in creating HTTPS server are below to reflect my certificate files robo.pem and roboprvkey.pem :
   
```
    extern const unsigned char robo_pem_start[] asm("_binary_robo_pem_start");
    extern const unsigned char robo_pem_end[] asm("_binary_robo_pem_end");
    conf.cacert_pem = robo_pem_start;
    conf.cacert_len = robo_pem_end - robo_pem_start;
```

    extern const unsigned char roboprvkey_pem_start[] asm("_binary_roboprvkey_pem_start");
    extern const unsigned char roboprvkey_pem_end[] asm("_binary_roboprvkey_pem_end");
    conf.prvtkey_pem = roboprvkey_pem_start;
    conf.prvtkey_len = roboprvkey_pem_end - roboprvkey_pem_start;

As I send POST request, in browser console I get:
 **POST https://192.168.1.222/echo net::ERR_CERT_AUTHORITY_INVALID**

In ESP IDF Command prompt I get error:

> AP Station connected
> I (1227163) tcpip_adapter: softAP assign IP to station,IP is: 192.168.1.123
> E (1227473) esp-tls: mbedtls_ssl_handshake returned -30592
> E (1227473) esp_https_server: esp_tls_create_server_session failed
> W (1227483) httpd: httpd_accept_conn: session creation failed
> W (1227483) httpd: httpd_server: error accepting new connection
> 

I created selfsigned certificate with following command using OPENSSL:
`req -x509 -newkey rsa:4096 -sha256 -days 3650 -nodes -keyout robo.key -out robo.crt -subj /CN=robo.com -addext subjectAltName=DNS:robo.com,DNS:robo.net,IP:192.168.1.222`
 (following https://stackoverflow.com/questions/10175812/how-to-create-a-self-signed-certificate-with-openssl).

Question is how to create self signed SSL certificates with OPENSSL when using static IP instead of mDNS. I am using static IP as with mDNS it is difficult to connect from mobile app.

AdityaHPatwardhan, 2020-04-14:
@Zeni241 , I tried your command to create certificates, they seem to be fine !  
Maybe the problem might be other than certificate (could be with the tools from which you send POST request), [here](https://blog.postman.com/2019/07/17/self-signed-ssl-certificate-troubleshooting/) is a some info about self-signed certs blocked by a tool in this field. 
As I have checked  a GET request from `https_client` on the server hosted on other esp32 was successful.

Zeni241, 2020-04-23:
Thanks for response. In fact I discovered this is an issue because the certificate is not issued by some known authority. I will have to add this certificate as trusted in my app.

closed 2020-04-23 by Zeni241