espressif/esp-idf #4893: esp_netif_set_ip_info always returning ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED

sharkx, 2020-03-05:
I have experienced an issue trying to change the SoftAP IP address while using the esp_netif_ APIs; more specifically, no matter what I do, I get the `ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED` error, even if I try to start/stop the DHCP server beforehand.

A call to `esp_netif_dhcps_get_status` always returns 0 for me (`ESP_NETIF_DHCP_INIT`, aka not yet started), which makes me think that maybe the test for a running DHCP server or client (file esp_netif_lwip.c:1202) should be changed from :

```c++
if (esp_netif->flags & ESP_NETIF_DHCP_SERVER) {
    if (esp_netif->dhcps_status != ESP_NETIF_DHCP_STOPPED) {
        return ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED;
    }
} else if (esp_netif->flags & ESP_NETIF_DHCP_CLIENT) {
    if (esp_netif->dhcpc_status != ESP_NETIF_DHCP_STOPPED) {
        return ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED;
    }
}
```

to

```c++
if (esp_netif->flags & ESP_NETIF_DHCP_SERVER) {
    if (esp_netif->dhcps_status == ESP_NETIF_DHCP_STARTED) {
        return ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED;
    }
} else if (esp_netif->flags & ESP_NETIF_DHCP_CLIENT) {
    if (esp_netif->dhcpc_status == ESP_NETIF_DHCP_STARTED) {
        return ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED;
    }
}
```
or some other equivalent.

david-cermak, 2020-03-06:
Hi @sharkx 

The `esp_netif_set_ip_info()` works okay for me, example:
``` c
    netif = esp_netif_create_default_wifi_ap();

    ... // init the wifi AP

    res = esp_netif_dhcps_get_status(netif, &status); // status : 1 (ESP_NETIF_DHCP_STARTED)
    ESP_ERROR_CHECK(esp_netif_dhcps_stop(netif));
    res = esp_netif_dhcps_get_status(netif, &status); // status: 2 (ESP_NETIF_DHCP_STOPPED)

    esp_netif_ip_info_t ip_info = { ...  };
    res = esp_netif_set_ip_info(netif, &ip_info); // res: 0 (ESP_OK)
    ESP_ERROR_CHECK(esp_netif_dhcps_start(netif));
```
after that the AP's DHCP sever uses the update ip_info

sharkx, 2020-03-06:
Thanks. Maybe it's because I have a different start-up sequence; I need the module to connect first to the network, then act as an AP for others (something like a mesh). Here's my scenario (briefly, without too much details) :

```
1. in app_main :
    1. esp_netif_init();
    2. esp_netif_create_default_wifi_sta()
    3. esp_netif_create_default_wifi_ap()
    4. esp_wifi_init()
    5. esp_wifi_set_mode(WIFI_MODE_STA)
    6. esp_wifi_start()
    7. esp_wifi_scan_start(...);

2. when I get WIFI_EVENT_SCAN_DONE
    1. esp_wifi_set_config(...)
    2. esp_wifi_connect()

3. when I get IP_EVENT_STA_GOT_IP
    1. esp_wifi_set_mode(WIFI_MODE_APSTA)
    2. esp_wifi_set_config(...)
    3. esp_netif_set_ip_info    <------------------------
    4. esp_netif_dhcps_option(...)
    5. esp_netif_dhcps_start()
```
It's the (3) that fails. I have used `esp_netif_dhcps_get_status` to check dhcp status before `esp_netif_set_ip_info` and it returns 0 (INIT); I have also added some ESP_LOGs in `esp_netif_lwip.c` and got `esp_netif->dhcps_status = 0`.

I have worked around this by changing the test from `!= ESP_NETIF_DHCP_STOPPED` to `== ESP_NETIF_DHCP_STARTED` and rebuilding the modules. After all, it's a tri-state value, and !RUNNING means any of the other two states, right ?

david-cermak, 2020-03-06:
> It's the (3) that fails. I have used esp_netif_dhcps_get_status to check dhcp status before esp_netif_set_ip_info and it returns 0 (INIT); I have also added some ESP_LOGs in esp_netif_lwip.c and got esp_netif->dhcps_status = 0.

Are you checking the right interface here? Please note that this
```
    2. sta_netif = esp_netif_create_default_wifi_sta()
    3. ap_netif = esp_netif_create_default_wifi_ap()
```
actually creates two instances of esp-netif (one AP and one station). Seems like you're checking on the station, not the access point. Please make sure run:
``` cpp
esp_netif_set_ip_info(ap_netif, &ip_info);
```

sharkx, 2020-03-06:
Triple-checked before raising the issue here.

Here's the actual code (without the error checkcing) :

```c++
uint8_t softAPMAC[6];
esp_wifi_get_mac(WIFI_IF_AP, softAPMAC);

// setting up sap_config struct
// ...

// go to APSTA mode
esp_wifi_set_mode(WIFI_MODE_APSTA);

softAPMAC[1] = getAPRank();
esp_wifi_set_mac(WIFI_IF_AP, softAPMAC);

esp_wifi_get_mac(WIFI_IF_AP, softAPMAC);
ESP_LOGI(TAG, "New MAC : "MACSTR"", MAC2STR(softAPMAC));

esp_wifi_set_config(ESP_IF_WIFI_AP, &sap_config);

// set my IP address
esp_netif_ip_info_t ap_info;
esp_netif_get_ip_info(g_sap_if, &ap_info);

ESP_LOGI(TAG, "Current IP : "IPSTR"", IP2STR(&ap_info.ip));

// reset_dhcp_server();

if ((ip4_addr1(&ap_info.ip) != 10) || (ip4_addr3(&ap_info.ip) != getAPRank()))
{
	ESP_LOGI(TAG, "Current IP does not match");
	IP4_ADDR(&ap_info.ip, 		 10,   0, getAPRank(), 1);
	IP4_ADDR(&ap_info.gw, 		 10,   0, getAPRank(), 1);
	IP4_ADDR(&ap_info.netmask, 	255, 255, 255, 0);

	esp_netif_set_ip_info(g_sap_if, &ap_info);
}
```

The line that says `reset_dhcp_server()` tries to start then stop the dhcp (initially I hought this might change the internal status, but doesn't seem to change anything); `esp_netif_set_ip_info` still returns RUNNING, but internal status is actually INIT.

The problem could be elsewhere, but if it is, I cannot find it.

sharkx, 2020-03-06:
I have just added this line of code before the running dhcp test (esp_netif_lwip.c:1198, just after the existing `ESP_LOGD`).

```c++
ESP_LOGE(TAG,
	"esp_netif->flags=%08x, esp_netif->dhcps_status=%08x",
	(int)esp_netif->flags,
	(int)esp_netif->dhcps_status);

```
and I get this in the terminat :
```esp_netif_lwip: esp_netif->flags=00000006, esp_netif->dhcps_status=00000000```

sharkx, 2020-03-06:
Forgot to mention that I updated my local copy of the esp-idf repo yesterday (5th of march 2020), just to be sure I'm not using outdated code.

david-cermak, 2020-03-06:
Still cannot reproduce. Tried to follow the steps outlined above, to essentially init both AP, STA. Start station, connect, start AP in AP+STA mode, alter the AP's IP (added the internal logs), still getting the expected:
```
E (20776) esp_netif_lwip: esp_netif->flags=00000006, esp_netif->dhcps_status=00000001
I (20776) wifi softAP: res = 0
I (20776) wifi softAP: status = 1
E (20786) esp_netif_lwip: esp_netif->flags=00000006, esp_netif->dhcps_status=00000002
I (20796) wifi softAP: res = 0
I (20796) wifi softAP: status = 2
E (20806) esp_netif_lwip: esp_netif->flags=00000006, esp_netif->dhcps_status=00000001
I (20816) wifi softAP: res = 0
I (20816) wifi softAP: status = 1
```

Could you please reduce the project to a simple snippet and post it somewhere?

Alvin1Zhang, 2020-05-25:
@sharkx Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

sharkx, 2020-05-25:
Hello,
I haven't experienced any more problems since I changed 
```
if (esp_netif->dhcps_status != ESP_NETIF_DHCP_STOPPED)
```
to
```
if (esp_netif->dhcps_status == ESP_NETIF_DHCP_STARTED)
```
Like described in the original report.
But if the problem cannot be reproduced on your side, it must be something I do differently or wrong.
Best regards.

Alvin1Zhang, 2020-08-19:
@sharkx Thanks for reporting, and sorry that we still cannot reproduce the issue. Feel free to reopen if the issue happens again at your side/help provide details when the issue happens.

closed 2020-08-19 by Alvin1Zhang

chipweinberger, 2022-03-31:
I am also hitting this issue using v4.4.

My DHCP status is "Init", then I call "stop".  I must be getting the chain of events wrong. Can someone share/describe a working example?

```
I (383) pd_wifi_start()
I (383) pd wifi: esp_netif_init()
I (393) pd wifi: esp_event_loop_create_default()
I (393) pd wifi: esp_netif_create_default_wifi_ap()
I (403) pd wifi: esp_wifi_init()
I (503) pd wifi: esp_event_handler_instance_register(ESP_EVENT_ANY_ID, ...)
I (513) pd wifi: esp_event_handler_instance_register(IP_EVENT_STA_GOT_IP, ...)
I (523) pd wifi: esp_wifi_set_mode(WIFI_MODE_APSTA)
I (533) pd wifi: esp_wifi_set_config(...)
I (1383) pd wifi: esp_wifi_start()
I (1473) pd wifi:  Wifi Event: STA Mode Start
I (1483) pd wifi: dhcp status: Init
I (1483) pd wifi: esp_netif_dhcpc_stop(...)
I (6493) pd wifi: esp_netif_set_ip_info(...)
E (6493) : e1 : [pd wifi] esp_netif_set_ip_info() failed, ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED
```