espressif/esp-idf #4895: ESP HFP API Build Error - Undefined reference to ...

searock35, 2020-03-05:
## Environment

- Development Kit:   ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): Master
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.2
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I am trying to use the HFP AG API that was recently added. I have written some code that begins to implement the API but I can't get the file to build properly. When I build, It fails on the "linking CXX executable" stage for the elf file. It says that there are undefined references to the functions I reference from the esp_hf_ag_api.h file.

### Expected Behavior
I expect the program to build properly.

### Actual Behavior
It fails during the build on the "Linking CXX Executable" part. See log below.

### Steps to reproduce

1. Include references to the esp_hf_ag_api file and attempt to build the project.
2. See the error unfold

### Code to reproduce this issue

[GIST](https://gist.github.com/searock35/2365883c07dcb3e510d2f43d711a5987)

## Debug Logs

```

C:\Users\cb1562\Documents\esp-idf\examples\bluetooth\bluedroid\classic_bt\hfp_fad>idf.py build
Executing action: all (aliases: build)
Running cmake in directory c:\users\cb1562\documents\esp-idf\examples\bluetooth\bluedroid\classic_bt\hfp_fad\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 c:\users\cb1562\documents\esp-idf\examples\bluetooth\bluedroid\classic_bt\hfp_fad"...
Warn about uninitialized values.
-- Found Git: C:/Users/cb1562/AppData/Local/Programs/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/cb1562/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/cb1562/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/cb1562/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/cb1562/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/cb1562/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Loading defaults file C:/Users/cb1562/Documents/esp-idf/examples/bluetooth/bluedroid/classic_bt/hfp_fad/sdkconfig.defaults...
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs-time.ld
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/cb1562/Documents/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Found PythonInterp: C:/Users/cb1562/.espressif/python_env/idf4.2_py3.5_env/Scripts/python.exe (found version "3.5.1")
[... 4 lines cut ...]

(skipped some lines here)

[96/99] Linking C static library esp-idf\spi_flash\libspi_flash.a
[97/99] Linking C static library esp-idf\main\libmain.a
[98/99] Linking C executable bootloader.elf
[99/99] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/cb1562/Documents/esp-idf/examples/bluetooth/bluedroid/classic_bt/hfp_fad/build/bootloader/bootloader.bin
[1072/1173] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ccache C:\Users\cb1562\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../../../../../../components/newlib/platform_include -I../../../../../../components/freertos/include -I../../../../../../components/freertos/xtensa/include -I../../../../../../components/heap/include -I../../../../../../components/log/include -I../../../../../../components/lwip/include/apps -I../../../../../../components/lwip/include/apps/sntp -I../../../../../../components/lwip/lwip/src/include -I../../../../../../components/lwip/port/esp32/include -I../../../../../../components/lwip/port/esp32/include/arch -I../../../../../../components/soc/src/esp32/. -I../../../../../../components/soc/src/esp32/include -I../../../../../../components/soc/include -I../../../../../../components/esp_rom/include -I../../../../../../components/esp_common/include -I../../../../../../components/xtensa/include -I../../../../../../components/xtensa/esp32/include -I../../../../../../components/esp32/include -I../../../../../../components/driver/include -I../../../../../../components/driver/esp32/include -I../../../../../../components/esp_ringbuf/include -I../../../../../../components/efuse/include -I../../../../../../components/efuse/esp32/include -I../../../../../../components/esp_timer/include -I../../../../../../components/soc/soc/esp32/. -I../../../../../../components/soc/soc/esp32/include -I../../../../../../components/soc/soc/esp32/../include -I../../../../../../components/vfs/include -I../../../../../../components/esp_wifi/include -I../../../../../../components/esp_wifi/esp32/include -I../../../../../../components/esp_event/include -I../../../../../../components/esp_netif/include -I../../../../../../components/esp_eth/include -I../../../../../../components/tcpip_adapter/include -I../../../../../../components/app_trace/include -I../../../../../../components/mbedtls/port/include -I../../../../../../components/mbedtls/mbedtls/include -I../../../../../../components/wpa_supplicant/include -I../../../../../../components/wpa_supplicant/port/include -I../../../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../../../components/bootloader_support/include -I../../../../../../components/app_update/include -I../../../../../../components/spi_flash/include -I../../../../../../components/nvs_flash/include -I../../../../../../components/pthread/include -I../../../../../../components/espcoredump/include -I../../../../../../components/perfmon/include -I../../../../../../components/asio/asio/asio/include -I../../../../../../components/asio/port/include -I../../../../../../components/bt/include -I../../../../../../components/bt/common/osi/include -I../../../../../../components/bt/host/bluedroid/api/include/api -I../../../../../../components/cbor/port/include -I../../../../../../components/coap/port/include -I../../../../../../components/coap/port/include/coap -I../../../../../../components/coap/libcoap/include -I../../../../../../components/coap/libcoap/include/coap2 -I../../../../../../components/console -I../../../../../../components/nghttp/port/include -I../../../../../../components/nghttp/nghttp2/lib/includes -I../../../../../../components/esp-tls -I../../../../../../components/esp_adc_cal/include -I../../../../../../components/esp_gdbstub/include -I../../../../../../components/tcp_transport/include -I../../../../../../components/esp_http_client/include -I../../../../../../components/esp_http_server/include -I../../../../../../components/esp_https_ota/include -I../../../../../../components/protobuf-c/protobuf-c -I../../../../../../components/protocomm/include/common -I../../../../../../components/protocomm/include/security -I../../../../../../components/protocomm/include/transports -I../../../../../../components/ [... cut ...] in.c
../main/main.c:30:10: fatal error: bt_app_hf.h: No such file or directory
 #include "bt_app_hf.h"
          ^~~~~~~~~~~~~
compilation terminated.
[1081/1173] Building C object esp-idf/openssl/CMakeFiles/__idf_openssl.dir/platform/ssl_pm.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

## Other items if possible

(I'm really sorry, I don't know how to link or attach my project files or logs here. Please ask me for more stuff when you will inevitably need it)

AbnerFederer, 2020-03-06:
Hi,

Maybe you can remove the `bt_app_hf.h` file.

Thanks
Abner

searock35, 2020-03-06:
Hi AbnerFederer,
Thanks for responding. I removed that reference before I ran into this build error but mistakenly left it in the code. I changed the code on the GIST link to reflect that. 
My apologies,
Corey

renzbagaporo, 2020-03-09:
So does that mean this can be closed @searock35?

searock35, 2020-03-11:
Sorry, I should have made it more clear that the problem still persists. I am gonna try building it in the Eclipse build environment soon and maybe that will help?

renzbagaporo, 2020-03-11:
Can you build the [HFP examples](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt)?

renzbagaporo, 2020-03-24:
Any updates @searock35 ?

searock35, 2020-03-24:
@renzbagaporo Sorry, I should've updated a while ago. I moved from my laptop to a different device and everything works fine there. I won't delve too deep into it now. Although I am closing this, I will answer your last question; I was able to build the examples, but my program I could not build. I might have forgotten to update the SDK config file when I did this, I'm not sure. Thanks for helping.

closed 2020-03-24 by searock35