espressif/esp-idf #4899: [BLE MESH] NODE_RESET_STATUS message may not be sent upon ESP_BLE_MESH_NODE_PROV_RESET_EVT

AlpinProgrammer, 2020-03-06:
## Environment

- Kit version : v3
- Module or chip used:  ESP32-WROVER-B
- IDF version v4.0-rc
- Build System:         Make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When resetting a node with a nRF mesh based app, NODE_RESET_STATUS may not be sent after receiving the ESP_BLE_MESH_NODE_PROV_RESET_EVT.

Looking further in the esp-idf source code, we were able to identify where this ACK can be sent. However, we do not have any way to ensure it.
The _node_reset_ function which calls function _bt_mesh_model_msg_init(&msg, OP_NODE_RESET_STATUS);_ is called. However, the function _node_reset_status_ is not called.

### Expected Behavior

After receiving ESP_BLE_MESH_NODE_PROV_RESET_EVT, the ESP32 sends an ACK message (NODE_RESET_STATUS).

### Actual Behavior

The ESP_BLE_MESH_NODE_PROV_RESET_EVT is obtained but the ACK message (NODE_RESET_STATUS) is rarely received by the app.

### Steps to reproduce

1. Provision the ESP32 with a mesh app
2. Reset the node

### Code to reproduce this issue

esp_ble_mesh node example

Thanks in advance for having a look :)

Campou, 2020-03-09:
Hi @AlpinProgrammer 

1. I used the branch [release/v4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) to verify this issue, and the node reset status can be received by the App. And the following is my test log on the device side and test video (within the zip file) of the App side.
    [image]
    [node-reset.mp4.zip](https://github.com/espressif/esp-idf/files/4304835/node-reset.mp4.zip)

2. Could you please checkout to the branch and have a test?

Thanks.

AlpinProgrammer, 2020-03-19:
Hi @Campou , thank you very much for your fast answer.
I will perform the test on the release/v4.0, check the results and let you know :)

AlpinProgrammer, 2020-03-27:
Hello again !
I checked on release/v4.0 branch and it works as expected.

Thanks for your precious help :)

PS: Do you know when a v4.0.1 adding those fixes is expected to be available ?

closed 2020-03-27 by AlpinProgrammer