espressif/esp-idf #4900: Maximum tcp client connection number does not increase.

TarikAkyuz, 2020-03-06:
A tcp server is established in Esp32(station mode).
I accept the clients with the code below.
```
        int sock = accept(listen_sock, (struct sockaddr *)&source_addr, &addr_len);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to accept connection: errno %d", errno);
            break;
        }
```

When 20 tcp clients send requests to connect at the same time, 9 can connect. But in my system, 1000 clients on the field will have to connect to the esp32 server at the same time.

Although I made "Maximum active TCP Connections" and "Maximum listening TCP Connections" 1000 in Menuconfig(LWIP->TCP), the number of connections did not change.

Only when I changed the "Max number of open sockets" in menuconfig, I was able to increase the number of connections.

Esp32 will be connected to the network in "station" mode and create tcp server. The other 1000 esp32 will connect to this as client.

Is it possible?How should I set tcp server, if possible?

freakyxue, 2020-03-12:
hi @TarikAkyuz 
Do you want to establish a 1000 socket connection on a device?
In menuconfig  ,The upper limit of socket is 16.
because sockets take up a certain amount of memory, and allowing fewer
sockets to be open at the same time conserves memory. Specify 
the maximum amount of sockets here. The valid value is from 1 to 16.

freakyxue, 2020-03-18:
hi @TarikAkyuz
Do you have any questions about this problem? I will close this problem. If yes, please reopen it

Alvin1Zhang, 2020-05-25:
@TarikAkyuz Thanks for reporting. Will close due to short of feedback, feel free to reopen if any more updates. Thanks.

closed 2020-05-25 by Alvin1Zhang