espressif/esp-idf #4901: [NVS] ram is not released after nvs_close

SamyRICHET, 2020-03-06:
hi, 

ESP-IDF: 3.3.1
dev_board: ESP32-WROOM-32
Toolchain: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
chip rev: 1

On my project, I need to use the nvs library for storing a significant amount of uint32_t.
But when I use nvs_close to release the RAM memory that used by NVS process, nothing is happening. I don't recovered the quantity of RAM that's used during NVS process.

On bellow, I copy paste a dirty code based on nvs_rw_value example that shows this issue.

```
#include <stdio.h>
#include <esp_event_loop.h>
#include <esp_log.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"

static const char *TAG = "app";
void app_main()
{
    // Initialize NVS
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    // Open
    printf("\n");
    printf("Opening Non-Volatile Storage (NVS) handle... ");

    ESP_LOGI(TAG, " free-ram:%u", esp_get_free_heap_size() );
	ESP_LOGI(TAG, " free-ram-minimun:%u", esp_get_minimum_free_heap_size() );

    nvs_handle my_handle;
    err = nvs_open("storage", NVS_READWRITE, &my_handle);

    ESP_LOGI(TAG, " free-ram:%u", esp_get_free_heap_size() );
	ESP_LOGI(TAG, " free-ram-minimun:%u", esp_get_minimum_free_heap_size() );

    if (err != ESP_OK) {
        printf("Error (%s) opening NVS handle!\n", esp_err_to_name(err));
    } else {
        printf("Done\n");

        // Read
        printf("Reading restart counter from NVS ... ");
        int32_t restart_counter=0; // value will default to 0, if not set yet in NVS
        for (int i = 0; i < 512; ++i)
        {
        	char* keyName;

			asprintf(&keyName, "%u", i);
        	err = nvs_get_i32(my_handle, keyName, &restart_counter);
        	/* code */
        }
        
        switch (err) {
            case ESP_OK:
                printf("Done\n");
                printf("Restart counter = %d\n", restart_counter);
                break;
            case ESP_ERR_NVS_NOT_FOUND:
                printf("The value is not initialized yet!\n");
                break;
            default :
                printf("Error (%s) reading!\n", esp_err_to_name(err));
        }

        // Write
        printf("Updating restart counter in NVS ... ");
        restart_counter++;
                for (int i = 0; i < 512; ++i)
        {
        	char* keyName;

			asprintf(&keyName, "%u", i);
        	err = nvs_set_i32(my_handle, keyName, restart_counter);
        	/* code */
        }
        
        printf((err != ESP_OK) ? "Failed!\n" : "Done\n");

        // Commit written value.
        // After setting any values, nvs_commit() must be called to ensure changes are written
        // to flash storage. Implementations may write to storage at other times,
        // but this is not guaranteed.
        printf("Committing updates in NVS ... ");
        err = nvs_commit(my_handle);
        printf((err != ESP_OK) ? "Failed!\n" : "Done\n");

		ESP_LOGI(TAG, " free-ram:%u", esp_get_free_heap_size() );
		ESP_LOGI(TAG, " free-ram-minimun:%u", esp_get_minimum_free_heap_size() );

        // Close
        nvs_close(my_handle);
    }

    printf("\n");

		ESP_LOGI(TAG, " free-ram:%u", esp_get_free_heap_size() );
		ESP_LOGI(TAG, " free-ram-minimun:%u", esp_get_minimum_free_heap_size() );

    // Restart module
    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}
```

Thanks you for your help.

Best regard.

igrr, 2020-03-06:
Could you please try adding a call to `nvs_flash_deinit`? NVS maintains a cache at partition level, not namespace level, so calling nvs_close doesn't free the cache. Calling nvs_flash_deinit should do the trick.

SamyRICHET, 2020-03-09:
I try to add nvs_flash_deinit but I have the same result.

igrr, 2020-03-10:
@SamyRICHET what difference between the initial and final free heap size are you observing, after adding nvs_flash_deinit call?

ameyinamdar, 2020-03-11:
@SamyRICHET I think there is memory leak in application code due to asprintf call which internally allocates the buffer of sufficient size to store the key value string. It never is not freed anywhere.

SamyRICHET, 2020-03-11:
@ameyinamdar You are right. It's my fault, I forgot to free the buffer. Thank you for your help.
@igrr thanks you too

closed 2020-03-11 by SamyRICHET