espressif/esp-idf #4902: can example

nopnop2002, 2020-03-06:
examples/peripherals/can/can_network/can_network_master and
examples/peripherals/can/can_network/can_network_slave don't work.

My Environment:
```
freeRTOS version:V8.2.0
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.1-dev-1935-g647cb628a
```

Transceiver:
SN65HVD230

Wire:
```
    ----------   ----------   --------------  
   |  Master  | |  Slave   | | Listen Only  | 
   |          | |          | |              | 
   | 21    22 | | 21    22 | | 21    22     | 
    ----------   ----------   --------------  
     |      |     |      |     |      |   
     |      |     |      |     |      |   
    ----------   ----------   ----------  
   | D      R | | D      R | | D      R | 
   |          | |          | |          | 
   |  VP230   | |  VP230   | |  VP230   | 
   |          | |          | |          | 
   | H      L | | H      L | | H      L | 
    ----------   ----------   ----------  
     |      |     |      |     |      |   
     |      |     |      |     |      |   
  |--x------|-----x------|-----x------|--| H
            |            |            |
  |---------x------------x------------x--| L

```

Master:
[image]

Slave:
[image]

Dazza0, 2020-03-09:
@nopnop2002 

It means that the CAN Master is not able to ping the slave. The ping is a single shot Data Frame that the CAN master will keep sending until the CAN slaves responds with another Data Frame.

Could you check that the CAN bus and transceivers are connected properly by confirming with a logic analyzer that the RX pin of the CAN Master and CAN Slave are actually receiving data?

nopnop2002, 2020-03-11:
@Dazzo0
I do not have a logic analyzer.
I will try it on esp-idf Ver3.3 on the weekend.

nopnop2002, 2020-03-14:
I tried with ESP-IDF Ver3.3 but it doesn't work.

The transmitter is a 5V transceiver and the receiver is a 3.3V transceiver.
There seems to be a problem here.

Sorry for the trouble.

closed 2020-03-14 by nopnop2002

xuhongv, 2022-05-23:
did you resolve it ? i have the same problem @nopnop2002

nopnop2002, 2022-05-23:
@xuhongv 

The problem was solved by adjusting the value of the terminating resistor.