espressif/esp-idf #4903: ESP peripheral + iOS 13.4: ignored BLE attribute read by type

zacwbond, 2020-03-06:
## Environment

- Development Kit:      none (custom HW)
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-369-g4dac7c7df
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): gcc8_2_0-esp-2019r1-linux-amd64
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

We have our ESP configured as a GATT server, and have Android and iOS apps that connect to it. 
However, we have just begun having a problem with iPhones running iOS 13.4, and only those iPhones which also support Bluetooth 5.  

It looks like a failure of the BLE stack to process a GATT Read By Type request for the device model string in the device information service shortly after discovery.  30 seconds later, the iOS device disconnects because it never receives a response.

In the ESP debug (BLE GATT and L2CAP debug is turned on), I can see that it receives a message:
```
D (219742) BT_L2CAP: L2CAP - rcv_cid CID: 0x0004
```

But there's nothing else until the disconnect message 30s later, whereas there's lots of L2CAP debug for all other messages.  It's as if the stack is just ignoring it.

I've attached a WireShark log, and the ESP debug log follows.
```
E (200913) Heap: 83000
E (200913) HeapMin: 78000
E (210913) Heap: 82992
E (210913) HeapMin: 78000
D (215435) BT_L2CAP: l2c_ble_link_adjust_allocation  num_hipri: 0  num_lowpri: 1  low_quota: 10  round_robin_quota: 0  qq: 10
D (215435) BT_L2CAP: l2c_ble_link_adjust_allocation LCB 0   Priority: 0  XmitQuota: 10
D (215436) BT_L2CAP:         SentNotAcked: 0  RRUnacked: 0
D (215437) BT_L2CAP: l2cu_allocate_ccb: cid 0x0000
D (215438) BT_L2CAP: l2c_link_adjust_chnl_allocation
D (215438) BT_L2CAP: CID:0x0043 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (215572) BT_L2CAP: l2cu_process_fixed_chnl_resp
D (215572) BT_GATT: GATT   ATT protocol channel with BDA: 52c52dd739e1 is connected
D (215572) BT_GATT: gatt_is_bda_in_the_srv_chg_clt_list :52-c5-2d-d7-39-e1
D (215573) BT_GATT: gatt_set_ch_state: old=0 new=4
D (215573) BT_GATT: gatt_connect_cback: from 52c52dd739e1 connected:1 conn_id=1 reason = 0x0000
D (215575) BT_GATT: gatt_get_ch_state: ch_state=4
I (215575) BT_GATT: GATT_GetConnIdIfConnected status=1

D (215578) BT_GATT: gatt_num_apps_hold_link   num=0
I (215578) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (215580) Esp Ble Server: gatt server event: ESP_GATTS_CONNECT_EVT on interface: 4
I (215581) Esp Ble Server: phone connected!
I (215581) Esp Ble Server: Restart advertising due to connect.
I (215582) Esp Ble Client: gatt server event: ESP_GATTC_CONNECT_EVT on interface: 3
E (215583) Esp Ble Client: Connection to E1:39:D7:2D:C5:52 was started when connection to 00:00:00:00:00:00 was already in progress.
I (215583) GOT ACK!:
D (215585) BT_L2CAP: l2c_link_check_send_pkts
D (215585) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (215586) BT_L2CAP: l2c_link_send_to_lower
D (215587) BT_L2CAP: TotalWin=9,Hndl=0x0,Quota=10,Unack=1,RRQuota=0,RRUnack=0
I (215590) SEND ACK!:
I (215591) Ble Driver: GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (215591) Ble Driver: Advertising Succeeded!
D (215631) BT_L2CAP: l2c_link_check_send_pkts
D (215631) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (215631) BT_L2CAP: TotalWin=10,LinkUnack(0x0)=0,RRCheck=0,RRUnack=0

D (215632) BT_L2CAP: l2cble_process_data_length_change_event TX data len = 251
D (215662) BT_L2CAP: L2CAP - rcv_cid CID: 0x0004

D (215662) BT_GATT: gatt_get_ch_state: ch_state=4
I (215662) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 52c52dd739e1
D (215663) BT_L2CAP: l2c_link_check_send_pkts
[... 320 lines cut ...]
D (249723) BT_GATT: ATT disconnected
I (249724) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (249725) Esp Ble Server: gatt server event: ESP_GATTS_DISCONNECT_EVT on interface: 4
I (249725) Esp Ble Server: phone disconnect!
I (249726) Esp Ble Server: Restart advertising due to disconnect.
I (249727) Esp Ble Client: gatt server event: ESP_GATTC_DISCONNECT_EVT on interface: 3
I (249728) Esp Ble Client: connection failed for unknown device
I (249728) GOT ACK!:
I (249729) Esp Ble Client: 52 c5 2d d7 39 e1
I (249730) Esp Ble Client: Restart scanning due to disconnect
I (249736) Ble Driver: GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (249736) Ble Driver: Advertising Succeeded!
E (250913) Heap: 82952
E (250913) HeapMin: 78000
I (252590) SEND ACK!:
V (257724) esp_adapter: thread sem get: sem=0x3ffe8098
D (259774) event: running post WIFI_EVENT:1 with handler 0x400e2ac4 on loop 0x3ffcbb24
E (259774) ESP Wifi: Event received: WIFI_EVENT_SCAN_DONE
V (259827) esp_adapter: thread sem get: sem=0x3ffe8098
I (259990) SEND ACK!:
E (260913) Heap: 83484
E (260913) HeapMin: 78000
```

[iOS_ConnFail_BLE5.pcapng.log](https://github.com/espressif/esp-idf/files/4300721/iOS_ConnFail_BLE5.pcapng.log)

TallDaveWeiss, 2020-03-31:
Any update on this issue?

Alvin1Zhang, 2020-03-31:
@TallDaveWeiss Thanks for asking the updates, sorry for the slow turnaround, we are looking into the issue, and will provide feedback once available. Thanks.

Alvin1Zhang, 2020-03-31:
@zacwbond @TallDaveWeiss Thanks for reporting and updating, we already have a fix internally now under reviewing, will update once it is available on GitHub. Thanks.

stuarthatchbaby, 2020-03-31:
We're seeing this issue too and it's preventing our customers from registering new devices!

This is critical - please treat this as your highest priority!

AndroidBaaaby, 2020-03-31:
@stuarthatchbaby Sure. We have put the MR in the highest priority.

stuarthatchbaby, 2020-03-31:
We are standing by to help confirm this resolution. Can we get details of the MR ?

hm0429, 2020-04-03:
any updates on this? thanks!

Alvin1Zhang, 2020-04-03:
@hm0429 @stuarthatchbaby @TallDaveWeiss @zacwbond Thanks for following up the updates. The fix has been merged and available on the master branch and release/4.1 branches on GitHub now. The fix will also be back ported to other release branches. Thanks for your patience.

canbempbell, 2020-04-07:
@Alvin1Zhang we're getting a very similar issue with 3.2.2 since iOS 13.4. Could you please specify the commit which fixes this issue and do you plan to backport it to 3.2? When it'll be available on 3.2?

Alvin1Zhang, 2020-04-08:
@canbempbell Thanks for the follow up. On master branch, the fix is https://github.com/espressif/esp-idf/commit/dcc2963d431ef7eccc4c0abd0a95781698bb4e75 ; on release/v4.1, the fix is https://github.com/espressif/esp-idf/commit/5e09d473c70367ce3899706db48ce3313162c0c4 . We do have plan to back port this fix to IDF v3.2, currently we are working on the release planning of IDF v3.2.4, which is the next bug fix release of IDF v3.2, will update you once the fix is available on release/3.2. Thanks.

canbempbell, 2020-04-08:
@Alvin1Zhang Good. Thanks a lot!

phatpaul, 2020-04-28:
I've been searching for this for a while now.  Our product is now broken on iOS 13.4 and Android 10 (at least on Samsung devices)!  https://esp32.com/viewtopic.php?f=13&t=15344

What about porting to IDF 3.3?
Thanks.

zacwbond, 2020-04-28:
FYI, my application has been migrated to Nimble instead of Bluedroid, so I'm not longer able to easily verify fixes for this issue.  I think one of the other folks who have experienced this problem will need to verify & close.  Thanks!

phatpaul, 2020-04-28:
I can confirm my device now works with iOS 13.4.1 on an iPhone 8 thanks to this fix.  

I still need ask my customers who were unable to connect to my device to upgrade the firmware and test with their devices:
iPhone XR, iOS 13.4.1
Samsung Note 10+, Android 10.?
Samsung S20 Ultra, Android 10.?

Alvin1Zhang, 2020-05-25:
@phatpaul Thanks for verifying. Would you please help share if any more updates? Thanks.

rahult-github, 2022-05-23:
Closing this ticket for lack of updates. Please feel to reopen if issue still observed.

closed 2022-05-23 by rahult-github