espressif/esp-idf #4905: UART_0 Missing Bytes

bsayiner, 2020-03-07:
## Environment
- Development Kit: Custom PCB
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-dev-532-g6f7be93d5
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: Eclipse
- Power Supply: External 3.3V
## Problem Description
Uart 0 Missing Bytes. When I read from UART_0 some bytes are missing. But If I write similar code with Arduino ESP, it is working very well. I'm sure that its not related with hardware. Interesting somethink if i touch rx pin uart its working normal.
### Expected Behavior
Expected receive bytes: 90 02 00 04 00 00 00 00 f4 00 00 00 a7 92
### Actual Behavior
Actual received bytes: 08 00 00 f4 00 00 00 a7 92
### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "esp_log.h"

#define ECHO_TEST_TXD  (GPIO_NUM_1)
#define ECHO_TEST_RXD  (GPIO_NUM_3)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)

#define BUF_SIZE (1024)

static const char *TAG = "UART_EXAMPLE";

static void echo_task(void *arg) {
	/* Configure parameters of an UART driver,
	 * communication pins and install the driver */
	uart_config_t uart_config = { .baud_rate = 115200, .data_bits = UART_DATA_8_BITS, .parity = UART_PARITY_DISABLE, .stop_bits = UART_STOP_BITS_1, .flow_ctrl = UART_HW_FLOWCTRL_DISABLE, .source_clk = UART_SCLK_APB, };
	uart_driver_install(UART_NUM_0, BUF_SIZE * 2, 0, 0, NULL, 0);
	uart_param_config(UART_NUM_0, &uart_config);
	uart_set_pin(UART_NUM_0, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);

	// Configure a temporary buffer for the incoming data

	uint8_t txBuffer[] = { 0x90, 0x02, 0x60, 0x02, 0x01, 0x00, 0x22, 0x01, 0xbd, 0x92 };
	uint8_t rxBuffer[BUF_SIZE];

	while (1) {
		uart_write_bytes(UART_NUM_0, (const char*) txBuffer, 10);
		// Read data from the UART
		uint64_t now = esp_timer_get_time();
		int num_of_bytes = uart_read_bytes(UART_NUM_0, rxBuffer, BUF_SIZE, 50 / portTICK_RATE_MS);
		// Write data back to the UART
		ESP_LOGI(TAG, "Elapsed Time: %d", (int ) (esp_timer_get_time() - now) / 1000);
		ESP_LOGI(TAG, "Number of bytes: %d", num_of_bytes);
		ESP_LOG_BUFFER_HEX_LEVEL(TAG, rxBuffer, num_of_bytes, ESP_LOG_INFO);
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}

void app_main(void) {
	gpio_config_t io_conf;
	io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
	io_conf.mode = GPIO_MODE_OUTPUT;
	io_conf.pin_bit_mask = 1ULL << GPIO_NUM_16;
	io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
	io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
	ESP_ERROR_CHECK(gpio_config(&io_conf));
	ESP_ERROR_CHECK(gpio_set_level(GPIO_NUM_16, 1));
	xTaskCreate(echo_task, "uart_echo_task", 4096, NULL, 10, NULL);
}

```
## Debug Logs
```
I (29) boot: ESP-IDF v4.2-dev-532-g6f7be93d5 2nd stage bootloader
I (29) boot: compile time 21:12:46
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x070b4 ( 28852) map
I (123) esp_image: segment 1: paddr=0x000170dc vaddr=0x3ffb0000 size=0x02180 (  8576) load
I (127) esp_image: segment 2: paddr=0x00019264 vaddr=0x40080000 size=0x00404 (  1028) load
I (130) esp_image: segment 3: paddr=0x00019670 vaddr=0x40080404 size=0x069a8 ( 27048) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15b8c ( 88972) map
I (184) esp_image: segment 5: paddr=0x00035bb4 vaddr=0x40086dac size=0x0398c ( 14732) load
I (197) boot: Loaded app from partition at offset 0x10000
I (197) boot: Disabling RNG early entropy source...
I (197) cpu_start: Pro cpu up.
I (201) cpu_start: Application information:
I (206) cpu_start: Project name:     app-template
I (211) cpu_start: App version:      1
I (216) cpu_start: Compile time:     Mar  6 2020 21:12:42
I (222) cpu_start: ELF file SHA256:  f26966330831c6e1...
I (228) cpu_start: ESP-IDF:          v4.2-dev-532-g6f7be93d5
I (234) cpu_start: Starting app cpu, entry point is 0x40081098
I (0) cpu_start: App cpu up.
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (251) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (258) heap_init: At 3FFB2970 len 0002D690 (181 KiB): DRAM
I (264) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (270) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (277) heap_init: At 4008A738 len 000158C8 (86 KiB): IRAM
I (283) cpu_start: Pro cpu start user code
I (301) spi_flash: detected chip: generic
I (301) spi_flash: flash io: dio
W (302) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (323) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (383) UART_EXAMPLE: Elapsed Time: 44
I (383) UART_EXAMPLE: Number of bytes: 0
I (1433) UART_EXAMPLE: Elapsed Time: 49
I (1433) UART_EXAMPLE: Number of bytes: 0
I (2483) UART_EXAMPLE: Elapsed Time: 49
I (2483) UART_EXAMPLE: Number of bytes: 0
I (3533) UART_EXAMPLE: Elapsed Time: 49
I (3533) UART_EXAMPLE: Number of bytes: 0
I (4583) UART_EXAMPLE: Elapsed Time: 49
I (4583) UART_EXAMPLE: Number of bytes: 0
I (5633) UART_EXAMPLE: Elapsed Time: 49
I (5633) UART_EXAMPLE: Number of bytes: 0
I (6683) UART_EXAMPLE: Elapsed Time: 49
I (6683) UART_EXAMPLE: Number of bytes: 0
I (7733) UART_EXAMPLE: Elapsed Time: 49
I (7733) UART_EXAMPLE: Number of bytes: 0
I (8783) UART_EXAMPLE: Elapsed Time: 49
I (8783) UART_EXAMPLE: Number of bytes: 0
I (9833) UART_EXAMPLE: Elapsed Time: 49
I (9833) UART_EXAMPLE: Number of bytes: 0
I (10883) UART_EXAMPLE: Elapsed Time: 49
I (10883) UART_EXAMPLE: Number of bytes: 0
I (11933) UART_EXAMPLE: Elapsed Time: 49
I (11933) UART_EXAMPLE: Number of bytes: 9
I (11933) UART_EXAMPLE: 02 00 00 f4 00 00 00 a7 92
I (12983) UART_EXAMPLE: Elapsed Time: 49
I (12983) UART_EXAMPLE: Number of bytes: 9
I (12983) UART_EXAMPLE: 02 00 00 f4 00 00 00 a7 92
I (14033) UART_EXAMPLE: Elapsed Time: 49
I (14033) UART_EXAMPLE: Number of bytes: 9
I (14033) UART_EXAMPLE: 02 00 00 f4 00 00 00 a7 92
I (15083) UART_EXAMPLE: Elapsed Time: 49
I (15083) UART_EXAMPLE: Number of bytes: 14
I (15083) UART_EXAMPLE: 90 02 00 04 00 00 00 00 f4 00 00 00 a7 92
I (16133) UART_EXAMPLE: Elapsed Time: 49
I (16133) UART_EXAMPLE: Number of bytes: 14
I (16133) UART_EXAMPLE: 90 02 00 04 00 00 00 00 f4 00 00 00 a7 92
I (17183) UART_EXAMPLE: Elapsed Time: 49
I (17183) UART_EXAMPLE: Number of bytes: 14
I (17183) UART_EXAMPLE: 90 02 00 04 00 00 00 00 f4 00 00 00 a7 92
I (18233) UART_EXAMPLE: Elapsed Time: 49
I (18233) UART_EXAMPLE: Number of bytes: 14
I (18233) UART_EXAMPLE: 90 02 00 04 00 00 00 00 f4 00 00 00 a7 92
I (19283) UART_EXAMPLE: Elapsed Time: 49
I (19283) UART_EXAMPLE: Number of bytes: 9
I (19283) UART_EXAMPLE: 08 00 00 f4 00 00 00 a7 92
I (20333) UART_EXAMPLE: Elapsed Time: 49
I (20333) UART_EXAMPLE: Number of bytes: 9
I (20333) UART_EXAMPLE: 10 00 00 f4 00 00 00 a7 92
I (21383) UART_EXAMPLE: Elapsed Time: 49
I (21383) UART_EXAMPLE: Number of bytes: 8
I (21383) UART_EXAMPLE: 40 00 f4 00 00 00 a7 92
I (22433) UART_EXAMPLE: Elapsed Time: 49
I (22433) UART_EXAMPLE: Number of bytes: 9
I (22433) UART_EXAMPLE: 08 00 00 f4 00 00 00 a7 92
```

koobest, 2020-03-16:
Hi,@bsayiner
Can you use UART1 to do this test? Log output is using UART0, But not sure if this is the cause
thanks!!

Alvin1Zhang, 2020-03-30:
@bsayiner Thanks for reporting, will close due to short of feedback, feel free to reopen with more feedback or details. Thanks.

closed 2020-03-30 by Alvin1Zhang

MoraesIBI, 2022-04-12:
Bom dia, vc teve sucesso com o código? por gentileza pode disponibilizar?
meu e-mail mxmoraes@gmail.com
Good morning, did you have success with the code? can you please make it available?