espressif/esp-idf #4906: Project doesn't compile with 8 ULP files index

sch0bert, 2020-03-07:
## Environment

- Development Kit:  **ANY**  [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): **ANY** [v1|v2|v3|v4]
- Module or chip used: **ANY** [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
   **ESP-IDF v4.2-dev-622-g63cd2b061**
- Build System: **idf.py**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   **xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0**
- Operating System:  **Windows**
- (Windows only) environment type: **ESP Command Prompt**
- Using an IDE?: **No**
- Power Supply:  **Doesn't apply**    [USB|external 5V|external 3.3V|Battery]

## Problem Description

My project has 8 assembly files to be compiled for ULP usage and seems like this is impacting somehow the cache from CMake at compilation time. In some point from the 

> Performing configure step for 'ulp_main'

The concatenation from all the paths gets cut in one address and that is causing the project to not compile. 

I looked in to google and seems like the cache doesn't have enough space for that many paths getting indexed in one command. I even get the next error:

> Error: could not load cache

I was able to compile with IDF v4.0 but wasn't functional with the issue that recently got fixed: [4713](https://github.com/espressif/esp-idf/issues/4713)
That's why I now moved to the master branch to get this fix but this is now happening. 

### Expected Behavior

Be able to compile.

### Actual Behavior

Not able to compile, it crashes on the try.

### Steps to reproduce

1. Make a project where ULP is enabled
2. Create 8 assembly files with multiple functionalities.
3. Try to compile the project with the files listed in your CMakeList.txt

## Debug Logs

Log is attached.
[log.txt](https://github.com/espressif/esp-idf/files/4302486/log.txt)

renzbagaporo, 2020-03-09:
Hi @sch0bert, can you try updating ESP-IDF to latest master? We recently fixed an issue with compiling multiple ULP assembly files that might be related: https://github.com/espressif/esp-idf/commit/46e6307e07cd2a19eabee1d56e1858568130ef5e.

renzbagaporo, 2020-03-09:
Hi @sch0bert, have you tried purging the build directory (deleting entire build folder) then rebuilding?

renzbagaporo, 2020-03-09:
Hi @sch0bert, have you tried purging the build directory and then rebuilding?

sch0bert, 2020-03-09:
> Hi @sch0bert, have you tried purging the build directory and then rebuilding?

Yes, I did a fullclean command and also retried deleting the build folder. Same result.

renzbagaporo, 2020-03-10:
Hi @sch0bert, there is an error `CMake Error: The source directory "C:/Users/phernandez/Destop/esp-idf/components/ulp/cmake" does not exist.` in the log you posted. Can you verify that this directory exists?

sch0bert, 2020-03-10:
That also caught my attention but once that I also saw:

` -DPYTHON=C:\Users\phernandez\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe -GNinja C:/Users/phernandez/Des || exit /b `

That make me wonder that because of the cache size, the path got cut off in the "k" part of "Desktop" and that's why this error appears also. I removed the ulp code from cmakelist.txt and re compiled it without a problem so this path should be ok.

renzbagaporo, 2020-03-11:
I can't seem to reproduce your problem @sch0bert. 

>That make me wonder that because of the cache size, the path got cut off in the "k" part of "Desktop" 
 
What I found out is that this sometimes commands get cut short for display purposes in the log. To verify the actual command you should take a look at a `ulp_main-configure-xxxxxxx.bat` in the build directory.

> I removed the ulp code from cmakelist.txt and re compiled it without a problem so this path should be ok, just getting cut.

sch0bert, 2020-03-16:
I'm still unable to compile my code with the latest idf version. 
I'm unable to find the ulp_main-configure-xxxxxxx.bat file in the build folder of my project :(

renzbagaporo, 2020-03-19:
Are you able to compile the ULP examples?

sch0bert, 2020-03-20:
I upgrated my IDF and seems there were a CMake upgrade, probably this fixed the problem because I now able to compile with ULP.

IDF version: ESP-IDF v4.2-dev-781-g47253a827-dirty

Thanks @renzbagaporo!

closed 2020-03-20 by sch0bert