espressif/esp-idf #4907: SDSPI cannot be mounted when the SPI bus has been initialized.

lovyan03, 2020-03-08:
The SPI bus may be shared by multiple devices.
However, if the SPI bus initialization is included in the SDSPI mounting process, and it has already been initialized, it will terminate with an error.

https://github.com/espressif/esp-idf/blob/master/components/driver/sdspi_host.c#L981

In this function, I think that the initialization of the SPI bus and the mounting of the SDSPI should be separated.
Or, if the SPI bus initialization fails, you should try to mount the SD without returning an error.

Currently, cannot unmount SDSPI and then mount it again.
For that purpose, it is necessary to release the SPI bus once.

best regard.

igrr, 2020-03-09:
Hi @lovyan03 `sdspi_host_init_slot` function is deprecated, it is suggested to use `sdspi_host_init_device` instead. The latter solves the problem with sharing the SPI bus between multiple devices.

https://github.com/espressif/esp-idf/blob/f7d1ed5f329c14c356a710c4c10ba50cea277b55/components/driver/sdspi_host.c#L958-L959

The sd_card example has been updated to split the SPI bus and sdspi initialization:
https://github.com/espressif/esp-idf/blob/f7d1ed5f329c14c356a710c4c10ba50cea277b55/examples/storage/sd_card/main/sd_card_example_main.c#L105-L126

lovyan03, 2020-03-09:
Thank you !! and sorry…
I could not check the latest repository.
I'll update it and check it.

closed 2020-03-09 by lovyan03

WangYifa, 2020-09-20:
@lovyan03 Hi I'm using M5Stack. Can you provide a demo code that coexists with SD card and LCD on the sharing SPI bus base on ESP-IDF?

lovyan03, 2020-09-20:
@WangYifa 
There's nothing to talk about in this thread.