espressif/esp-idf #4908: Problem with resetting UART after SW reset or OTA

Arthedian, 2020-03-08:
## Environment

- Development Kit:     None
- Module or chip used:  ESP32-WROOM-32D|
- IDF version (run ``git describe --tags`` to find it): v4.0
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Fedora
- Using an IDE?: [No|Yes (please give details)]: Clion
- Power Supply:        external 3.3V

## Problem Description

 Sometime after SW reset or OTA some bytes still stuck in UART TX buffer (I am using UART2). If I want to send something after SW reset, than it will send the rest of TX buffer and only part of my message. I noticed the same bug referenced https://github.com/espressif/arduino-esp32/issues/662 , according to that issue, it should be fixed, but I have the same issue right now with newer ESP IDF.

Arthedian, 2020-03-10:
I tested my code more, and I found how to fix it in my code, but still I am not sure, what precisely is happening. I have UART0 (pin 1 and 3) connected to GSM (SIM800L), UART1 (pin 32 and 35) connected to debug print (ESP IDF logging library) and UART2 (26, 27 and rts 25) to RS485 (MAX485). If I switched off the debug print, than everything is fine. Why?

koobest, 2020-03-16:
@Arthedian 
Can you provide the code that reproduces this issue?
thanks

closed 2020-03-31 by espressif-bot