espressif/esp-idf #4909: SPI Example is quite complicated

maludwig, 2020-03-09:
Hey friends,

## Problem

It took me quite a while to get started when I was trying to get SPI up and running. My hope was to port this Arduino library:

https://github.com/bolderflight/MPU9250/blob/master/src/MPU9250.cpp

Over into ESP-IDF native code.

## Current State

There is already an example of how to use SPI here:

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_master

But it's quite complex, and has a lot of LCD-specific code that needs to be disentangled from the actual SPI logic.

## Desired Solution

I'd like for a simpler SPI example to exist. I've made a pretty simple thing, that I could contribute, that just queries the "WHO_AM_I" register on the MPU9255 (which always just returns 115), but I'm not sure if you guys do contributions to your examples, and if so, what standards they should have? The MPU9255 is an end-of-life, odd motion processing unit, not really standard kit. It's not likely someone would just have one lying around. It's also really complicated, though the "WHO_AM_I" register is quite simple.

Alternatively, if you guys could make a simpler SPI example, that would be very awesome.

ginkgm, 2020-04-08:
We are considering this.. You can also check the `sender` in the `spi_slave` example folder, it's also an spi master full-duplex example. We also have another spi master half-duplex mode example, in the spi_master/hd_eeprom folder.

Do you have suggestions for other devices which are quite easy to get, and easy to write examples?

0xjakob, 2021-05-20:
Hi @maludwig!

Are you OK with an SPI example in C++? We are currently working on C++ libraries on top of the IDF C libraries. By coincidence, the example is reading a register from the *MPU9250*.

If you are interested, please let us know!