espressif/esp-idf #4910: commit b393203 "Merge branch 'feature/x509_cert_management" breaks my project

ataweg, 2020-03-09:
Windows 10 pro using the cmd shell ends in

The cryptography package is not installed.Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Traceback (most recent call last):
  File "C:/Espressif/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 35, in <module>
    from cryptography import x509
  File "D:\Users\Axel\AppData\Roaming\Python\Python37\site-packages\cryptography\x509\__init__.py", line 8, in <module>
    from cryptography.x509.base import (
  File "D:\Users\Axel\AppData\Roaming\Python\Python37\site-packages\cryptography\x509\base.py", line 16, in <module>
    from cryptography.x509.extensions import Extension, ExtensionType
  File "D:\Users\Axel\AppData\Roaming\Python\Python37\site-packages\cryptography\x509\extensions.py", line 18, in <module>
    from cryptography.hazmat.primitives import constant_time, serialization
  File "D:\Users\Axel\AppData\Roaming\Python\Python37\site-packages\cryptography\hazmat\primitives\constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ModuleNotFoundError: No module named 'cryptography.hazmat.bindings._constant_time'

the previous commit is ok.

igrr, 2020-03-09:
Can you please try to go to C:\Espressif\esp-idf\ (in your Command Prompt window) and run install.bat to install the missing package?

ataweg, 2020-03-09:
**running install.bat doesn't help.**

1)
**_I use foolowing setting to bild my project:**
cd /D F:\Projects\Carrera124\Firmware\Wifi-ControlUnit\source
set PWD=.
set IDF_PATH=C:\Espressif\esp-idf
PATH=C:\Python37;%PATH%
PATH=d:\users\axel\appdata\roaming\python\python37;%PATH%

PATH=C:\msys32\usr\bin;%PATH%
PATH=C:\Espressif\esptool\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin;%PATH%

cls&make -j 5 flash monitor_

2) 
**edit esp-idf/components/mbedtls/CMakeLists.txt**
_idf_build_get_property(idf_target IDF_TARGET)
idf_build_get_property(python PYTHON)

idf_component_register(SRCS "esp_crt_bundle/esp_crt_bundle.c"
    INCLUDE_DIRS "port/include" "mbedtls/include" "esp_crt_bundle/include"
    REQUIRES lwip_
**--> doesn't help**

3)
**copy d:\users\axel\appdata\roaming\python\python37 to C:\Python37**
**--> doesn't help**

igrr, 2020-03-10:
I see you are not using export.bat to set up the PATH variable, is there a reason to that?

In addition, you are adding what looks like 3 different Python paths, one being from MSYS, and you are not adding the IDF-specific Python virtual environment created by install.bat.

Can you please try using install.bat and export.bat as recommended in the Getting Started guides instead, and don't add Msys to your PATH? The latter frequently causes issues with installation of Python packages which depend on precompiled libraries, such as `cryptography`.

Please also note that in my previous message, the last suggestion was to modify CMakeLists.txt file adding a line of code; not to copy d:\users\axel\appdata\roaming\python\python37 to C:\Python37.

ataweg, 2020-03-10:
I also see it that way, it seems to be a problem with the installation of necessary and missing Python libraries.  In my case, however, msys doesn't matter, because while running "install.bat" with the installation of Python packages, msys is not in the search path. So my question is, why does the IDF-specific virtual Python environment not work together with the installed Python package? I have seen that missing Python libraries are copied to the virtual environment (here d:\users\xxx\appdata\roaming\python\python37). Why not into the appropriate Python folder if a Python is already installed?

One more word on msys. I need it to build my project. As you have surely seen it is in old fashioned make project. Without msys for "make -j 5" the message comes: "C:\Espressif\esp-idf/make/project.mk:114: *** IDF_PATH cannot contain colons. If overriding IDF_PATH on Windows, use MSYS Unix-style /c/dir instead of C:/dir.  Stop."
Changing the IDF_PATH to Unix style "set IDF_PATH=\C\Espressif\esp-idf" leads to:
Makefile:20: \C\Espressif\esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target `\C\Espressif\esp-idf/make/project.mk'.  Stop.

As you can see, the classic "make" with the IDF still has its problems. I can't imagine if all customers will want to switch to Cmake. Therefore an Esp project should continue to work with "make".

ataweg, 2020-03-10:
wrong button pressed, so reopened.

igrr, 2020-03-10:
> One more word on msys. I need it to build my project. As you have surely seen it is in old fashioned make project.

Apologies, no I haven't seen this. Kind request to fill in the default issue template next time you are filing an issue — this ensures that we have enough context to help you.

Are you using the pre-packaged MSYS environment (https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_gcc8_toolchain-20191231.zip)?

ataweg, 2020-03-10:
I got it. Now my project can be build with the head revicsion of esp-idf. I removed my python package and setup my toolchain new again.
What i did:
1) update msys
set PATH=C:\cygwin\bin;%PATH%
C:\msys32\mingw32.exe
1a) in the msys shell:
export IDF_PATH=C:/Espressif/esp-idf
pushd C:/Espressif/esp-idf/tools/windows
./windows_install_prerequisites.sh
echo export IDF_PATH="C:/Espressif/esp-idf"  > C:/msys32/etc/profile.d/export_idf_path.sh
echo export PATH="/C/Python38:$PATH"        >> C:/msys32/etc/profile.d/export_idf_path.sh
cat C:/msys32/etc/profile.d/export_idf_path.sh

2) Install Python 3.8.2 from https://www.python.org/ftp/python/3.8.2/python-3.8.2.exe
Install with precompiled standard library --> C:\python38

3) install missing python packages
python -m pip install -r C:\espressif\esp-idf\requirements.txt

Now I'm satisfied.
Thank you for your support.
Axel

Alvin1Zhang, 2020-03-17:
@ataweg Thanks for reporting and sharing updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-03-17 by Alvin1Zhang