espressif/esp-idf #4911: bug fixing in esp-idf installation suggest

bioBean, 2020-03-10:
when I install the esp-idf on windows, there is bug in python code that can cease the installation:

the python file in "esp-idf-v4.0\tools\idf_tools.py", line 1183, the argument "--no-site-packages" for virtualenv seems like deprecated in latest version of virtualenv, instead they use "--system-site-packages" argument to express an opposite flag for that, which means there is no need for --no-site-package argument because it is now a default option, otherwise it cannot recognize that parameter and it could lead to failing of installation.
[image]

igrr, 2020-03-10:
Hi @bioBean, this has been fixed in commit https://github.com/espressif/esp-idf/commit/7ccf6ebadae432e24d933a5d4090b6f2373cf2d0 on release/v4.0 branch. The fix will be part of the next bugfix release (v4.0.1).

bioBean, 2020-03-28:
Ok, I got it, I think that problem is because I downloaded that idf from official website instead of github, that's it, thanks

closed 2020-03-28 by bioBean