espressif/esp-idf #4912: ESP32 A2DP Sink - PCM5102 only cracks and pops

mbt28, 2020-03-10:
Hi

I am testing A2DP example with ESP-IDF 4.0. The Bluetooth receiving function works, my phone can discover the ESP and streams the music sucessfully. I can also see the incoming audio packets and audio metadata no problem with that.

There are 2 issues, if I dont disable absolute volume control, esp32 changes the volume level randomly. The second issue is, I only hear pops and cracks from the PCM5101. 

I set PCM5101 I2S slave and SCL is grounded so I use only 3 wires for I2S. I tried to change the I2S format as below too but no success with the audio at all. When I change the format I dont hear anything at all. 

communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,

Is there anyway for me to debug I2S hardware? I feel like the issue doesnt come from bluetooth but from I2S buffer, frequency or format. I saw similar issues with the audio but I am not sure if they have the same root causes.

I am using this DAC Board and powering esp32 and pcm5101 with the usb battery: https://www.aliexpress.com/item/33061919226.html?spm=a2g0o.productlist.0.0.603b3bb1WbpD7H&algo_pvid=75fb0aad-7892-4e19-93b1-29a16069da75&algo_expid=75fb0aad-7892-4e19-93b1-29a16069da75-13&btsid=0b0a0ad815838300728938446e0e91&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Thanks

KenWood5020, 2020-04-09:
Dear mbt28,
unfortunately I do not have a solution, but instead I face the same problems. I just wanted to inform you what I found out so far.
1. Random volume change
There is a task running which simulates volume changes. Please see the log in you monitor

> I (760920) RCTG: start volume change simulation

The task is started by function bt_av_hdl_avrc_tg_evt, please see file bt_app_av.c line 316. The task's function is volume_change_simulation (same file, from line 304).
My next step is to simply delete this task and see what's gonna happen.

2. Noise
My setup is different to yours. I do not use I2S, but instead try to run internal DACs, but the result is the same: A lot of noise instead of a clear sound. I spent a lot of time measuring with my limited skills and equipment and found, that wiring and power supply do have a tremendous impact on the noise. I used a 5V power supply connected to mains. And I really could hear mains in my speaker. Using condensators did help, but were not able to suppress the noise in total. It stopped as soon as I changed my power supply. Also, you can hear a Bluetooth connection, if wires pass the ESP to close. To sum it up: Check your wiring carefully.
I also did some experiments with the Arduino IDE and other examples. The sound quality was way better. So there must be also kind of bad configuration in the example. I am still searching...

Farzinkh, 2021-07-27:
Hi any solution yet? I have the same problem too much noises especially when playing pop musics and i am using  i2s protocol.

FedericoBusero, 2021-12-05:
According to https://github.com/espressif/esp-idf/issues/4264#issuecomment-654535425 using I2S_COMM_FORMAT_STAND_I2S (IDF >=4.2) should fix it.

jack0c, 2023-04-12:
Close this since we have https://github.com/espressif/esp-idf/issues/4264 and long time no update on this

closed 2023-04-12 by jack0c