espressif/esp-idf #4913: NIMBLE crashes after reading characteristics as a client

Mair, 2020-03-10:
## Environment

- Development Kit:      ESP32 DOIT (ESP32 WROOM 4MB)

- Kit version: DevKit: v1
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.0-beta2-71-gac74ca653

- Build System:         [CMake|idf.py]
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0 
- Operating System:     [Windows 10]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (vs code)]
- Power Supply:         [USB]

## Problem Description
calling ble_gattc_disc_all_chrs() causes a guru meditation application crash after the characteristic is correctly read

### Expected Behavior
application program counter continues normally

### Actual Behavior
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

### Steps to reproduce

1. In my GAP call back I scan for a device
2. After locating the device I want I Connect to it successfully
3. My GAP connection calls ble_gattc_disc_all_svcs() which works successfully
4. In my discover service callback I call ble_gattc_disc_all_chrs()
5. This throws an exception once it completes running

I have checked heap and stack there appears to be enough memory available 

### Code to reproduce this issue
 below is pseudo code summery 
[click me for full file gist](https://gist.github.com/Mair/4614dbece5b38cf0cf083c9f0642fdf4)

```cpp

static int get_val(uint16_t conn_handle,  const struct ble_gatt_error *error,   const struct ble_gatt_chr *chr, void *arg)
{
    ...
    return 0;
    // crash here
}

static int findPrimaryServices(uint16_t conn_handle, const struct ble_gatt_error *error, const struct ble_gatt_svc *service, void *arg)
{
    if (error->status == 0)
    {
        char buffer[80];
        memset(buffer, 0, sizeof(buffer));
        ble_uuid_to_str(&service->uuid.u, buffer);
        ESP_LOGI("CONNECT", "SERVICE %s found", buffer);
        if (strcmp(buffer, "0x180a") == 0)
        {
            ESP_LOGI("CONNECT", "Found DEVICE_INFO_SERVICE!!!");
            int res = ble_gattc_disc_all_chrs(conn_handle, service->start_handle, service->end_handle, get_val, "NULL");
            ESP_LOGI("deb", "res= %d", res);
        }
    } 
    return 0;
}

static int ble_gap_event(struct ble_gap_event *event, void *arg)
{
    struct ble_hs_adv_fields fields;

    switch (event->type)
    {
    case BLE_GAP_EVENT_DISC:
        ble_hs_adv_parse_fields(&fields, event->disc.data, event->disc.length_data);
     
      if (... found what we were looking for)
        {
            ble_gap_disc_cancel();
            ble_gap_connect(BLE_OWN_ADDR_PUBLIC, &event->disc.addr, 10000, NULL, ble_gap_event, NULL);
        }

        break;

    case BLE_GAP_EVENT_CONNECT:
        if (event->connect.status == 0)
        {
            ble_gap_conn_find(event->connect.conn_handle, NULL);
            ble_gattc_disc_all_svcs(event->connect.conn_handle, findPrimaryServices, NULL);
        }
        break;
       ......
    }
    return 0;
}

```

## Debug Logs

```
GAP procedure initiated: stop advertising.
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates
Discovered device with name: "Blank" Looking for device with name "Blank"
FOUND BLE Device
GAP procedure initiated: connect; peer_addr_type=1 peer_addr=4c:24:bc:85:99:dc scan_itvl=16 scan_window=1n=16 max_ce_len=768 own_addr_type=0
I (957) GAP: BLE_GAP_EVENT_CONNECT OK
I (957) CONNECT: Finding Device
GATT procedure initiated: discover all services
I (1117) CONNECT: SERVICE 0x1800 found
I (1117) CONNECT: SERVICE 0x1801 found
I (1217) CONNECT: SERVICE d0611e78-bbb4-4591-a5f8-487910ae4366 found
I (1367) CONNECT: SERVICE 9fa480e0-4967-4542-9390-d343dc5d04ae found
I (1467) CONNECT: SERVICE 0x180f found
I (1467) CONNECT: SERVICE 0x1805 found
I (1467) CONNECT: SERVICE 0x180a found
I (1467) CONNECT: Found DEVICE_INFO_SERVICE!!!
GATT procedure initiated: discover all characteristics; start_handle=79 end_handle=83
I (1477) deb: res= 0
I (1567) CONNECT: char 0x2a29 found
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d8156  PS      : 0x00060930  A0      : 0x800d8a5d  A1      : 0x3ffc73e0  
0x400d8156: ble_gattc_rx_read_type_adata at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/s

A2      : 0x3ffc0aac  A3      : 0x3ffc7420  A4      : 0x3f400e98  A5      : 0x0000001f  
A6      : 0x00000001  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffc73b0  
A10     : 0x3ffc0aac  A11     : 0x3f403d54  A12     : 0x00000004  A13     : 0x3ffc73e0  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 30ed0dad09291d7a79ee56179fbd51af34019136bd214536c4e48b2b022b71e3

Backtrace: 0x400d8153:0x3ffc73e0 0x400d8a5a:0x3ffc7410 0x400d6a6d:0x3ffc7450 0x400d606a:0x3ffc7480 0x400db:0x3ffc7510 0x4008fc41:0x3ffc7530
0x400d8153: ble_gattc_rx_read_type_adata at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/s

0x400d8a5a: ble_att_clt_rx_read_type at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/b

0x400d6a6d: ble_att_rx at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x400d606a: ble_hs_hci_evt_acl_process at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src

0x400d5611: ble_hs_process_rx_data_queue at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/s

0x400d5623: ble_hs_event_rx_data at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_h

0x400dd602: ble_npl_event_run at C:/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/inc
 (inlined by) nimble_port_run at C:/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimbl

0x400d4ebb: host_task at c:\_esp32\esp32-course\_19_bluetooth\_19_6_gapp_client\build/../main/main.c:129

0x4008fc41: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port.c:143

Rebooting...

```

## Other items if possible

- [x ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ x] coredump 
[debug.zip](https://github.com/espressif/esp-idf/files/4311632/debug.zip)
(This provides stacks of tasks.)

Any help or advice is high;y appreciated

AussieSusan, 2020-03-11:
I had the same problem but I found that it appeared from several lines within ble_gattc.c.
The original line (for me) was 4543 (in the ble_gattc_rx_read_type_complete function) which, with the preceding lines, is:

    proc = BLE_GATTC_RX_EXTRACT_RX_ENTRY(
        conn_handle, ble_gattc_rx_read_type_complete_entries,
        &rx_entry);
    if (proc != NULL) { 
        rc = rx_entry->cb(proc, status);   <==== 4543

Using printf statements, it turns out that 'rx_entry' is being returned as NULL from the BLE_GATTC_RX_EXTRACT_RX_ENTRY macro. This is supported by the fact that the EXCVADDR value is 0x00...4 and the 'cb' item in the struct that rx_entry is supposed to be pointing to has an offset of 4.

By changing the if statement to be 

    if (proc != NULL && rx_entry != NULL) {

in all places where the code pattern appears, the problem goes away and the library *appears* to be working.

As to why this occurs, I can only guess but the example code only seems to have one characteristic for the service whereas the one I was using as 3 characteristics. What seems to happen is that the queue that the macro tries to scan for a matching entry from the first characteristic is altered by the subsequent characteristic and so the macro cannot find a match. However it DOES return a non-NULL 'proc' value but also a NULL 'rx_entry'.

II strongly suggest that my solution is a very poor workaround as I really don't understand how the whole state machine works and there is almost certainly a better solution to be had.

Susan

AussieSusan, 2020-03-11:
Just after entering the above note, I looked at https://github.com/micropython/micropython/issues/5686 which seems to indicate the ther eare some patches available that also address (probably properly!) this issue.
Susan

Mair, 2020-03-12:
Thanks Susan,
Good to know its not just me having this issue

Alvin1Zhang, 2020-03-13:
@Mair @AussieSusan Thanks for reporting and sharing the updates. We will look into. @prasad-alatkar

AussieSusan, 2020-03-13:
In the other forum I linked to (above) I have reported that the issue still exists for me even after using the patched version they mention.
Susan

dhrishi, 2020-03-18:
Hi @Mair @AussieSusan  If you want to stop the service discovery when you find the required service then you should return `BLE_HS_EDONE` (instead of 0) from the service discovery callback  (findPrimaryServices). I have tried this and it works as expected.

Apart from this, I agree that there shouldn't be any crash while discovering the GATT database in any way. We will check it and fix it. Thanks!

AussieSusan, 2020-03-19:
@dhrishi - I must admit that I can't find where you have set BLE_HS_EDONE in the code you link to.
Regardless, I don't see that this is my problem. In my case I have several services within each of the (multiple) devices that I may find in a scan. Therefore I need to find ALL of the services in a device. For EACH service I need to read the characteristic value and it is reading that characteristic (whether after finding one service or all of them and working through them one at a time later) that the crash occurs.
Susan

dhrishi, 2020-03-19:
@AussieSusan It is the code provided by @Mair and I have just marked the location where he needs to return BLE_HS_EDONE. The solution is in the context of the current issue where the characteristic discovery is started immediately when a particular service (0x180a in this case) is found.

Mair, 2020-03-19:
@dhrishi Thanks mate,
That function now works however, moving to the next level where I try to read the attribute I get a very similar crash
```c++

int attr_read(uint16_t conn_handle,  const struct ble_gatt_error *error,   struct ble_gatt_attr *attr, void *arg)
{
    if (error->status == 0)
    {
        printf("printing val\n");
        //printf("value was %.*s\n", attr->om->om_len, attr->om->om_data);
        //return BLE_HS_EDONE;
    }
    return 0;
}

static int get_val(uint16_t conn_handle,  const struct ble_gatt_error *error,   const struct ble_gatt_chr *chr, void *arg)
{
    if (error->status == 0)
    {
        char buffer[80];
        memset(buffer, 0, sizeof(buffer));
        ble_uuid_to_str(&chr->uuid.u, buffer);
        ESP_LOGI("CONNECT", "char %s found", buffer);
        if(strcmp(buffer,"0x2a29") == 0)
        {
            printf("got name\n");
            ble_gattc_read(conn_handle, chr->val_handle,  attr_read, NULL);
            return BLE_HS_EDONE;
        }
    } 
    return 0;
}

static int findPrimaryServices(uint16_t conn_handle, const struct ble_gatt_error *error, const struct ble_gatt_svc *service, void *arg)
{
    if (error->status == 0)
    {
        char buffer[80];
        memset(buffer, 0, sizeof(buffer));
        ble_uuid_to_str(&service->uuid.u, buffer);
        ESP_LOGI("CONNECT", "SERVICE %s found", buffer);
        if (strcmp(buffer, "0x180a") == 0)
        {
            ESP_LOGI("CONNECT", "Found DEVICE_INFO_SERVICE!!!");
            int res = ble_gattc_disc_all_chrs(conn_handle, service->start_handle, service->end_handle, get_val, NULL);
            ESP_LOGI("deb", "res= %d", res);
            return BLE_HS_EDONE;
        }
    } 
    return 0;
}

```

 (971) GAP: BLE_GAP_EVENT_CONNECT OK
I (971) CONNECT: Finding Device
GATT procedure initiated: discover all services
I (1071) CONNECT: SERVICE 0x1800 found
I (1071) CONNECT: SERVICE 0x1801 found
I (1171) CONNECT: SERVICE d0611e78-bbb4-4591-a5f8-487910ae4366 found
I (1321) CONNECT: SERVICE 9fa480e0-4967-4542-9390-d343dc5d04ae found
I (1421) CONNECT: SERVICE 0x180f found
I (1421) CONNECT: SERVICE 0x1805 found
I (1421) CONNECT: SERVICE 0x180a found
I (1431) CONNECT: Found DEVICE_INFO_SERVICE!!!
GATT procedure initiated: discover all characteristics; start_handle=79 end_handle=83
I (1441) deb: res= 0
I (1521) CONNECT: char 0x2a29 found
got name
GATT procedure initiated: read; att_handle=81
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d821e  PS      : 0x00060930  A0      : 0x800d8b25  A1      : 0x3ffc7820  
0x400d821e: ble_gattc_rx_read_type_adata at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4519

A2      : 0x3ffc0eec  A3      : 0x3ffc7860  A4      : 0x3f400ed8  A5      : 0x0000001f  
A6      : 0x00000001  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffc77f0  
A10     : 0x3ffc0eec  A11     : 0x3f403db0  A12     : 0x00000004  A13     : 0x3ffc7820  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 7e93c0a91a84c20c9fe6813f62b03c982d793006a3fc66fc4ff376f15e3d2253

Backtrace: 0x400d821b:0x3ffc7820 0x400d8b22:0x3ffc7850 0x400d6acd:0x3ffc7890 0x400d60ca:0x3ffc78c0 0x400d5671:0x3ffc78f0 0x400d5683:0x3ffc7910 0x400dd6ca:0x3ffc7930 0x400d4eeb:0x3ffc7950 0x40090385:0x3ffc7970
0x400d821b: ble_gattc_rx_read_type_adata at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4518

0x400d8b22: ble_att_clt_rx_read_type at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:448

0x400d6acd: ble_att_rx at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x400d60ca: ble_hs_hci_evt_acl_process at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:870

0x400d5671: ble_hs_process_rx_data_queue at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x400d5683: ble_hs_event_rx_data at C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:539

0x400dd6ca: ble_npl_event_run at C:/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121
 (inlined by) nimble_port_run at C:/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:82

0x400d4eeb: host_task at c:\_esp32\esp32-course\_19_bluetooth\_19_6_gapp_client\build/../main/main.c:149

0x40090385: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port.c:143

dhrishi, 2020-03-19:
@Mair Thanks for the feedback. I see that in your code you are calling `ble_gattc_read` as soon as the characteristic is found. So, what you are trying to do here is to perform an ATT operation before the previous one is complete. This is not allowed as per the specification.

The cleaner and the right approach is, you should complete the entire GATT database discovery and then perform the read/write operation. You can save the value handle (and other required parameters) of the characteristic of your interest. See https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/blecent for more details

If you want to continue with your approach, once the characteristic is found, you can send a message to some other thread (say main) with the details of read and then return BLE_HS_EDONE. Your other thread, waits on a message and when it receives one, it performs the read operation `ble_gattc_read`

Mair, 2020-03-23:
Hi @dhrishi I got it working. Thanks for the tip
```c++

int attr_read(uint16_t conn_handle, const struct ble_gatt_error *error, struct ble_gatt_attr *attr, void *arg)
{
    printf("error status %d\n",error->status);
    if (error->status == 0)
    {
        printf("printing val\n");
        printf("value was %.*s\n", attr->om->om_len, attr->om->om_data);
        return BLE_HS_EDONE;
    }
    return 0;
}

static int get_val(uint16_t conn_handle, const struct ble_gatt_error *error, const struct ble_gatt_chr *chr, void *arg)
{
    char buffer[80];
    static uint16_t val_handle;

    switch (error->status)
    {
    case BLE_HS_EDONE:
        printf("got here\n");
        ble_gattc_read(conn_handle, val_handle, attr_read, NULL);
        break;
    case 0:
        memset(buffer, 0, sizeof(buffer));
        ble_uuid_to_str(&chr->uuid.u, buffer);
        ESP_LOGI("CONNECT", "char %s found", buffer);
        if (strcmp(buffer, "e05bbe04-e5c5-4d0e-97ec-5784e82a3f44") == 0)
        {
            printf("got name\n");
            val_handle = chr->val_handle;
        }
    }
    return 0;
}

static int findPrimaryServices(uint16_t conn_handle, const struct ble_gatt_error *error, const struct ble_gatt_svc *service, void *arg)
{
    char buffer[80];
    static uint16_t start = 0, end = 0;
    switch (error->status)
    {
    case BLE_HS_EDONE:

        ble_gattc_disc_all_chrs(conn_handle, start, end, get_val, NULL);
        break;
    case 0:
        memset(buffer, 0, sizeof(buffer));
        ble_uuid_to_str(&service->uuid.u, buffer);
        ESP_LOGI("CONNECT", "SERVICE %s found", buffer);
        if (strcmp(buffer, "46965d7c-756f-4f07-910e-fb1b2d616482") == 0)
        {
            ESP_LOGI("CONNECT", "Found DEVICE_INFO_SERVICE!!!");
            start = service->start_handle;
            end = service->end_handle;
        }
        break;

    default:
        break;
    }

    return 0;
}
```

closed 2020-03-23 by Mair

agners, 2020-04-12:
I hit the same issue @AussieSusan sees, I also while trying to use the same BLE sensor using MicroPython (see https://github.com/micropython/micropython/issues/5686).

The hint that the service discovery need to be stopped before starting the characteristic discovery was indeed helpful. Unfortunately, in MicroPython it is not possible to return `BLE_HS_EDONE` (the API does not allow it). Also is the status not transmitted to Python code, so user code currently has no way to know when service discovery finished. Currently I had to implement a work around which waits a while until characteristic discovery is triggered.

@dhrishi you noted
> Apart from this, I agree that there shouldn't be any crash while discovering the GATT database in any way. We will check it and fix it. Thanks!

Is this still planned? Should there maybe a new ticket for this/this ticket kept open?

dhrishi, 2020-04-13:
Hi @agners We plan to do that, but that's not on priority list. I would suggest that, in this case, you complete the service/characteristic discovery and save the details of attributes of your interest. Once done, you can use the same for GATT read/write operations

agners, 2020-04-13:
@dhrishi thanks for your answer. From what I see if `status` is set to `BLE_HS_EDONE` the service/characteristic discovery can be determined as being done. Unfortunately, this is (currently) not available from MicroPython. Do you happen to know if there another way to determine whether service/characteristic discovery is done?

dhrishi, 2020-04-13:
@agners 

>  Also is the status not transmitted to Python code, so user code currently has no way to know when service discovery finished. 

Oops, sorry for missing this in your previous reply. 
Unfortunately, if that's not propogated by the micropython code, I think that needs to be fixed. Apart from this, I don't see a way out here.

dhrishi, 2020-04-22:
@agners I think this could be done by directly using APIs `ble_gattc_disc_svc_by_uuid` and `ble_gattc_disc_chrs_by_uuid` to discover the service and characteristic of your interest respectively instead of discovering the entire GATT tree. If you are still at it, you might want to give it a try :-)

agners, 2020-04-22:
@dhrishi hm, quickly checked, both are not available from MicroPython as of now. But since those are rather straight forward APIs of the standard Nimble API and seem like really useful/sensible, I can try adding them. Thanks for the hint!

agners, 2020-04-26:
@dhrishi I did implement those two API's today in MicroPython and made use of them (see https://github.com/micropython/micropython/pull/5976). It works perfectly for service, e.g. I do get that single IRQ and then immediately start characteristic discovery. However, even though I use `ble_gattc_disc_chrs_by_uuid`, I do see crashes in the following read. Only when inserting a delay of several 100th of milliseconds things start to get stable (200 seems to be on the edge, with that it crashes every ~5 read).

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40165694  PS      : 0x00060930  A0      : 0x8016f6fb  A1      : 0x3ffd21b0
A2      : 0x3ffc57c0  A3      : 0x00000000  A4      : 0x3f42e359  A5      : 0x3f4028e9
A6      : 0x3f433af2  A7      : 0x3f433b58  A8      : 0x00000000  A9      : 0x3ffd2170
A10     : 0x3ffc57c0  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffd21b0
A14     : 0x00000003  A15     : 0x3ffc3a6c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x40097474  LEND    : 0x400974a2  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
```

Any idea what that might be?

agners, 2020-04-27:
@dhrishi another question which came up here you might know: Is using this API generating less traffic on air? E.g. is this a host only filtering or is this really something which generates less requests? (see also https://github.com/micropython/micropython/pull/5976).