espressif/esp-idf #4914: esp32ulp_mapgen.py fails to run on ESP-IDF 4.0

jgcjoec, 2020-03-10:
## Environment

- Development Kit:      HiLetgo ESP32
- Module or chip used:  ESP32-WROOM-32
- IDF version 4.0
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I am trying to compile a ULP program from GitHub and I receive the following error towards the end of the build process:
cmd.exe /C "cd /D C:\Temp\ulp-i2c\build\esp-idf\main\ulp_main && C:\Temp\ESP32\Download\components\ulp\esp32ulp_mapgen.py -s ulp_main.sym -o ulp_main"
Usage: esp32ulp_mapgen.py [options]

This application generates .h and .ld files for symbols defined in input file.
The input symbols file can be generated using nm utility like this: esp32-ulp-
nm -g -f posix <elf_file> > <symbols_file>

Options:
  -h, --help            show this help message and exit
  -s SYMFILE, --symfile=SYMFILE
                        symbols file name
  -o OUTFILE, --outputfile=OUTFILE
                        destination .h and .ld files name prefix
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

I added the following print statement to esp32ulp_mapgen.py just below line 39:
print(options)

Re-running idf.py build then produced this:
cmd.exe /C "cd /D C:\Temp\ulp-i2c\build\esp-idf\main\ulp_main && C:\Temp\ESP32\Download\components\ulp\esp32ulp_mapgen.py -s ulp_main.sym -o ulp_main"
{'symfile': None, 'outputfile': None}
Usage: esp32ulp_mapgen.py [options]

This application generates .h and .ld files for symbols defined in input file.
The input symbols file can be generated using nm utility like this: esp32-ulp-
nm -g -f posix <elf_file> > <symbols_file>

Options:
  -h, --help            show this help message and exit
  -s SYMFILE, --symfile=SYMFILE
                        symbols file name
  -o OUTFILE, --outputfile=OUTFILE
                        destination .h and .ld files name prefix
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Note that "symfile" and "outputfile" are both "None", even though those arguments are clearly supplied. Maybe a problem with Python?

### Expected Behavior
esp32ulp_mapgen.py should run successfully

### Actual Behavior
esp32ulp_mapgen.py is exiting with code 1

### Steps to reproduce
1. Install ESP-IDF v4.0, allowing it to install Git, Python, etc
2. Compile a ULP program. I am trying to compile https://github.com/xlfe/ulp-i2c... I compared the makefile, component.mk and cmakelists.txt against those in ulp_adc in the ESP-IDF examples directory and adjusted as needed. The same error might occur out-of-the-box with ulp_adc.
3. Observe error message from esp32ulp_mapgen.py

jgcjoec, 2020-03-10:
I confirmed the same behavior occurs when attempting to compile the "ulp_adc" that ships with ESP-IDF v4.0:

[1/7] Generating esp32.ulp.ld
[2/7] Generating adc.ulp.S
[3/7] Building ASM object CMakeFiles/ulp_main.dir/adc.ulp.S.obj
[4/7] Linking ASM executable ulp_main
[5/7] Generating ulp_main.bin
[6/7] Generating ulp_main.sym
[7/7] Generating ulp_main.ld, ulp_main.h
FAILED: ulp_main.ld ulp_main.h
cmd.exe /C "cd /D C:\Temp\ESP32\Download\examples\system\ulp_adc\build\esp-idf\main\ulp_main && C:\Temp\ESP32\Download\components\ulp\esp32ulp_mapgen.py -s ulp_main.sym -o ulp_main"
Usage: esp32ulp_mapgen.py [options]

This application generates .h and .ld files for symbols defined in input file.
The input symbols file can be generated using nm utility like this: esp32-ulp-
nm -g -f posix <elf_file> > <symbols_file>

Options:
  -h, --help            show this help message and exit
  -s SYMFILE, --symfile=SYMFILE
                        symbols file name
  -o OUTFILE, --outputfile=OUTFILE
                        destination .h and .ld files name prefix
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

jgcjoec, 2020-03-10:
I created a sample python script to better demonstrate this behavior:
`
#!/usr/bin/env python

from optparse import OptionParser

def main():
    parser = OptionParser()
    parser.add_option("-s", "--symfile", dest="symfile",
                      help="symbols file name", metavar="SYMFILE")
    parser.add_option("-o", "--outputfile", dest="outputfile",
                      help="destination .h and .ld files name prefix", metavar="OUTFILE")

    (options, args) = parser.parse_args()
    print(options)
    print(args)
 
    return 0

if __name__ == "__main__":
    exit(main())
`

jgcjoec, 2020-03-10:
On a whim, I decided to try something different... when the call to foo.py is prefixed by "python", it works... I'm lost...

[image]

igrr, 2020-03-10:
Hi @jgcjoec, on release/v4.0 branch this issue was fixed in commit https://github.com/espressif/esp-idf/commit/a8fe083017057b6c2dbd76ef5fd060b16ff06790, the fix will be part of the next bugfix release (v4.0.1).

jgcjoec, 2020-03-10:
Ok, thank you!

Alvin1Zhang, 2020-05-25:
@jgcjoec IDF v4.0.1 has been released, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-25 by Alvin1Zhang

Alvin1Zhang, 2020-05-27:
@jgcjoec Thanks for sharing the updates and glad to hear the issue is resolved. Thanks.