espressif/esp-idf #4915: I (39) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader

mattkefford, 2020-03-11:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v4.1
- Module or chip used: ESP32-WROVER
- IDF version: v4.0
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ?
- Using an IDE?: Yes Visual Studio + VisualGDB
- Power Supply: USB

## Problem Description
Recently switched from IDF v3.3.1 to v4.0.
The code compiles and flashes and boots fine, but during boot log I now see the message: 
"I (39) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader"

Previously this has always managed to pull in the version of IDF being used. 

I have checked my paths and they are correct:
[image]

Anybody have any clue how to fix this? I've looked in the project properties within visual studio and cannot see any other place to specify the tools or IDF path. 

## Debug Logs
[image]

igrr, 2020-03-17:
@mattkefford It's a bit hard to troubleshoot this, since I'm not familiar with VisualGDB internals, but can you check if VisualGDB makes `git` available on the `PATH` when invoking `cmake`? This error may happen if `git` is not present on the PATH.

Alvin1Zhang, 2020-05-13:
@mattkefford Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

projectgus, 2020-06-03:
Closing due to a lack of updates.

closed 2020-06-03 by projectgus

mattkefford, 2020-06-03:
My apologies. As you may be aware, there is currently a global pandemic occurring, I have since lost my job. I no longer have access to this build environment or the computer I was using and I am no longer developing for the ESP32.

Thanks for your efforts to debug the issue. I hope it serves as useful for others.

projectgus, 2020-06-04:
Hi @mattkefford,

Very sorry to hear that. Wish you the best in this tough time, hope you find a new role soon.

Angus