espressif/esp-idf #4917: NVS_get_i32 returns wrong value

kaedros, 2020-03-11:
Hello,

I'm having some trouble with NVS. Basically, I try read some variables to it 
that does not yet exist and, in this case, this variable would be created with a certain value.

To make things easier, I have created a [repository ](https://github.com/kaedros/ESP32_Bug_NVS_Get) with a demo of the problem and all description.

What I have tried so far:

Checked partition sizes and alignments
Checked the NVS read and write process and it seems to be correct
Removed from the project all code that is not related to the NVS

igrr, 2020-03-11:
Since the program starts running on the board immediately after flashing, I think by the time you open `monitor`, the value is already set. Do you think this explains the observed behavior?
Perhaps you can modify the program to wait until a key press, before testing NVS behavior. Then it will not have a chance to set the value between the time it is flashed and the time the monitor runs.

kaedros, 2020-03-11:
Hello @igrr !

You are completely right.
"make monitor" perform a reset on ESP32 causing to run a second time.

There is a command "simple_monitor" that does not perform the reset, it only opens the serial port, but if used in sequence to the "make flash", there is usually no way to interrupt the monitor.

Thank You!

Alvin1Zhang, 2020-03-12:
@kaedros Thanks for reporting. Has your issue been resolved? Could this issue be closed? Thanks.

kaedros, 2020-03-12:
Yes, thank you!

closed 2020-03-12 by kaedros