espressif/esp-idf #4918: 'mem_start' may be used uninitialized in esp_bt_controller_mem_release

uname2219, 2020-03-11:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-2079-g5dbabae9d
- Build System:         make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     linux
- Using an IDE?: No
- Power Supply: usb

## Problem Description

I attempted to build a project that includes Bluetooth with performance optimizations: 
```
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BT_ENABLED=y
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BTDM=y
```
When I make the project, the build fails inside components/bt/controller/bt.c with:
```
/home/user/Development/esp32/esp-idf-v4.1/components/bt/controller/bt.c: In function 'esp_bt_controller_mem_release':
/home/user/Development/esp32/esp-idf-v4.1/components/bt/controller/bt.c:961:15: error: 'mem_start' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     int ret = heap_caps_add_region(start, end);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Development/esp32/esp-idf-v4.1/components/bt/controller/bt.c:975:14: note: 'mem_start' was declared here
     intptr_t mem_start, mem_end;
              ^~~~~~~~~
cc1: some warnings being treated as errors
```

### Expected Behavior

Perhaps initializing mem_start to NULL so the compiler can detect it has been initialized? Then the rest of the build can continue. Under default optimizations the project builds without error.

Alvin1Zhang, 2020-03-12:
@uname2219 Thanks for reporting.

AbnerFederer, 2020-03-12:
Please apply this patch. 
[optimization_compiler_fix.zip](https://github.com/espressif/esp-idf/files/4323078/optimization_compiler_fix.zip)

uname2219, 2020-03-12:
Thank you, with this patch the project builds and runs. I look forward to seeing this merged in.

Alvin1Zhang, 2020-03-13:
@uname2219 Thanks for reporting and updates. We have a fix internally and under reviewing. Will update once the fix gets synchronized into GitHub. Thanks.

Alvin1Zhang, 2020-03-17:
@uname2219 Thanks for reporting, the fix has been available on GitHub. Thanks.

closed 2020-03-17 by Alvin1Zhang

ROYMURLI, 2020-04-16:
Facing same issue. How to apply this patch. Guide me. 
OS- Windows using (msys32)
IDF- v4.1-dev-1086-g93a8603c5

AbnerFederer, 2020-04-20:
@ROYMURLI 

`git apply [route to patch]`