espressif/esp-idf #4919: First i2c read (after write) returns 0???...=??, nth read return previous (n-1)

MrCavemen, 2020-03-11:
## Environment

- Development Kit:        none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0
- Build System:         idf.py
- Compiler version:    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

I've connected 2 esp32's using i2c, master and slave.
I want to do a combined write and read from the master.
(I couldn't figure out how to make this work by reading registers, so I'm just reading the buffer.)

The master sends some messages the slave should ignore.
Then the master sends a message for which he should "reply" (by storing the reply in its buffer)
The master should then read the message stored in the buffer.

However, the first time the master tries to read the buffer, he gets O????????????????????????????????=??(+original string send by master??), instead the expected reply.
The second time, the master reads the reply from the first command, instead of the second, and so on.

I'm guessing that the master does not wait until there is an actual value in the buffer.
However, I have no explanation to why/how the original string send by the master is attached to the reply, as they should be in different ringbuffers...

### Expected Behavior

Master should reads the correct value from the buffer (once stored) when requested (instead of one cycle later).

Expected example:
Master writes: IGNORE THIS 0001
Master writes: IGNORE THIS 0002
Master writes: WHATS MY NUMBER? 0003
Master reads: 0003
Master writes: IGNORE THIS 0004
Master writes: IGNORE THIS 0005
Master writes: WHATS MY NUMBER? 0006
Master reads: 0006
...

### Actual Behavior

Master reads strange things the first time, then it reads the previous reply.

Actual behavior:
Master writes: IGNORE THIS 0001
Master writes: IGNORE THIS 0002
Master writes: WHATS MY NUMBER? 0003
Master reads: **O????????????????????????????????=??WHATS MY NUMBER? 0003**
Master writes: IGNORE THIS 0004
Master writes: IGNORE THIS 0005
Master writes: WHATS MY NUMBER? 0006
Master reads: **0003**
Master writes: IGNORE THIS 0007
Master writes: IGNORE THIS 0008
Master writes: WHATS MY NUMBER? 0009
Master reads: **0006**
...

### Code to reproduce this issue

I've removed the ret checks, but all are ESP_OK.
This "works" when writing and sending in different command (with stop cmd before restart), and waiting 1 second between them.
But this would defeat the purpose as we want to guarantee that the response is the first item in the buffer and that no other cmd would be read in between them (SCL line should thus need to stay low, therefor I need to do this in one cmd and without the stop. I want to use this in a multi-master setup.)

master:
```cpp
#define ACK_CHECK_EN                0x1
#define ACK_CHECK_DIS               0x0
#define ACK_VAL                     0x0
#define NACK_VAL                    0x1

void i2c_master_init() {
    i2c_config_t conf;
    conf.mode               = I2C_MODE_MASTER;
    conf.sda_io_num         = I2C_MASTER_SDA_IO;
    conf.sda_pullup_en      = GPIO_PULLUP_DISABLE;
    conf.scl_io_num         = I2C_MASTER_SCL_IO;
    conf.scl_pullup_en      = GPIO_PULLUP_DISABLE;
    conf.master.clk_speed   = I2C_MASTER_FREQ_HZ;
    esp_err_t ret = i2c_param_config(I2C_MASTER_NUM, &conf);
    ret = i2c_driver_install(I2C_MASTER_NUM, conf.mode, 0, 0, 0);
}

void i2c_master_read_slave(const uint8_t slave_add, uint8_t *data_wr, uint8_t* data_rd, size_t size_rd, bool local) {
    printf("Master writes: %s\n", data_wr);
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( slave_add << 1 ) | I2C_MASTER_WRITE, ACK_CHECK_EN);
    i2c_master_write(cmd, data_wr, RW_TEST_LENGTH, ACK_CHECK_EN);
    i2c_master_start(cmd);  // send restart signal (yes, no stop before here)
    i2c_master_write_byte(cmd, ( slave_add << 1 ) | I2C_MASTER_READ, ACK_CHECK_EN);
    i2c_master_read(cmd, data_rd, size_rd, ACK_VAL); // no extra i2c_master_read with size_rd -1 as this is already in the i2c_master_read function
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_MASTER_NUM, cmd, TICKS_TO_WAIT);
    i2c_cmd_link_delete(cmd);
}

void app_main() {
    i2c_master_init();

    char command_1[] = "IGNORE THIS 0000";
    char command_2[] = "WHATS MY NUMBER? 0000";
    uint8_t *data_rd = (uint8_t*) calloc(RW_TEST_LENGTH, sizeof(uint8_t));
    uint8_t *data_wr = (uint8_t*) calloc(RW_TEST_LENGTH, sizeof(uint8_t));

    for (unsigned int nr = 0;;) {
        for (int i = 0; i < 5; ++i) {
            nr = (nr+1) % 10000;
            sprintf(command_1+sizeof(command_1)-5, "%.4u", nr);
            memset(data_wr, 0, RW_TEST_LENGTH);        // ensuring data_wr is clean
            memcpy(data_wr, command_1, sizeof(command_1));  // setting cmd in it
            write_to_control_unit(data_wr);             // write to control unit
        }
        nr = (nr+1) % 10000;
        sprintf(&command_2[sizeof(command_2)-5], "%.4u", nr);
        memset(data_wr, 0, RW_TEST_LENGTH*sizeof(uint8_t));
        memcpy(data_wr, command_2, sizeof(command_2));
        read_from_control_unit(data_wr, data_rd, RW_TEST_LENGTH);   // write to CU and read response
        
        printf("Master reads: %s\n", data_rd);
    }
}

```

slave:
```cpp
void i2c_slave_init() {
    i2c_config_t conf_slave;
    conf_slave.sda_io_num       = I2C_SLAVE_SDA_IO;
    conf_slave.sda_pullup_en    = GPIO_PULLUP_DISABLE;
    conf_slave.scl_io_num       = I2C_SLAVE_SCL_IO;
    conf_slave.scl_pullup_en    = GPIO_PULLUP_DISABLE;
    conf_slave.mode             = I2C_MODE_SLAVE;
    conf_slave.slave.addr_10bit_en = 0;
    conf_slave.slave.slave_addr = ESP_SLAVE_ADDR;
    esp_err_t ret = i2c_param_config(I2C_SLAVE_NUM, &conf_slave);
    i2c_driver_install(I2C_SLAVE_NUM, conf_slave.mode, 1024, 1024, 0);
}

void app_main() {
    i2c_slave_init();

    char command_2[] = "WHATS MY NUMBER? ";

    uint8_t *data_rd = (uint8_t *) calloc(RW_TEST_LENGTH, sizeof(uint8_t));
    uint8_t *data_wr = (uint8_t *) calloc(RW_TEST_LENGTH, sizeof(uint8_t));

    for (;;) {
        int size_rd = i2c_slave_read_buffer(I2C_SLAVE_NUM, data_rd, RW_TEST_LENGTH, portMAX_DELAY);
        printf("SLAVE READ [%i] bytes: %s\n", size_rd, data_rd);
        
        bool cmd_found = true;
        for (i = 0; i < strlen(command_2); ++i) {
            if (data_rd[i] != command_2[i]) {
                cmd_found = false;
                break;
            }
        }
        if (cmd_found) {
            memset(data_wr, 0, RW_TEST_LENGTH); // ensuring data_wr is clean
            memcpy(data_wr, data_rd+i, 4);    // setting XXXX nr            
            size_t d_size = i2c_slave_write_buffer(I2C_SLAVE_NUM, data_wr, RW_TEST_LENGTH, TICKS_TO_WAIT);            
        }
    }
}
```

koobest, 2020-03-12:
Hi, @MrCavemen
The slave does not support SCL stretch and the current driver do not support `restart read` operation (start+write+write+reastart+write+read+stop)  in slave mode.  If the application does not write data to the tx ringbuffer in advance(calling i2c_slave_write_buffer), some garbage values will be read by the master. 

In your example,  when the master sends "WHATS MY NUMBER?"  message to the slave, the slave receives and stores it in the hardware rxfifo.  then the master read the slave, but the application  dose not feed the txbuffer  in advance,  some gaebrg value will be read.

thanks!!

Alvin1Zhang, 2020-03-25:
@MrCavemen Thanks for reporting. Please help check the notes by @koobest , will close for now, thanks.

closed 2020-03-25 by Alvin1Zhang