espressif/esp-idf #4920: vTaskDelete(NULL) and eTaskGetState()

nopnop2002, 2020-03-11:
Calling vTaskDelete(NULL) will temporarily change the status to eDeleted, then return to eReady.
Is this correct behavior? ?

Quote from your documentation:
```
INCLUDE_vTaskDelete must be defined as 1 for this function to be available. See the configuration section for more information.
```
But there is no configuration section in your document.

My Environment:
```
freeRTOS version:V8.2.0
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.1-dev-1935-g647cb628a
```

My Code:
```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/ringbuf.h"
#include "esp_system.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

static void test_task(void *pvParameters)
{
        ESP_LOGI(pcTaskGetTaskName(0), "Start");
        for(int i=0;i<10;i++) {
                vTaskDelay(100);
        }
        ESP_LOGI(pcTaskGetTaskName(0), "Finish");
        vTaskDelete(NULL);
}

void app_main(void)
{
        TaskHandle_t xHandle = NULL;
        xTaskCreate(&test_task, "TEST", 8192, NULL, 3, &xHandle);
        while(1) {
                eTaskState taskState = eTaskGetState( xHandle );
                ESP_LOGI(TAG, "taskState=%d",taskState);
                vTaskDelay(1000 / portTICK_PERIOD_MS);
        }

}
```

Result:
```
I (0) cpu_start: App cpu up.
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (252) heap_init: At 3FFB2930 len 0002D6D0 (181 KiB): DRAM
I (258) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (264) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (271) heap_init: At 40089BAC len 00016454 (89 KiB): IRAM
I (277) cpu_start: Pro cpu start user code
I (295) spi_flash: detected chip: generic
I (296) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (306) MAIN: taskState=1
I (306) TEST: Start
I (1306) MAIN: taskState=2
I (2306) MAIN: taskState=2
I (3306) MAIN: taskState=2
I (4306) MAIN: taskState=2
I (5306) MAIN: taskState=2
I (6306) MAIN: taskState=2
I (7306) MAIN: taskState=2
I (8306) MAIN: taskState=2
I (9306) MAIN: taskState=2
I (10306) TEST: Finish
I (10306) MAIN: taskState=4
I (11306) MAIN: taskState=1
I (12306) MAIN: taskState=1
I (13306) MAIN: taskState=1
I (14306) MAIN: taskState=1
```

Alvin1Zhang, 2020-03-12:
@nopnop2002 Thanks for reporting.

igrr, 2020-03-12:
Checking task status after a task is deleted might be valid for a task with statically allocated TCB. But in your case the TCB is originally allocated from the heap, so the memory is released back to the heap when the task is deleted. By passing the task handle (which in FreeRTOS is a pointer to TCB) to the eTaskGetState function, you are essentially causing a "use after free", reading some value from a now-deallocated heap block. This behavior is the same in upstream FreeRTOS.

If you want to get notified that a task is deleted, a more reliable alternative is to register a task deletion callback.

nopnop2002, 2020-03-12:
@igrr
Thank you for response.

Which API is used to register the task deletion callback?

BTW I was able to know that the task was gone by the following code.

This is my code:
```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/ringbuf.h"
#include "esp_system.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

static void test_task(void *pvParameters)
{
                ESP_LOGI(pcTaskGetTaskName(0), "Start");
                for(int i=0;i<5;i++) {
                                vTaskDelay(100);
                }
                ESP_LOGI(pcTaskGetTaskName(0), "Finish");
                vTaskDelete(NULL);
}

void app_main(void)
{
                TaskHandle_t xHandle = NULL;
                xTaskCreate(&test_task, "TEST", 8192, NULL, 3, &xHandle);

                UBaseType_t uxArraySize = uxTaskGetNumberOfTasks();
                ESP_LOGI(TAG, "uxArraySize=%d", uxArraySize);
                TaskStatus_t *pxTaskStatusArray;
                pxTaskStatusArray = pvPortMalloc( uxArraySize * sizeof( TaskStatus_t ) );
                if( pxTaskStatusArray == NULL ) {
                                ESP_LOGE(TAG, "pvPortMalloc fail");
                }
                while(1) {
                        eTaskState taskState = eTaskGetState( xHandle );
                        ESP_LOGI(TAG, "taskState=%d",taskState);

                        /* Generate raw status information about each task. */
                        uint32_t ulTotalRunTime;
                        uxArraySize = uxTaskGetSystemState( pxTaskStatusArray, uxArraySize, &ulTotalRunTime );
                        for( int x = 0; x < uxArraySize; x++ ) {
                           ESP_LOGI(TAG, "pcTaskName=%s eCurrentState=%d xTaskNumber=%lu, uxCurrentPriority=%lu",
                                        pxTaskStatusArray[ x ].pcTaskName,
                                        pxTaskStatusArray[ x ].eCurrentState,
                                        (unsigned long)pxTaskStatusArray[ x ].xTaskNumber,
                                        (unsigned long)pxTaskStatusArray[ x ].uxCurrentPriority);
                        }

                        vTaskDelay(1000 / portTICK_PERIOD_MS);
                }

}
```

Result:
When the TEST task ends, the TEST task entry disappears from the TCB.

```
I (0) cpu_start: App cpu up.
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (252) heap_init: At 3FFB2930 len 0002D6D0 (181 KiB): DRAM
[... 65 lines cut ...]
I (6636) MAIN: taskState=1
I (6636) MAIN: pcTaskName=main eCurrentState=1 xTaskNumber=5, uxCurrentPriority=1
I (6636) MAIN: pcTaskName=IDLE1 eCurrentState=1 xTaskNumber=7, uxCurrentPriority=0
I (6636) MAIN: pcTaskName=IDLE0 eCurrentState=1 xTaskNumber=6, uxCurrentPriority=0
I (6646) MAIN: pcTaskName=Tmr Svc eCurrentState=2 xTaskNumber=8, uxCurrentPriority=1
I (6656) MAIN: pcTaskName=esp_timer eCurrentState=2 xTaskNumber=1, uxCurrentPriority=22
I (6666) MAIN: pcTaskName=ipc1 eCurrentState=2 xTaskNumber=3, uxCurrentPriority=24
I (6676) MAIN: pcTaskName=ipc0 eCurrentState=2 xTaskNumber=2, uxCurrentPriority=24
I (7686) MAIN: taskState=1
I (7686) MAIN: pcTaskName=main eCurrentState=1 xTaskNumber=5, uxCurrentPriority=1
I (7686) MAIN: pcTaskName=IDLE1 eCurrentState=1 xTaskNumber=7, uxCurrentPriority=0
I (7686) MAIN: pcTaskName=IDLE0 eCurrentState=1 xTaskNumber=6, uxCurrentPriority=0
I (7696) MAIN: pcTaskName=Tmr Svc eCurrentState=2 xTaskNumber=8, uxCurrentPriority=1
I (7706) MAIN: pcTaskName=ipc1 eCurrentState=2 xTaskNumber=3, uxCurrentPriority=24
I (7716) MAIN: pcTaskName=ipc0 eCurrentState=2 xTaskNumber=2, uxCurrentPriority=24
I (7726) MAIN: pcTaskName=esp_timer eCurrentState=2 xTaskNumber=1, uxCurrentPriority=22
I (8736) MAIN: taskState=1
I (8736) MAIN: pcTaskName=main eCurrentState=1 xTaskNumber=5, uxCurrentPriority=1
I (8736) MAIN: pcTaskName=IDLE1 eCurrentState=1 xTaskNumber=7, uxCurrentPriority=0
I (8736) MAIN: pcTaskName=IDLE0 eCurrentState=1 xTaskNumber=6, uxCurrentPriority=0
I (8746) MAIN: pcTaskName=Tmr Svc eCurrentState=2 xTaskNumber=8, uxCurrentPriority=1
I (8756) MAIN: pcTaskName=ipc0 eCurrentState=2 xTaskNumber=2, uxCurrentPriority=24
I (8766) MAIN: pcTaskName=esp_timer eCurrentState=2 xTaskNumber=1, uxCurrentPriority=22
I (8776) MAIN: pcTaskName=ipc1 eCurrentState=2 xTaskNumber=3, uxCurrentPriority=24
```

igrr, 2020-03-12:
Using uxTaskGetSystemState is a good approach. As for the callback, you can try https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html#_CPPv447vTaskSetThreadLocalStoragePointerAndDelCallback12TaskHandle_t10BaseType_tPv27TlsDeleteCallbackFunction_t.

nopnop2002, 2020-03-12:
I'll try vTaskSetThreadLocalStoragePointerAndDelCallback()
Thank you for your support.

closed 2020-03-12 by nopnop2002