espressif/esp-idf #4921: adc driver cause reset

wgcod, 2020-03-12:
There is no regular occurrence, resulting in a reset, and usually still can not run after the reset, because the same error, unless the hardware switch. Hardware problems?
[image]
[image]

costaud, 2020-03-12:
One question, did you call adc1_get_raw in an interrupt handler ?

wgcod, 2020-03-12:
No, it's all read in one thread.

koobest, 2020-03-17:
Hi, @wgcod 
Did you enable power management?

wgcod, 2020-03-17:
No.

koobest, 2020-03-17:
Can you provide code to reproduce this issue? From the log you provided. We can not find out  why this happens. 
thanks !!

csobrinho, 2020-03-31:
Also having this issue. This is the thread code I have:

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "driver/adc.h"
#include "esp_err.h"

#include "context.h"  // For context_t
#include "error.h"    // For ARG_UNUSED
#include "distance.h"

#define GP2D12_CH ADC1_CHANNEL_6
#define GP2D12_SAMPLES 16

static const char *TAG = "distance";

static void distance_task(void *arg) {
    ARG_UNUSED(arg);

    while (true) {
        uint32_t adc_reading = 0;
        for (int i = 0; i < GP2D12_SAMPLES; i++) {
            adc_reading += adc1_get_raw(GP2D12_CH);
        }
        adc_reading /= GP2D12_SAMPLES;
        ESP_LOGI(TAG, "GP2D12 ADC%d CH%d Raw: %d\t\n", 1, GP2D12_CH, adc_reading);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

esp_err_t distance_init(context_t *context) {
    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(GP2D12_CH, ADC_ATTEN_DB_11);
    xTaskCreatePinnedToCore(distance_task, "distance", 2048, context, configMAX_PRIORITIES - 2, NULL, tskNO_AFFINITY);

    return ESP_OK;
}
```

Log:
```
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40029d3d
0x40029d3d: esp_digital_reset at /Users/sobrinho/esp/esp-idf/components/esp_system/port/panic_handler.c:530 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1814
load:0x40050000,len:0x14a0
load:0x40054000,len:0x210c
entry 0x400502d8
I (50) boot: ESP-IDF v4.2-dev-910-g4fe04f115 2nd stage bootloader
I (51) boot: compile time 19:43:19
I (51) boot: chip revision: 0
I (54) boot.esp32s2: SPI Speed      : 40MHz
I (59) boot.esp32s2: SPI Mode       : DIO
I (63) boot.esp32s2: SPI Flash Size : 4MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (107) boot: End of partition table
[... 45 lines cut ...]
0x40084740: distance_task at /Users/sobrinho/CLionProjects/hydroponics/build/../main/sensors/distance.c:16

A10     : 0x8007ffff  A11     : 0x00000000  A12     : 0x00060520  A13     : 0x00000000
A14     : 0x3ffff718  A15     : 0x40084740  SAR     : 0x0000001a  EXCCAUSE: 0x00000005
0x40084740: distance_task at /Users/sobrinho/CLionProjects/hydroponics/build/../main/sensors/distance.c:16

EXCVADDR: 0x00000000  LBEG    : 0x00060520  LEND    : 0x00000000  LCOUNT  : 0x40024fd4
0x40024fd4: xt_highint4 at /Users/sobrinho/esp/esp-idf/components/esp_system/port/esp32s2/dport_panic_highint_hdl.S:58

Backtrace:0x4008ee4a:0x3ffc75a0 0x40088129:0x3ffc75c0 0x4008859d:0x3ffc75e0 0x4008474c:0x3ffc7610 0x4002c689:0x3ffc7640
0x4008ee4a: adc_ll_rtc_convert_is_done at /Users/sobrinho/esp/esp-idf/components/soc/src/esp32s2/include/hal/adc_ll.h:317
 (inlined by) adc_hal_convert at /Users/sobrinho/esp/esp-idf/components/soc/src/hal/adc_hal.c:62

0x40088129: adc_convert at /Users/sobrinho/esp/esp-idf/components/driver/adc.c:190

0x4008859d: adc1_get_raw at /Users/sobrinho/esp/esp-idf/components/driver/adc.c:323

0x4008474c: distance_task at /Users/sobrinho/CLionProjects/hydroponics/build/../main/sensors/distance.c:22 (discriminator 3)

0x4002c689: vPortTaskWrapper at /Users/sobrinho/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: f4db837a57d8ae4d

Rebooting...
```

wgcod, 2020-03-31:
Our problem should be the same, but I can't find a way to reproduce it. Do you have any way

fuAlex, 2020-03-31:
@wgcod @csobrinho Please tell me the version of the compiler you are using.

wgcod, 2020-03-31:
[image]

bhusang, 2020-03-31:
In your log you have the following line:

E (355) gpio: gpio_set_direction(263): GPIO number error

Is this expected for your code? It seems like there might be a setup issue for the GPIO that you are using. Can you show your initialization code?

fuAlex, 2020-03-31:
@wgcod @csobrinho We want need your chip version and type. this info can get when your download the flash. like this:
[image]

csobrinho, 2020-04-03:
As far as I know, that gpio is for a separate piece of code (1 wire initialization) and it's not related to this adc pin.

I never did any initialization for the gpio since I assumed that adc config would take care of that. Is that assumption wrong? I simply followed the docs and the esp-idf adc example that contains code for both ESP32 and ESP32-S2 targets.

Happy to help diagnose this issue. Thanks!

csobrinho, 2020-04-04:
My main:
```c
#include "context.h"
#include "sensors/distance.h"

static context_t *context;

void app_main() {
    context = context_create();
    ESP_ERROR_CHECK(distance_init(context));
}
```

My distance.c:
```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "driver/adc.h"
#include "esp_err.h"

#include "context.h"
#include "error.h"
#include "distance.h"

#define GP2D12_CH ADC1_CHANNEL_6
#define GP2D12_SAMPLES 16

static const char *TAG = "distance";

static void distance_task(void *arg) {
    ARG_UNUSED(arg);

    while (true) {
        uint32_t adc_reading = 0;
        for (int i = 0; i < GP2D12_SAMPLES; i++) {
            adc_reading += adc1_get_raw(GP2D12_CH);
        }
        adc_reading /= GP2D12_SAMPLES;
        ESP_LOGI(TAG, "GP2D12 ADC%d CH%d Raw: %d", 1, GP2D12_CH, adc_reading);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

esp_err_t distance_init(context_t *context) {
    adc_power_on();
    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(GP2D12_CH, ADC_ATTEN_DB_11);
    xTaskCreatePinnedToCore(distance_task, "distance", 2048, context, configMAX_PRIORITIES - 2, NULL, tskNO_AFFINITY);

    return ESP_OK;
}
```

My distance.h:
```c
#ifndef HYDROPONICS_SENSORS_DISTANCE_H
#define HYDROPONICS_SENSORS_DISTANCE_H

#include "esp_err.h"

#include "context.h"

esp_err_t distance_init(context_t *context);

#endif //HYDROPONICS_SENSORS_DISTANCE_H
```

```
Executing action: flash
[... 790 lines cut ...]
load:0x40054000,len:0x210c
entry 0x400502d8
I (50) boot: ESP-IDF v4.2-dev-910-g4fe04f115 2nd stage bootloader
I (51) boot: compile time 19:43:19
I (51) boot: chip revision: 0
I (54) boot.esp32s2: SPI Speed      : 40MHz
I (59) boot.esp32s2: SPI Mode       : DIO
I (63) boot.esp32s2: SPI Flash Size : 4MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (107) boot: End of partition table
I (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x050b8 ( 20664) map
I (126) esp_image: segment 1: paddr=0x000150e0 vaddr=0x3ffbcae0 size=0x01e0c (  7692) load
I (131) esp_image: segment 2: paddr=0x00016ef4 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /Users/sobrinho/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (138) esp_image: segment 3: paddr=0x00017300 vaddr=0x40024404 size=0x086d0 ( 34512) load
I (158) esp_image: segment 4: paddr=0x0001f9d8 vaddr=0x00000000 size=0x00640 (  1600)
I (159) esp_image: segment 5: paddr=0x00020020 vaddr=0x40080020 size=0x13058 ( 77912) map
0x40080020: _stext at ??:?
```

csobrinho, 2020-04-04:
Easier way to reproduce:

```
cd examples/peripherals/adc2
git clean -fdx
export IDF_TARGET=esp32s2; idf.py flash monitor
```

Yields:
```
(...SNIP...)
[85/93] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/esp32s2/touch_sensor_hal.c.obj
[86/93] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/esp32s2/rtc_clk.c.obj
[87/93] Linking C static library esp-idf/soc/libsoc.a
[88/93] Linking C static library esp-idf/micro-ecc/libmicro-ecc.a
[89/93] Linking C static library esp-idf/bootloader_support/libbootloader_support.a
[90/93] Linking C static library esp-idf/efuse/libefuse.a
[91/93] Linking C static library esp-idf/main/libmain.a
[92/93] Linking C executable bootloader.elf
[93/93] Generating binary image from built executable
esptool.py v3.0-dev
Generated /Users/sobrinho/esp/esp-idf/examples/peripherals/adc2/build/bootloader/bootloader.bin
[908/909] Generating binary image from built executable
esptool.py v3.0-dev
Generated /Users/sobrinho/esp/esp-idf/examples/peripherals/adc2/build/adc2.bin
[908/909] cd /Users/sobrinho/esp/esp-idf/components/esptool_py && /Users/sobrinho/Development/homebr...mples/peripherals/adc2/build" -P /Users/sobrinho/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32s2 -p /dev/cu.SLAB_USBtoUART -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 adc2.bin
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S2
Features: WiFi
Crystal is 40MHz
MAC: 7c:df:a1:00:39:b6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 4032.2 kbit/s)...
Hash of data verified.
Compressed 20000 bytes to 12140...
Writing at 0x00001000... (100 %)
Wrote 20000 bytes (12140 compressed) at 0x00001000 in 0.1 seconds (effective 1119.4 kbit/s)...
Hash of data verified.
Compressed 144368 bytes to 74561...
Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 144368 bytes (74561 compressed) at 0x00010000 in 1.2 seconds (effective 982.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/sobrinho/esp/esp-idf/examples/peripherals/adc2
Executing "/Users/sobrinho/.espressif/python_env/idf4.2_py2.7_env/bin/python /Users/sobrinho/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.SLAB_USBtoUART -b 115200 --toolchain-prefix xtensa-esp32s2-elf- /Users/sobrinho/esp/esp-idf/examples/peripherals/adc2/build/adc2.elf -m '/Users/sobrinho/.espressif/python_env/idf4.2_py2.7_env/bin/python' '/Users/sobrinho/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe8100,len:0x4
[... 71 lines cut ...]
0x400826d1: app_main at /Users/sobrinho/esp/esp-idf/examples/peripherals/adc2/build/../main/adc2_example_main.c:49

0x4008122b: main_task at /Users/sobrinho/esp/esp-idf/components/esp32s2/cpu_start.c:416

0x4002a615: vPortTaskWrapper at /Users/sobrinho/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 6dcc0ab71b90f7fb

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40027e75
0x40027e75: esp_digital_reset at /Users/sobrinho/esp/esp-idf/components/esp_system/port/panic_handler.c:530 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1808
load:0x40050000,len:0x14a0
load:0x40054000,len:0x210c
entry 0x400502d8
I (50) boot: ESP-IDF v4.2-dev-910-g4fe04f115 2nd stage bootloader
(...SNIP...)
```

csobrinho, 2020-04-04:
Just upgraded my ESP-IDF to `ge599b794b` and the ADC seems to be working now. I'm assuming https://github.com/espressif/esp-idf/commit/baa7898e35102a8106b58b9d0b01e7b69f72e00a fixed it, thanks!!

```
Generated /Users/sobrinho/esp/esp-idf/examples/peripherals/adc/build/adc.bin
[913/914] cd /Users/sobrinho/esp/esp-idf/components/esptool_py && /Users/sobrinho/Development/homebr...amples/peripherals/adc/build" -P /Users/sobrinho/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32s2 -p /dev/cu.SLAB_USBtoUART -b 3000000 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 adc.bin
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S2
Features: WiFi
Crystal is 40MHz
MAC: 7c:df:a1:00:39:b6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 3000000
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 5806.0 kbit/s)...
Hash of data verified.
Compressed 20000 bytes to 12143...
Writing at 0x00001000... (100 %)
Wrote 20000 bytes (12143 compressed) at 0x00001000 in 0.1 seconds (effective 2847.7 kbit/s)...
Hash of data verified.
Compressed 181056 bytes to 101067...
Writing at 0x00010000... (14 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (42 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (71 %)
Writing at 0x00024000... (85 %)
Writing at 0x00028000... (100 %)
Wrote 181056 bytes (101067 compressed) at 0x00010000 in 1.3 seconds (effective 1099.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/sobrinho/esp/esp-idf/examples/peripherals/adc
Executing "/Users/sobrinho/.espressif/python_env/idf4.2_py2.7_env/bin/python /Users/sobrinho/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.SLAB_USBtoUART -b 115200 --toolchain-prefix xtensa-esp32s2-elf- /Users/sobrinho/esp/esp-idf/examples/peripherals/adc/build/adc.elf -m '/Users/sobrinho/.espressif/python_env/idf4.2_py2.7_env/bin/python' '/Users/sobrinho/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x180c
load:0x40050000,len:0x14a0
load:0x40054000,len:0x210c
entry 0x400502d8
I (48) boot: ESP-IDF v4.2-dev-1034-ge599b794b-dirty 2nd stage bootloader
I (48) boot: compile time 01:02:37
I (48) boot: chip revision: 0
I (52) boot.esp32s2: SPI Speed      : 40MHz
I (57) boot.esp32s2: SPI Mode       : DIO
I (61) boot.esp32s2: SPI Flash Size : 2MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x05f90 ( 24464) map
I (125) esp_image: segment 1: paddr=0x00015fb8 vaddr=0x3ffbe940 size=0x020e8 (  8424) load
I (129) esp_image: segment 2: paddr=0x000180a8 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /Users/sobrinho/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (136) esp_image: segment 3: paddr=0x000184b4 vaddr=0x40024404 size=0x07b64 ( 31588) load
I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x19918 (104728) map
0x40080020: _stext at ??:?

I (185) esp_image: segment 5: paddr=0x00039940 vaddr=0x4002bf68 size=0x029d8 ( 10712) load
0x4002bf68: rwdt_ll_write_protect_enable at /Users/sobrinho/esp/esp-idf/examples/peripherals/adc/build/../../../../components/soc/src/esp32s2/include/hal/rwdt_ll.h:264
 (inlined by) wdt_hal_init at /Users/sobrinho/esp/esp-idf/components/soc/src/hal/wdt_hal_iram.c:62

I (194) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
I (196) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (204) cpu_start: Pro cpu up.
I (208) cpu_start: Application information:
I (212) cpu_start: Project name:     adc
I (217) cpu_start: App version:      v4.2-dev-1034-ge599b794b-dirty
I (224) cpu_start: Compile time:     Apr  4 2020 01:02:24
I (230) cpu_start: ELF file SHA256:  af8ec44bf7cf2d42...
I (236) cpu_start: ESP-IDF:          v4.2-dev-1034-ge599b794b-dirty
I (243) cpu_start: Single core mode
I (247) heap_init: Initializing. RAM available for dynamic allocation:
I (254) heap_init: At 3FFC1410 len 0003ABF0 (234 KiB): D/IRAM
I (261) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (267) cpu_start: Pro cpu start user code
I (324) spi_flash: detected chip: generic
I (325) spi_flash: flash io: dio
W (325) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (335) cpu_start: Starting scheduler on PRO CPU.
ADC1 CH6 Raw: 6060
ADC1 CH6 Raw: 2748
ADC1 CH6 Raw: 2766
ADC1 CH6 Raw: 2731
ADC1 CH6 Raw: 2694
ADC1 CH6 Raw: 2797
ADC1 CH6 Raw: 2802
ADC1 CH6 Raw: 2809
ADC1 CH6 Raw: 2877
```

AxelLin, 2020-04-05:
> Just upgraded my ESP-IDF to `ge599b794b` and the ADC seems to be working now. I'm assuming [baa7898](https://github.com/espressif/esp-idf/commit/baa7898e35102a8106b58b9d0b01e7b69f72e00a) fixed it, thanks!!
> 
But IMHO, the https://github.com/espressif/esp-idf/commit/baa7898e35102a8106b58b9d0b01e7b69f72e00a includes too many changes so it's difficult to know which part is refactor and which part is bug fix.
The "fix adc driver" does not tell what was really fixed.
Please don't mix refactor with bug fixes.

costaud, 2020-04-07:
@csobrinho Seems you are using esp32s2 chip, and the issue you found has been fixed on master branch. 
@wgcod  Seems you are using esp32 chip, could you please provide some piece of code for us to check. It would be great if you can provide a full test code although this can not be reproduced regularly.

Alvin1Zhang, 2020-05-26:
@csobrinho @wgcod Thanks for sharing the details, would you please help share if any updates for the issue? Thanks.

costaud, 2020-05-26:
I'll close this issue for now,  
@wgcod Feel free to reopen this issue if you find more information.

closed 2020-05-26 by costaud