espressif/esp-idf #4922: WIFI TimeSync

sajjadakhter, 2020-03-12:
is there any chip that supports Wifi TimeSync
Similar question was asked almost 2 years ago, i thought to check if any thing change 
even if its planned for future.

sajjadakhter, 2020-03-12:
https://github.com/espressif/esp-idf/issues/2326

liuzfesp, 2020-03-12:
Hi @sajjadakhter I think ESP32 S2 chip supports it, but the software is not ready yet. Please correct me if I'm wrong @sagb2015 .

sagb2015, 2020-04-13:
@sajjadakhter Yes ESP32 S2 chip support timesync. We are in process of implementing FTM feature as per IEEE specification.

sajjadakhter, 2020-04-13:
> @sajjadakhter Yes ESP32 S2 chip support timesync. We are in process of implementing FTM feature as per IEEE specification.

Any timeline? will it be able to sync over esp-now as well or only wifi?

dinuchiriac, 2020-05-27:
I'm interested in this as well. Do we have an estimated timeline when will be able to test TimeSync?

sagb2015, 2020-05-27:
@dinuchiriac We are currently targeting the availability by Mid July.

dinuchiriacsonos, 2020-06-01:
@sagb2015  Thanks,   Looking forward to testing it.

reletobg, 2020-08-29:
Is there any updates around the TimeSync? @sagb2015

sagb2015, 2020-09-04:
@reletobg  This feature is under test. However, we need more time to evaluate accuracy under different environmental conditions.

sagb2015, 2020-12-11:
Beta release for the feature is [here](https://github.com/espressif/esp-idf/tree/feature/ftm_support) . The example can be found [here](https://github.com/espressif/esp-idf/tree/feature/ftm_support/examples/wifi/ftm).

sajjadakhter, 2020-12-11:
Is there an example to use FTM for time sync. Trying to save myself some coding if its already done.

sagb2015, 2020-12-11:
@sajjadakhter  I need to apologise on this. Actually,  the Timesync program from WFA and FTM (WiFi location) are two different things even though FTM hardware can be used for Timesync (T1-T4) values to be precise.  We will look into this feature request.

VladimirP1, 2023-01-21:
Any update on this?
I would be interested in accurate time sync over wifi

nachiketkukade, 2025-03-17:
@sajjadakhter , WiFi TimeSync program doesn't exist anymore OR delayed. As of now - except 802.1AS-2020 over Wi-Fi - there is no standard way to perform Wi-Fi based time synchronisation. We do not support 802.1AS, however it's possible to perform Time Synchronisation between 2 ESP devices using FTM. 
@VladimirP1 , if interested we can provide patches that can work as PoC for FTM based Time Synchronisation