espressif/esp-idf #4923: RX IR signal from remote

varunsavita, 2020-03-12:
I want to read data from my AC remote that's why
Hardware-  TSOP with esp32.
Code - Example - rmt_nec_tx_rx
Changed in code - **line** - rmt_driver_install(rmt_rx.channel, 10000, 0);
I got error 
"
E (19513) rmt: RMT[0] ERR
E (19513) rmt: status: 0x13000040
I (24653) NEC: RMT TX DATA
I (36783) NEC: RMT TX DATA
I (48963) NEC: RMT TX DATA
E (57923) rmt: RMT RX BUFFER FULL
E (59353) rmt: RMT RX BUFFER FULL
E (59513) rmt: RMT RX BUFFER FULL
I (61143) NEC: RMT TX DATA
E (61513) rmt: RMT RX BUFFER FULL
E (61683) rmt: RMT RX BUFFER FULL
"
Please guide me.

suda-morris, 2020-03-12:
Hi @varunsavita 
We have a newer example for IR Protocol : https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/ir_protocols

varunsavita, 2020-03-12:
I was trying according to you
@suda-morris 
Code - ir_protocols
just build
got error
 1- 
   rmt_config_t rmt_rx_config = RMT_DEFAULT_CONFIG_RX(CONFIG_EXAMPLE_RMT_RX_GPIO, example_rx_channel);
                                  ^~~~~~~~~~~~~~~~~~~~~
                                  RMT_APB_CONF_REG
../main/ir_protocols_main.c:36:34: error: invalid initializer
../main/ir_protocols_main.c: In function 'example_ir_tx_task':
../main/ir_protocols_main.c:84:34: error: implicit declaration of function 'RMT_DEFAULT_CONFIG_TX'; did you mean 'IR_PARSER_DEFAULT_CONFIG'? [-Werror=implicit-function-declaration]
     rmt_config_t rmt_tx_config = RMT_DEFAULT_CONFIG_TX(CONFIG_EXAMPLE_RMT_TX_GPIO, example_tx_channel);
                                  ^~~~~~~~~~~~~~~~~~~~~
                                  IR_PARSER_DEFAULT_CONFIG

2 - 
    NEC_CHECK(rmt_get_counter_clock((rmt_channel_t)config->dev_hdl, &counter_clk_hz) == ESP_OK,
               ^~~~~~~~~~~~~~~~~~~~~
../components/infrared_tools/src/ir_builder_rmt_nec.c:24:15: note: in definition of macro 'NEC_CHECK'
         if (!(a))                                                                 \
               ^
cc1.exe: some warnings being treated as errors

negativekelvin, 2020-03-12:
You have to build the example against matching idf version not old version

varunsavita, 2020-03-16:
I have downloaded freshest IDF from 
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-1-install-prerequisites
git clone --recursive https://github.com/espressif/esp-idf.git
Is it work?

varunsavita, 2020-03-18:
@suda-morris @FayeY @igrr  Hello guys
Thanks for supporting me.
I am able to receive data from the AC remote.
I used both receivers IRM 3638 and tsop1838.

Alvin1Zhang, 2020-03-19:
@varunsavita Glad to hear that you are able to receive data from the AC remote. @negativekelvin Thanks for the notes. Feel free to reopen if the issue still happens. Thanks.

closed 2020-03-19 by Alvin1Zhang

varunsavita, 2020-03-19:
@Alvin1Zhang Hello
My bad, Currently Data is not receiving.
No Idea what is going on.