espressif/esp-idf #4924: console example: rapid input triggers task watchdog in linenoise

uname2219, 2020-03-12:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.1-dev-2079-g5dbabae9d
- Build System:         make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Rapidly pressing the enter key along with garbage characters in the console example will cause the console to become non-responsive. At that point the task watchdog will trigger on the main task. This will continue until the next power cycle even with no more input.

I noticed this problem in my application on ESP-IDF 4.1, 4.0, and the master branch before that. I use a terminal window with 120 columns and when rapidly entering commands or just random characters, I would see the terminal appear to freeze and then start to print the task watchdog message.

Perhaps an indicator of the problem, I sometimes see the first letter of the prompt on the far right column (120) of the terminal. It then repeats the entire prompt on the next line.

I imagine the problem is all my keypresses are interfering with the expected ANSI sequences linenoise is sending and receiving to create a terminal. However, I can't have the console put the device in a useless state by filling one core.

### Steps to reproduce

1. Compile the console example in examples/system/console
2. Flash the application and get the prompt 'esp32>'
3. Rapidly bang away at your keyboard with a large number of enter keypresses
4. Console will stop responding and then the task watchdog message will start printing

### Code to reproduce this issue

The unmodified examples/system/console application.

## Debug Logs
Note the output is wider than the default window.

```
This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
esp32> a
Unrecognized command
esp32> a
Unrecognized command
esp32> a
Unrecognized command
esp32> 
esp32> 
                                                                                                                       eesp32> 
esp32> a
Unrecognized command
esp32> 
esp32> 
esp32> f
Unrecognized command
esp32> f
Unrecognized command
esp32> 
                                                                                                                       eesp32> f
Unrecognized command
esp32> 
                                                                                                                       eesp32> r
Unrecognized command
esp32> 
                                                                                                                       eE (18556) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18556) task_wdt:  - IDLE0 (CPU 0)
E (18556) task_wdt: Tasks currently running:
E (18556) task_wdt: CPU 0: main
E (18556) task_wdt: CPU 1: IDLE1
E (23556) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (23556) task_wdt:  - IDLE0 (CPU 0)
E (23556) task_wdt: Tasks currently running:
E (23556) task_wdt: CPU 0: main
E (23556) task_wdt: CPU 1: IDLE1
E (28556) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (28556) task_wdt:  - IDLE0 (CPU 0)
E (28556) task_wdt: Tasks currently running:
E (28556) task_wdt: CPU 0: main
E (28556) task_wdt: CPU 1: IDLE1

```

## linenoise tracing

I have attempted to trace this code in the linenoise library. First to linenoiseEdit:

```cpp
int pos1 = getCursorPosition();
    if (fwrite(prompt,l.plen,1,stdout) == -1) return -1;
    int pos2 = getCursorPosition();
    if (pos1 >= 0 && pos2 >= 0) {
        l.plen = pos2 - pos1;
    }
```

I added a line of debugging:

```cpp
int pos1 = getCursorPosition();
    if (fwrite(prompt,l.plen,1,stdout) == -1) return -1;
    int pos2 = getCursorPosition();
    if (pos1 >= 0 && pos2 >= 0) {
        printf("l.cols=%i, pos1=%i, pos2=%i\n", l.cols, pos1, pos2);
        l.plen = pos2 - pos1;
    }
```

With this new code, I can see a problem just by pressing enter twice in quick succession. The output I get on a 120 column terminal is:

```
esp32> 
                                                                                                                       e
sp32> l.cols=80, pos1=120, pos2=7
```

Which means the getColumns() call has failed and returned its error value of 80. Also, I see pos1 = 120, and pos2 = 7. So then l.plen is assigned 7 - 120 = -113 which seems like a potential problem for the length of the prompt.

Inside the getColumns() function, I see errors occurring in getCursorPosition(), both to calculate start and cols.

## Workaround

The solution I've been using is to have a 'dumb' prompt, with no ANSI sequences. The code above from linenoiseEdit is just trying to determine the effective length of the prompt on the screen (ignoring any escape sequences). If I comment out the calculation, then my console appears stable.

```cpp
//    int pos1 = getCursorPosition();
    if (fwrite(prompt,l.plen,1,stdout) == -1) return -1;
/*    int pos2 = getCursorPosition();
    if (pos1 >= 0 && pos2 >= 0) {
        l.plen = pos2 - pos1;
    }*/

```

tanxn5, 2020-08-15:
`static int getCursorPosition() {
    char buf[32];
    int cols, rows;
    unsigned int i = 0;

    /* Report cursor location */
    fprintf(stdout, "\x1b[6n");

    /* Read the response: ESC [ rows ; cols R */
    for (i = 0; i < 32; i++) {
        if (fread(buf + 0, 1, 1, stdin) != 1) break;
        if (buf[0] == ESC) break;
    }
    for (i = 1; i < sizeof(buf) - 1; i++) {
        if (fread(buf + i, 1, 1, stdin) != 1) break;
        if (buf[i] == 'R') break;
    }
    buf[i] = '\0';
    /* Parse it. */
    if (buf[0] != ESC || buf[1] != '[') return -1;
    if (sscanf(buf + 2, "%d;%d", &rows, &cols) != 2) return -1;
    return cols;
}`

tanxn5, 2020-08-15:
~~~c
static int getCursorPosition() {
    char buf[32];
    int cols, rows;
    unsigned int i = 0;

    /* Report cursor location */
    fprintf(stdout, "\x1b[6n");

    /* Read the response: ESC [ rows ; cols R */
    for (i = 0; i < 32; i++) {
        if (fread(buf + 0, 1, 1, stdin) != 1) break;
        if (buf[0] == ESC) break;
    }
    for (i = 1; i < sizeof(buf) - 1; i++) {
        if (fread(buf + i, 1, 1, stdin) != 1) break;
        if (buf[i] == 'R') break;
    }
    buf[i] = '\0';
    /* Parse it. */
    if (buf[0] != ESC || buf[1] != '[') return -1;
    if (sscanf(buf + 2, "%d;%d", &rows, &cols) != 2) return -1;
    return cols;
}
~~~

ArwedL, 2020-12-06:
+1 having same problem (enabling linenoise from within telnet client)

closed 2024-01-30 by espressif-bot