espressif/esp-idf #4927: Change in generated symbol names for embedded files between ESP-IDF v3.2.2 and v4.0

tonystuart, 2020-03-12:
I recently migrated a project from ESP-IDF release 3.2.2 with Make to ESP-IDF release 4.0 with CMake.

The project includes a number of machine generated files that are embedded using idf_component_register and EMBED_FILES.

Some of the machine generated filenames begin with a digit and some begin with a letter. 

In ESP-IDF release 4.0, it appears that the generated symbol names for those filenames that begin with a digit include an extra underscore (e.g. _binary__12345_start versus _binary_abcde_start).

This is presumably to ensure that the other symbol names in the generated assembly language file (i.e. those without the \_binary\_ prefix) are valid symbol names and don't begin with a digit.

However, I couldn't find this documented on the Build System page:

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#embedding-binary-data

If this is the intended behavior, it would be great if the documentation could be updated to make this clear.

Alvin1Zhang, 2020-03-13:
@tonystuart Thanks for reporting, we will look into. Thanks.

renzbagaporo, 2020-06-02:
Thanks for bringing this into our attention @tonystuart. This looks unintentional to me, but I have to verify with some colleagues.

Alvin1Zhang, 2020-07-10:
Thanks for reporting and sorry for late reply, the fix is available at https://github.com/espressif/esp-idf/commit/b3e77c7ce92ce08ddf2d003a0ce7733d5147e69b. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-10 by Alvin1Zhang