espressif/esp-idf #4928: BT related panic (rwble.c)

fpgamaster, 2020-03-13:
Hi All,

I've read many threads about similar problem but it seems it is not fixed yet :(
Any ideas how to mitigate or fix this are welcome.

BLE and WiFi coexists, GATTS and GATTC too. Can provide some code if needed.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit: ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-PICO-D4
- IDF version: v4.2-dev-662-g06fa9476a
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Panic in few hours of uptime:
===============
1-st panic in few hours
===============

ASSERT_PARAM(512 0), in rwble.c at line 238

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

Core 0 register dump:

PC      : 0x40089079  PS      : 0x00060034  A0      : 0x8008d278  A1      : 0x3ffbf6b0  

[0;33m0x40089079: r_assert_param at ??:?
[0m
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  

A6      : 0x3ffbdbe4  A7      : 0x60008050  A8      : 0x80089079  A9      : 0x3ffbf690  

A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  

A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x40088fb1  LEND    : 0x40088fb8  LCOUNT  : 0x00000000  

[0;33m0x40088fb1: r_assert_param at ??:?
[0m
[0;33m0x40088fb8: r_assert_param at ??:?
[0m
Core 0 was running in ISR context:

EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40089079

[0;33m0x40089079: r_assert_param at ??:?
[0m

ELF file SHA256: 57fea2d08d5f0aaaa8ba74aa2c3d33855a1e16654354b554f3d6aab2c397c1c5

Backtrace: 0x40089076:0x3ffbf6b0 0x4008d275:0x3ffbf6d0 0x4008dea3:0x3ffbf6f0 0x40086089:0x3ffbf710 0x4000bfed:0x3ffbbe90 0x4009eee5:0x3ffbbea0 0x4009cd50:0x3ffbbec0 0x4009cf44:0x3ffbbee0 0x4009ec85:0x3ffbbf00

[0;33m0x40089076: r_assert_param at ??:?
[0m
[0;33m0x4008d275: r_rwble_isr at ??:?
[0m
[0;33m0x4008dea3: r_rwbtdm_isr_wrapper at intc.c:?
[0m
[0;33m0x40086089: _xt_lowint1 at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1192
[0m
[0;33m0x4009eee5: vPortExitCritical at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:419
[0m
[0;33m0x4009cd50: prvGetExpectedIdleTime at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:2202
[0m
[0;33m0x4009cf44: prvIdleTask at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:3409
[0m
[0;33m0x4009ec85: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Core 1 register dump:

PC      : 0x400d18e4  PS      : 0x00060634  A0      : 0x800d246b  A1      : 0x3ffbc610  

[0;33m0x400d18e4: esp_pm_impl_waiti at /home/xxxxx/esp/esp-idf/components/esp32/pm_esp32.c:480
[0m
A2      : 0x00000001  A3      : 0x00060620  A4      : 0x00060620  A5      : 0x00060623  

A6      : 0x00000001  A7      : 0x00060023  A8      : 0x00000001  A9      : 0x3ffbc600  

A10     : 0x00000003  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00060623  

A14     : 0x00000001  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 57fea2d08d5f0aaaa8ba74aa2c3d33855a1e16654354b554f3d6aab2c397c1c5

Backtrace: 0x400d18e1:0x3ffbc610 0x400d2468:0x3ffbc630 0x4009cf41:0x3ffbc650 0x4009ec85:0x3ffbc670

[0;33m0x400d18e1: esp_pm_impl_waiti at /home/xxxxx/esp/esp-idf/components/esp32/pm_esp32.c:475
[0m
[0;33m0x400d2468: esp_vApplicationIdleHook at /home/xxxxx/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63
[0m
[0;33m0x4009cf41: prvIdleTask at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:3385
[0m
[0;33m0x4009ec85: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Rebooting...

[0;32mI (1022) cpu_start: Pro cpu up.[0m

================
2-nd panic in few hours
================

ASSERT_PARAM(512 0), in rwble.c at line 238

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

Core 0 register dump:

PC      : 0x4008907e  PS      : 0x00060034  A0      : 0x8008d278  A1      : 0x3ffbf6b0  

[0;33m0x4008907e: r_assert_param at ??:?
[0m
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000200  A5      : 0x60008054  

A6      : 0x3ffbdbe4  A7      : 0x60008050  A8      : 0x80089079  A9      : 0x3ffbf690  

A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  

A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x40088fb1  LEND    : 0x40088fb8  LCOUNT  : 0x00000000  

[0;33m0x40088fb1: r_assert_param at ??:?
[0m
[0;33m0x40088fb8: r_assert_param at ??:?
[0m
Core 0 was running in ISR context:

EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008907e

[0;33m0x4008907e: r_assert_param at ??:?
[0m

ELF file SHA256: 57fea2d08d5f0aaaa8ba74aa2c3d33855a1e16654354b554f3d6aab2c397c1c5

Backtrace: 0x4008907b:0x3ffbf6b0 0x4008d275:0x3ffbf6d0 0x4008dea3:0x3ffbf6f0 0x40086089:0x3ffbf710 0x4000bfed:0x3ffbbe40 0x4009eee5:0x3ffbbe50 0x400d1ea0:0x3ffbbe70 0x400d1eab:0x3ffbbea0 0x400d2455:0x3ffbbec0 0x4009cf41:0x3ffbbee0 0x4009ec85:0x3ffbbf00

[0;33m0x4008907b: r_assert_param at ??:?
[0m
[0;33m0x4008d275: r_rwble_isr at ??:?
[0m
[0;33m0x4008dea3: r_rwbtdm_isr_wrapper at intc.c:?
[0m
[0;33m0x40086089: _xt_lowint1 at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1192
[0m
[0;33m0x4009eee5: vPortExitCritical at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:419
[0m
[0;33m0x400d1ea0: esp_task_wdt_reset at /home/xxxxx/esp/esp-idf/components/esp32/task_wdt.c:322
[0m
[0;33m0x400d1eab: idle_hook_cb at /home/xxxxx/esp/esp-idf/components/esp32/task_wdt.c:79
[0m
[0;33m0x400d2455: esp_vApplicationIdleHook at /home/xxxxx/esp/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)
[0m
[0;33m0x4009cf41: prvIdleTask at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:3385
[0m
[0;33m0x4009ec85: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Core 1 register dump:

PC      : 0x400d245c  PS      : 0x00060a34  A0      : 0x8009cf44  A1      : 0x3ffbc630  

[0;33m0x400d245c: esp_vApplicationIdleHook at /home/xxxxx/esp/esp-idf/components/esp_common/src/freertos_hooks.c:50 (discriminator 2)
[0m
A2      : 0x00000005  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00060023  

A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ffc8d98  A9      : 0x3ffc8d64  

A10     : 0x00000000  A11     : 0x00060a20  A12     : 0x00000014  A13     : 0x000000a5  

A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 57fea2d08d5f0aaaa8ba74aa2c3d33855a1e16654354b554f3d6aab2c397c1c5

Backtrace: 0x400d2459:0x3ffbc630 0x4009cf41:0x3ffbc650 0x4009ec85:0x3ffbc670

[0;33m0x400d2459: esp_vApplicationIdleHook at /home/xxxxx/esp/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)
[0m
[0;33m0x4009cf41: prvIdleTask at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:3385
[0m
[0;33m0x4009ec85: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Rebooting...

[0;32mI (1022) cpu_start: Pro cpu up.[0m

GYC-Espressif, 2020-03-13:
Hi. Can you describe how this problem happened? Have you disabled and enabled the Bluetooth controller for BT? Can you provide some code to reproduce the problem? Thank you.

fpgamaster, 2020-03-13:
Hi,
No, the BT controller is not disabled and enabled after the initial enable at startup. Unfortunately it is a big project and it is almost impossible to isolate a simple code which reproduces the issue. Basically it is a device with WiFi, GATTC and GATTS enabled, also PM in light sleep mode is used. There is no heavy traffic on WIFI and BT just 200-300 bytes in 20-30 seconds. There are no stack issues or buffer overflows. It seems that rwble.c is not a part of the edp-idf project but of BT firmware... I will try to disable PM and to make sure that every task related to BT and WiFi runs on same CPU core but actually I do not have any idea whether it is related or not.

fpgamaster, 2020-03-14:
Another panic, might be more useful:

ASSERT_PARAM(512 0), in rwble.c at line 238

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

Core 0 register dump:

PC      : 0x40088e4d  PS      : 0x00060034  A0      : 0x8008d04c  A1      : 0x3ffbf6b0  

[0;33m0x40088e4d: r_assert_param at ??:?
[0m
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  

A6      : 0x3ffbdbe4  A7      : 0x60008050  A8      : 0x80088e4d  A9      : 0x3ffbf690  

A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  

A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x40088d85  LEND    : 0x40088d8c  LCOUNT  : 0x00000000  

[0;33m0x40088d85: r_assert_param at ??:?
[0m
[0;33m0x40088d8c: r_assert_param at ??:?
[0m
Core 0 was running in ISR context:

EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40088e4d

[0;33m0x40088e4d: r_assert_param at ??:?
[0m

ELF file SHA256: a3dd858b37fda9572dd4291b3852ad1083d4b55bf0cc7578bd318dcf68722c59

Backtrace: 0x40088e4a:0x3ffbf6b0 0x4008d049:0x3ffbf6d0 0x4008dc77:0x3ffbf6f0 0x4008602d:0x3ffbf710 0x4000bfed:0x3ffdd3b0 0x4009e929:0x3ffdd3c0 0x4009c026:0x3ffdd3e0 0x400d2cb1:0x3ffdd420 0x4008347d:0x3ffdd440 0x400837e0:0x3ffdd480 0x400fa796:0x3ffdd4c0 0x400d2b4d:0x3ffdd4e0 0x400f4b9d:0x3ffdd500 0x400f7391:0x3ffdd550 0x400f741d:0x3ffdd570 0x40092dbe:0x3ffdd590 0x4009e6c9:0x3ffdd5d0

[0;33m0x40088e4a: r_assert_param at ??:?
[0m
[0;33m0x4008d049: r_rwble_isr at ??:?
[0m
[0;33m0x4008dc77: r_rwbtdm_isr_wrapper at intc.c:?
[0m
[0;33m0x4008602d: _xt_lowint1 at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1192
[0m
[0;33m0x4009e929: vPortExitCritical at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:419
[0m
[0;33m0x4009c026: xQueueGenericReceive at /home/xxxxx/esp/esp-idf/components/freertos/queue.c:1541
[0m
[0;33m0x400d2cb1: semphr_take_wrapper at /home/xxxxx/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:277
[0m
[0;33m0x4008347d: coex_bb_reset_unlock at ??:?
[0m
[0;33m0x400837e0: coex_core_request at ??:?
[0m
[0;33m0x400fa796: coex_wifi_request at ??:?
[0m
[0;33m0x400d2b4d: coex_wifi_request_wrapper at /home/xxxxx/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:478
[0m
[0;33m0x400f4b9d: pm_tbtt_process at ??:?
[0m
[0;33m0x400f7391: dbg_lmac_init at ??:?
[0m
[0;33m0x400f741d: pp_timer_do_process at ??:?
[0m
[0;33m0x40092dbe: ppTask at ??:?
[0m
[0;33m0x4009e6c9: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Core 1 register dump:

PC      : 0x4009e812  PS      : 0x00060534  A0      : 0x8009c749  A1      : 0x3ffbc540  

[0;33m0x4009e812: compare_and_set_native at /home/xxxxx/esp/esp-idf/components/soc/include/soc/compare_set.h:28
 (inlined by) spinlock_acquire at /home/xxxxx/esp/esp-idf/components/soc/include/soc/spinlock.h:91
 (inlined by) vPortCPUAcquireMutex at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:394
[0m
A2      : 0x3ffbf0f4  A3      : 0x00060520  A4      : 0x00000014  A5      : 0x000000a5  

A6      : 0x000000a5  A7      : 0x00060023  A8      : 0x0000abab  A9      : 0x0000abab  

A10     : 0x0000cdcd  A11     : 0xb33fffff  A12     : 0x00060520  A13     : 0x00060523  

A14     : 0x00000001  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: a3dd858b37fda9572dd4291b3852ad1083d4b55bf0cc7578bd318dcf68722c59

Backtrace: 0x4009e80f:0x3ffbc540 0x4009c746:0x3ffbc560 0x4009c988:0x3ffbc580 0x4009e6c9:0x3ffbc5a0

[0;33m0x4009e80f: spinlock_acquire at /home/xxxxx/esp/esp-idf/components/soc/include/soc/spinlock.h:75
 (inlined by) vPortCPUAcquireMutex at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:394
[0m
[0;33m0x4009c746: prvGetExpectedIdleTime at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:2175
[0m
[0;33m0x4009c988: prvIdleTask at /home/xxxxx/esp/esp-idf/components/freertos/tasks.c:3409
[0m
[0;33m0x4009e6c9: vPortTaskWrapper at /home/xxxxx/esp/esp-idf/components/freertos/xtensa/port.c:143
[0m

Rebooting...

maarthala, 2020-06-01:
Hi. Did you find any fix for this.
I too have similar issue 
ASSERT_PARAM(512 0), in rwble.c at line 238
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40082f4a  PS      : 0x00060034  A0      : 0x80087274  A1      : 0x3ffbe370  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdc28  A7      : 0x60008050  A8      : 0x80082f45  A9      : 0x3ffbe350  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40082e7d  LEND    : 0x40082e84  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x4019283b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082f4a

Backtrace:0x40082f47:0x3ffbe370 0x40087271:0x3ffbe390 0x40087e9f:0x3ffbe3b0 0x400826e1:0x3ffbe3d0 0x4000bfed:0x3ffb7530 0x4008e699:0x3ffb7540 0x40091018:0x3ffb7560 0x4010a75f:0x3ffb75a0 0x4008ad63:0x3ffb75c0 0x40159e31:0x3ffb75f0 0x401641bd:0x3ffb7610 0x40160d5d:0x3ffb7630 0x400d47b4:0x3ffb7650 0x400d4809:0x3ffb7670 0x4008e4ea:0x3ffb7690
Core  1 register dump:
PC      : 0x4000bff0  PS      : 0x00060b34  A0      : 0x8008e60a  A1      : 0x3ffdfa20  
A2      : 0x00000000  A3      : 0x00060b23  A4      : 0x00060b20  A5      : 0x000000a5  
A6      : 0x000000a5  A7      : 0x00060023  A8      : 0x8008f217  A9      : 0x3ffbe960  
A10     : 0x00000008  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x000000a5  
A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x0000000c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4000bfed:0x3ffdfa20 0x4008e607:0x3ffdfa30 0x40091ecc:0x3ffdfa50 0x400d4ef8:0x3ffdfa70 0x4008e4ea:0x3ffdfaa0

GYC-Espressif, 2020-06-02:
Hello, @maarthala Could you tell me you IDF version and the commit ID ? And I also want to know the program action you did before the crash. Thank you.

AxelLin, 2020-08-03:
> Hi. Did you find any fix for this.
> I too have similar issue
> ASSERT_PARAM(512 0), in rwble.c at line 238
> Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0).

Maybe you need to wait for below fix for ASSERT_PARAM (512, 0)
https://github.com/espressif/esp-idf/issues/5018

AxelLin, 2020-09-28:
@fpgamaster 
The ASSERT_PARAM (512, 0) fix is already in 4.2 branch, do you still have the issue after update esp-idf?

closed 2022-05-17 by jack0c