espressif/esp-idf #4929: Graceful recovery for (BLE) provisioning failure

armandr, 2020-03-13:
The current process ble_prov is works well as long as the wifi credentials provide are correct and the device connects.

If the user provides invalid credentials, the only way seems to be to reset ESP32 and start the process over.

I tried to stop and re-start the provisioning after detecting a failure [AUTH or  AP Not found] and received the following errors:

```
I (62008) app_main: Starting BLE provisioning
E (62008) app_prov: bt_controller_mem_release failed 259
I (62018) BTDM_INIT: BT controller compile version [79dd6f1]
I (62018) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (62028) phy_init: failed to allocate memory for RF calibration data
panic.

```

Is there a better way to do this?

dhrishi, 2020-03-17:
@armandr Ideally, the provisioning stop API is to be used only when it is not required in the current boot cycle.

For errors like AP not found or auth failures, after a specific number of connection attempts, you can factory reset and restart the device in the example code itself (`nvs_flash_erase()` followed by `esp_restart()`)

armandr, 2020-03-17:
@dhrishi that's the issue.  If you call provisioning stop API, then you get the errors I noted above.

The issue with factory reset is that you lose other settings stored on the device. A unscheduled restart on our device could lead to equipment damage.

There should a graceful way to roll-back and re-try provisioning without putting the user through a factory reset and restart.  That's a high cost for a typo on wifi.

AshUK, 2020-07-03:
https://github.com/espressif/esp-idf/issues/5261#issuecomment-652523709

Would also like to see graceful handling of invalid credentials. We have used the esp erase and restart technique by emitting a special event to notify other areas of the application of the impending restart this adds complexity to the devices fw but it works. However it means the provisioning application has to go through the scanning and provisioning dance again the delay of which is notable.

My preference would be to have an attempts counter of three, after which the client is disconnected. I'm not sure what this would mean for the softap clients.

karunt, 2022-01-18:
This is a very frustrating part of wifi provisioning in ESP32.  Options to consider (which I myself am looking to implement):
1. Don't erase all of NVS Flash, but just send over empty ssid and password values to nvs when you try to re-provision (by sending a message to a custom GATT characteristic that uses a custom callback already registered in the main app.
2. As opposed to doing a factory reset, just call esp_prov_mgr_start_provisioning again, which will cause the chip to restart (not sure if doing so causes the chip to lose all information stored previously like a factory reset would do).
3. Use your client side app to test credentials first to make sure they are correct (by trying to connect to the same network) and then send them over to the esp32 chip if correct.

These are preliminary thoughts - I have yet to test them all.

gulliverrr, 2022-11-04:
Re-provisioning if initial credentials were correct is supported with this commit on master:
https://github.com/espressif/esp-idf/commit/9bc1cc706e394466dcd202b6c4a4586d8742bab2
Is not what you asked but may help someone else who stumbles up on this in the future...

jack0c, 2024-02-07:
@gulliverrr @karunt @AshUK @armandr @dhrishi From my understanding, there are no problems of the principle to support this. The lasted Blufi support many event to make connection finished before bluetooth down: https://github.com/espressif/esp-idf/commit/0c8272b2e0d9c50eb2a573f1936d28fa53617c96

closed 2025-03-14 by Alvin1Zhang