espressif/esp-idf #4930: possible issue in mdns

chegewara, 2020-03-13:
I have very simple http server serving 2 files with GET and 1 POST link. Also in code i have implemented mdns.

I am assuming it is issue with mdns because:
- all is working fine after reboot, i can access server with esp32.local,
- http server works fine,
- after some time i still can access esp32 with IP, but connecting with esp32.local is printing such logs:
`ip4_output_if: Packets larger than MTU, discarded!!!`

esp-idf v3.3 commit 6f9a7264ce20c6132fbd8309112630d0eb490fe4

david-cermak, 2020-04-24:
@chegewara Sorry for not responding in more timely manner.
* Are there any other mdns devices in the local network? Is that reproducible also if you have only one device, the esp32.local?
* Could you please enable the IP fragmentation if disabled, to see if the issue still exists? `CONFIG_LWIP_IP_FRAG=y`

chegewara, 2020-04-30:
Hi @david-cermak 

> * Are there any other mdns devices in the local network? Is that reproducible also if you have only one device, the esp32.local?

yes, there is few more mdns devicesin the same network, but only 1 esp32

> * Could you please enable the IP fragmentation if disabled, to see if the issue still exists? `CONFIG_LWIP_IP_FRAG=y`

enabling this option seems to fix issue

Thanks for help.

closed 2020-04-30 by chegewara

AxelLin, 2020-04-30:
> > * Could you please enable the IP fragmentation if disabled, to see if the issue still exists? `CONFIG_LWIP_IP_FRAG=y`
> 
> enabling this option seems to fix issue

But why mdns needs CONFIG_LWIP_IP_FRAG=y to work?
If mdns really needs CONFIG_LWIP_IP_FRAG=y, had better show some warning if CONFIG_LWIP_IP_FRAG=n.

chegewara, 2020-04-30:
I found that another option has been enabled, will test if this change anything and report:
`CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE=y`

chegewara, 2020-04-30:
> I found that another option has been enabled, will test if this change anything and report:
> `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE=y`

Nope, this option changes nothing, but with `CONFIG_LWIP_IP_FRAG=y` server is working very stable now. (maybe it was not mdns)