espressif/esp-idf #4931: Error compiling latest master branch.

KaeLL, 2020-03-13:
Git describe: v4.2-dev-674-g640c7c510

```
In file included from esp_idf/test_master/esp-idf/components/bt/controller/bt.c:38:
esp_idf/test_master/esp-idf/components/bt/controller/bt.c: In function 'esp_bt_controller_mem_release':
esp_idf/test_master/esp-idf/components/log/include/esp_log.h:322:47: error: 'mem_start' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                               ^~~~~~~~~~~~~
esp_idf/test_master/esp-idf/components/bt/controller/bt.c:975:14: note: 'mem_start' was declared here
     intptr_t mem_start, mem_end;
```

dobairoland, 2020-03-14:
This looks like a duplicate of https://github.com/espressif/esp-idf/issues/4918.

KaeLL, 2020-03-14:
it is. You may close it, then.

Alvin1Zhang, 2020-03-16:
@KaeLL Thanks for reporting, let us track in https://github.com/espressif/esp-idf/issues/4918, thanks.

closed 2020-03-16 by Alvin1Zhang