espressif/esp-idf #4933: Unbielivable error in esp_wifi.c: invalid magic number

awander, 2020-03-14:
In the C++ code bellow (Base on ESP-IDF/release/v4.1):
`
esp_err_t HalWifiImpl::init() {
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    printf("feature_caps = %llu", cfg->feature_caps);  // OK // feature_caps = 0
    printf("magic = 0x%08X", cfg->magic);              // OK // magic = 0x1F2F3F4F
    RET_ON_ERROR( esp_wifi_init(&cfg) );        // !!! Unbelievable error here !!!
    RET_ON_ERROR( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
...
}`

When the parameter "wifi_init_config_t *cfg" is passed into "components/esp_wifi/wifi_init.c: esp_wifi_init()", it's not the same as passed !
`
esp_err_t esp_wifi_init(const wifi_init_config_t *config)
{
...
    // cfg->* // The most fields are correct, except the last two bellow:
    printf("feature_caps = %llu", cfg->feature_caps);  // Unbelievable ! // feature_caps = 2247084347616133120
    printf("magic = 0x%08X", cfg->magic);    // Unbelievable ! //magic = 0x3F403FF8
    esp_err_t result = esp_wifi_init_internal(config);
    if (result == ESP_OK) {
        esp_wifi_set_debug_log();
...
`

It's unbelievable ! It just happened suddenly, I don't know why ...

I mixed C/C++ program in my project, there are many strange problems ! This is the most unbelievable one !

I think it maybe the difference of struct field initialization between C and C++, or the field alignment    difference ?

awander, 2020-03-14:
I solved this problem temporarily by:

chegewara, 2020-03-14:
Try this:
```
wifi_init_config_t cfg = {0};
cfg = WIFI_INIT_CONFIG_DEFAULT();
```

liuzfesp, 2020-03-16:
Hi @awander, have you call "git submodule update" when you upgrade your IDF?

Alvin1Zhang, 2020-03-24:
@awander Glad that the issue gets resolved. Please help check the note by @liuzfesp , thanks.

closed 2020-03-24 by Alvin1Zhang