espressif/esp-idf #4936: idf.py flash error

GarMingLi, 2020-03-15:
My board model is nodemcu-32s and the flash size is 32Mbit.
The serial number is "Com9".
After compiling, I executed the command "idf.py-p /dev/ttyS9 flash" and the following error occurred:
```
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory /home/ljm/esp32/esp-idf/examples/get-started/hello_world/build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory /home/ljm/esp32/esp-idf/examples/get-started/hello_world/build
Executing "/home/ljm/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/ljm/esp32/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS9 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash @flash_project_args"...
esptool.py -p /dev/ttyS9 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v2.9-dev
Serial port /dev/ttyS9
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:55:37:24
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
esptool.py failed with exit code 2
```

igrr, 2020-03-16:
@GarMingLi Could you please try the troubleshooting instructions listed here: https://github.com/espressif/esptool#troubleshooting?

projectgus, 2020-03-31:
Hi @GarMingLi ,

Did you have any luck with troubleshooting? Can you tell us anything about your hardware setup?

Also, what happens if you set the serial port to `COM9` instead of `/dev/ttyS9`? Are you using WSL?

Alvin1Zhang, 2020-05-13:
@GarMingLi Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

projectgus, 2020-06-03:
Closing due to a lack of updates.

closed 2020-06-03 by projectgus