espressif/esp-idf #4937: NimBLE build errors

AxelLin, 2020-03-16:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-701-g0ae960f2fe92
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

There are some build issues with NimBLE Options:

1. If unchecked BT_NIMBLE_NVS_PERSIST, I got below build error:

In file included from /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_sm.c:51:
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/../store/config/src/ble_store_config_priv.h:54:19: error: static declaration of 'ble_store_persist_peer_records' follows non-static declaration
 static inline int ble_store_persist_peer_records(void)      { return 0; }
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_sm.c:50:
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv_priv.h:41:5: note: previous declaration of 'ble_store_persist_peer_records' was here
 int ble_store_persist_peer_records(void);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[660/1077] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

2. If unchecked BT_NIMBLE_SM_LEGACY, I got below build error:

/home/axel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs_resolv.c.obj):(.literal.ble_hs_resolv_gen_priv_addr+0x4): undefined reference to `ble_sm_alg_encrypt'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs_resolv.c.obj): in function `ble_hs_resolv_gen_priv_addr':
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:377: undefined reference to `ble_sm_alg_encrypt'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs_resolv.c.obj): in function `ble_hs_resolv_rpa':
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:685: undefined reference to `ble_sm_alg_encrypt'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

dhrishi, 2020-03-16:
@AxelLin Thanks for reporting this issue! We will look into it and update.

Alvin1Zhang, 2020-03-23:
@AxelLin Thanks for reporting, we have a fix which is now under internal reviewing, will update once it is available on GitHub. Thanks.

AxelLin, 2020-04-01:
Fixed by 38fc24291418b6aa204090451c82bdba78c0ba66

closed 2020-04-01 by AxelLin