espressif/esp-idf #4939: clang: error: linker command failed with exit code 1 (use -v to see invocation) error when I do make

mohammedkw11, 2020-03-16:
Hi all,

I'm installing esp-IDF for first time following the this instruction:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

During prerequisites installing in this link https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup-scratch.html I came to install crosstool.

following this command:
`$ git clone https://github.com/espressif/crosstool-NG.git`
`$ cd crosstool-NG`
`$ git checkout esp-2019r2`
`$ git submodule update --init`
`$ ./bootstrap && ./configure --enable-local && make`

when I come to make, I receive this errors
`ld: symbol(s) not found for architecture x86_64`
`clang: error: linker command failed with exit code 1 (use -v to see invocation)`
`gmake[3]: *** [Makefile:515: conf] Error 1`
`gmake[3]: Leaving directory '/Volumes/ctng/crosstool-NG/kconfig'`
`gmake[2]: *** [Makefile:430: all] Error 2`
`gmake[2]: Leaving directory '/Volumes/ctng/crosstool-NG/kconfig'`
`gmake[1]: *** [Makefile:2383: all-recursive] Error 1`
`gmake[1]: Leaving directory '/Volumes/ctng/crosstool-NG'`
`make: *** [all] Error 2`

My machine: MacOS Catalina

Any help will be a prciated!

igrr, 2020-03-16:
@mohammedkw11 Do you actually need to compile the toolchain from scratch? If not, please follow the instructions for Step 1 at https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html, then proceed to Step 3 (link at "Next steps" here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html#next-steps)

mohammedkw11, 2020-03-17:
Hi @igrr 
Thanks for helping.

I went through the experiment and I want to implement this project https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server into my esp32 wroom.

I always gets this error when I run make menuconfigure
`make: *** No rule to make target `menuconfigure'.  Stop.`

Any idea?

igrr, 2020-03-17:
Can you please:
1) provide the information requested in the issue template, such as the version of IDF, your operating system, the build system you are using;
2) confirm whether you can successfully compile some simple example such as hello_world from IDF, as suggested in the getting started guide?

mohammedkw11, 2020-03-19:
Hi @igrr thanks for replaying!
I started from beginning, but this time I run the experiment on Ubuntu 18.04 and it goes smooth until I come to run the hello_world example in make flash I receive this error
`/home/mohammed/esp/esp-idf/components/wolfssl/src/tls.c:98:10: error: #error Please configure your TLS 1.3 DH key size using either: HAVE_FFDHE_2048, HAVE_FFDHE_3072, HAVE_FFDHE_4096, HAVE_FFDHE_6144 or HAVE_FFDHE_8192
         #error Please configure your TLS 1.3 DH key size using either: HAVE_FFDHE_2048, HAVE_FFDHE_3072, HAVE_FFDHE_4096, HAVE_FFDHE_6144 or HAVE_FFDHE_8192
          ^~~~~
`

On other example from here: wolfs_server experiment on esp32-wroom from this link https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server

I always receive the error when I run make menuconfigure
`Toolchain path: /home/mohammed/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc`
`Toolchain version: esp-2019r2`
`Compiler version: 8.2.0`
`Python requirements from /home/mohammed/esp/esp-idf/requirements.txt are satisfied.`
`make: *** No rule to make target 'menuconfigure'.  Stop.`

igrr, 2020-03-19:
Where did you find instructions to run `make menuconfigure`, can you please give a link?

For checking the hello-world app, you can remove `wolfssl` from the components directory.

mohammedkw11, 2020-03-19:
Thanks @igrr I could run the hello_world !

but for wolfssl_server I still have same error.

the instruction of running make menuconfigure is here https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server

mahavirj, 2020-03-20:
@mohammedkw11 Do you think examples hosted at https://github.com/espressif/esp-wolfssl can help to achieve your purpose?

mohammedkw11, 2020-03-20:
thanks @mahavirj for replying.

I haven't try yet, but if it's upgradable to TLS 1.3 then I could work on it. I'm looking to implement TLS 1.3 in my project, and I haven't found any library built this version.  please, if you have any topics related to server/client communication using TLS 1.3 refer it !

Thanks again!

AdityaHPatwardhan, 2020-03-20:
@mohammedkw11 , example [https_server](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server) works for wolfSSL, for that 
*(esp-idf must be installed)

mahavirj, 2020-03-20:
@mohammedkw11 No, TLS1.3 is not supported. Then I think you will to build version of `wolfssl` yourself. Since we do not control `https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/`, it is difficult to comment, could you file issue in that repository?

mohammedkw11, 2020-03-20:
hi @AdityaHPatwardhan , thanks for the help and I would go through this experience as a start point of understanding this protocol and I will modify the changes in order to achieve 1.3 version.

thanks in advance

mohammedkw11, 2020-03-20:
dear @mahavirj , I went to their website and I found they have like a ticket query that I fill errors I receive and they reply by email.  Thanks so much for the help!

Alvin1Zhang, 2020-05-26:
@mohammedkw11 Thanks for reporting and updates, glad that you got reply from that repository. Feel free to reopen if any more updates. Thanks.

closed 2020-05-26 by Alvin1Zhang