espressif/esp-idf #4940: Failed to join ESP32 AP if NimBLE is also enabled

AxelLin, 2020-03-16:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-701-g0ae960f2fe92
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

My device is running as AP mode and I have NimBLE enabled at the same time.
Then I try to join the ESP32 AP, but it fails as below log:

I (99622) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
I (99623) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
I (99803) app: station c8:ff:28:04:f8:09 join, AID=1
I (144591) wifi:station: c8:ff:28:04:f8:09 leave, AID = 1, bss_flags is 134243, bss:0x3ffbadc4
I (144592) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
I (144596) app: station c8:ff:28:04:f8:09 leave, AID=1

It's 100% reproducable in my test.

For comparison, I also test with Bluedroid enabled and it is fine. Only NimBLE has this issue.

Below is the successfully join log when build with bluedroid enabled:
I (33898) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
I (33899) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
I (33940) app: station c8:ff:28:04:f8:09 join, AID=1
I (33953) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.10.2
I (33953) app: STA IP assigned

AxelLin, 2020-03-16:
I try to enable wifi debug and get below log:

Build with NimBLE enabled: (Failure case, not able to join the AP)

I (1015) wifi:wifi driver task: 3ffd056c, prio:23, stack:3584, core=0
I (1039) wifi:wifi firmware version: d76a2dd
I (1040) wifi:config NVS flash: enabled
I (1040) wifi:config nano formating: disabled
I (1040) wifi:Init dynamic tx buffer num: 32
I (1044) wifi:Init data frame dynamic rx buffer num: 32
I (1049) wifi:Init management frame dynamic rx buffer num: 32
I (1055) wifi:Init management short buffer num: 32
I (1059) wifi:Init static rx buffer size: 1600
I (1063) wifi:Init static rx buffer num: 10
I (1067) wifi:Init dynamic rx buffer num: 32
I (1811) wifi:set country: cc=US schan=1 nchan=11 policy=0
E (2371) wifi:filter: set rx policy=0
I (2372) wifi:mode : softAP (30:ae:a4:09:75:2d)
E (2373) wifi:filter: set rx policy=9
I (2375) wifi:Total power save buffer number: 16
I (2376) wifi:Init max length of beacon: 752/752
I (2381) wifi:Init max length of beacon: 752/752
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=0 duration=forever
E (57069) wifi:ap recv assoc/reassoc request
I (57070) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
E (57070) wifi:bss=0x3ffbae3c, resp=16 join
I (57073) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
E (57078) wifi:join success, ap send assoc response
E (102819) wifi:ap recv deauth/disassoc from sta c8:ff:28:04:f8:09, reason:3
I (102820) wifi:station: c8:ff:28:04:f8:09 leave, AID = 1, bss_flags is 134243, bss:0x3ffbae3c
I (102824) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
E (122139) wifi:ap recv assoc/reassoc request
I (122140) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
E (122141) wifi:bss=0x3ffbae08, resp=16 join
I (122143) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
E (122149) wifi:join success, ap send assoc response

Build with Bluedroid enabled: (Working case)

I (1112) wifi:wifi driver task: 3ffd57bc, prio:23, stack:3584, core=0
I (1136) wifi:wifi firmware version: d76a2dd
I (1136) wifi:config NVS flash: enabled
I (1137) wifi:config nano formating: disabled
I (1137) wifi:Init dynamic tx buffer num: 32
I (1141) wifi:Init data frame dynamic rx buffer num: 32
I (1146) wifi:Init management frame dynamic rx buffer num: 32
I (1151) wifi:Init management short buffer num: 32
I (1156) wifi:Init static rx buffer size: 1600
I (1160) wifi:Init static rx buffer num: 10
I (1164) wifi:Init dynamic rx buffer num: 32
I (2096) wifi:set country: cc=US schan=1 nchan=11 policy=0
E (2689) wifi:filter: set rx policy=0
I (2691) wifi:mode : softAP (30:ae:a4:09:75:2d)
E (2691) wifi:filter: set rx policy=9
I (2694) wifi:Total power save buffer number: 16
I (2695) wifi:Init max length of beacon: 752/752
I (2699) wifi:Init max length of beacon: 752/752
E (26596) wifi:ap recv assoc/reassoc request
I (26597) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
E (26597) wifi:bss=0x3ffbaf20, resp=16 join
I (26600) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
E (26605) wifi:join success, ap send assoc response
I (29701) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.10.2

AxelLin, 2020-04-01:
Test again with v4.2-dev-994-gc9f29e0b5998, the issue is still there.

If below ble_gap_disc() is called, I can not join the ESP32 AP.
    rc = ble_gap_disc(own_addr_type, BLE_HS_FOREVER, &disc_params,
                      blecent_gap_event, NULL);

And take a look at the wifi log, it looks like join is success but the station cannot get IP.

TimXia, 2020-06-01:
@AxelLin  Sorry, ESP32 hasn't supported softAP and Bluetooth coexistence now.

AxelLin, 2020-06-01:
> softAP

@TimXia 
I'm surprised about that since there is no document mentions that it's not supported.
In additional, if build with bluedroid stack it actually **works** with softAP.
Only nimBLE does not work with softAP. (This looks like a bug rather than unsupported feature).

TimXia, 2020-06-01:
@AxelLin  There is only one antenna on ESP32 which means Wi-Fi and Bluetooth can not receive at the same time. Station which connects to ESP32(softAP) may send packets to ESP32 when antenna is taken by Bluetooth. As a result, ESP32 can not receive these packets. So softAP and Bluetooth coexistence is not well supported on ESP32.

AxelLin, 2020-06-01:
@TimXia 
(I'm aware ESP32 only has one antenna)
That does not explain why bluedroid works but NimBLE does not work.

I reported this issue based on the fact of observing device behavior.
But I don't quite understand your reply, do you mean espressif has no interest to address this issue?

TimXia, 2020-06-01:
@AxelLin What did you mean by 'work'? Did you test the throughput between ESP32 softAP and station? Or just mean station can connect to ESP32 softAP? As there is only one antenna, the throughput between ESP32 softAP and station is poor if Bluetooth is working, because there is no protocol that can guarantee the throughput.

AxelLin, 2020-06-01:
@TimXia 

When build with Bluedroid+softAP (ESP32 as softAP), I also run a tcp server on esp32.
My laptop can associate the ESP32 AP and run a tcp client program to connect the tcp server (to receive BLE data).
It actually works well.

I only use BLE function, so I think I can use NimBLE instead of Bluedroid.
But when testing NimBLE+sofAT, my laptop cannot get IP address at all. (see the log in my original post) This is nothing it do with throughput, it simply does not work with NimBLE+softAP.

TimXia, 2020-06-01:
@AxelLin OK. There may be something wrong with NimBLE and softAP coexistence. If throughput is not mandatory required, softAP can works with Bluetooth.

AxelLin, 2020-11-05:
I verify again with v4.3-dev-1720-g494a124d961c, still fail to join ESP32 AP if NimBLE is also enabled.

Below is today's test log:
There is a difference compare to my original post:
It shows "DHCP server assigned IP to a station, IP is: 192.168.10.2",
but on the STA (my linux laptop) the ifconfig does not show IP address.
i.e. The STA actually does not get the IP address.

I (73285) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<0,0>, prof:6
I (73287) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
W (73854) wifi:<ba-add>idx:4 (ifx:1, c8:ff:28:04:f8:09), tid:0, ssn:0, winSize:64
W (154326) wifi:<ba-del>idx
I (154327) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<0,0>, prof:6
I (154327) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
I (158333) wifi:station: c8:ff:28:04:f8:09 leave, AID = 1, bss_flags is 134242, bss:0x3fff050c
I (158334) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<0,0>, prof:6
I (173275) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<0,0>, prof:6
I (173277) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
W (173424) wifi:<ba-add>idx:4 (ifx:1, c8:ff:28:04:f8:09), tid:6, ssn:2, winSize:64
I (173426) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.10.2
I (218275) wifi:station: c8:ff:28:04:f8:09 leave, AID = 1, bss_flags is 134243, bss:0x3fff0be4
I (218276) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<0,0>, prof:6
W (218279) wifi:<ba-del>idx
I (235326) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<0,0>, prof:6
I (235327) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
I (235428) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.10.2
W (236514) wifi:<ba-add>idx:4 (ifx:1, c8:ff:28:04:f8:09), tid:0, ssn:0, winSize:64
W (237205) wifi:<ba-del>idx
W (237233) wifi:<ba-add>idx:4 (ifx:1, c8:ff:28:04:f8:09), tid:0, ssn:5, winSize:64

TimXia, 2020-12-15:
@AxelLin If do not call rc = ble_gap_disc(own_addr_type, BLE_HS_FOREVER, &disc_params, blecent_gap_event, NULL), will the issue exist? If BLE scan with window approximate to interval, softAP will have little chance to communicate.
With Bluedroid, did you do BLE scan too?

AxelLin, 2020-12-15:
No issue if not calling ble_gap_disc(), but that is what I need to call.
I test the same functionality with both bluedroid and nimble, so yes, I did test BLE scan with bluedroid.
The point is it works for bluedroid but does not work for nimble.
i.e. something wrong with nimble.

TimXia, 2020-12-15:
@AxelLin Are the BLE scan parameter of bluedroid same with that of nimble? I mean interval and window.

AxelLin, 2020-12-15:
The bluedroid test setting:
```
static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30,
    .scan_duplicate         = BLE_SCAN_DUPLICATE_DISABLE
};
```
The nimble test setting:
```
    struct ble_gap_disc_params disc_params;
    int rc;

    /* Figure out address to use while advertising (no privacy for now) */
    rc = ble_hs_id_infer_auto(0, &own_addr_type);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error determining address type; rc=%d\n", rc);
        return;
    }

    disc_params.filter_duplicates = 0;
    disc_params.passive = 0; 

    /* Use defaults for the rest of the parameters. */
    disc_params.itvl = 0;
    disc_params.window = 0;
    disc_params.filter_policy = 0;
    disc_params.limited = 0;

    rc = ble_gap_disc(own_addr_type, BLE_HS_FOREVER, &disc_params,
                      blecent_gap_event, NULL);

```

TimXia, 2020-12-15:
@AxelLin With bluedroid, the interval is 0x50 and the window is 0x30. There is 0x20 remained for softAP to communicate. But with nimble, the default interval is 0x30 and the window is 0x30 too. There is no time for softAP to use. So, please set interval as 0x50 and window as 0x30 with nimble and then test again.

AxelLin, 2020-12-15:
If set:
    disc_params.itvl = 0x50;
    disc_params.window = 0x30;
Then it works in AP mode.

2 questions:
1) Why it works in STA mode even when both itvl and window are 0? i.e. STA still has chance to communicate?
2) How to decide a good value for itvl/window setting? (especially the suitable setting is different for AP and STA cases)

TimXia, 2020-12-15:
@AxelLin Coexistence firmware does something to make it work in STA mode. In STA mode, it is better to set itvl equal to window. In AP mode, AP uses the rest time of the window of BLE scan. So if AP wants to use more time, reduce the window of BLE scan. Vice versa.

closed 2020-12-15 by AxelLin

ahzf, 2021-11-06:
Hi, I have the same problem, but when I'm in the role of BLE_ROLE_PERIPHERAL.

Are there equal parameters for:
```
disc_params.itvl   = 0x50;
disc_params.window = 0x30;
```
so, that the AP mode will work again?