espressif/esp-idf #4942: Station failed to join AP in WEP mode

AxelLin, 2020-03-16:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-701-g0ae960f2fe92
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

Join an AP in WEP mode failed.
The AP configuration is WEP Open 128Bit.

I (1509) app: Joining AP: phoenix
I (1593) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1595) wifi:mode : sta (30:ae:a4:09:75:2c)
E (2567) wifi:Invalid WEP key 1234567890abc

The same error log with shared auth:
I (1509) app: Joining AP: phoenix
I (1595) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1597) wifi:mode : sta (30:ae:a4:09:75:2c)
E (2670) wifi:Invalid WEP key 1234567890abc

i.e. It does not work in both WEP Open and WEP shared key.

BTW, I have no idea why it shows "1234567890abc" is an invalid WEP key.
(My understanding is the key is 13 ASCII or 26 HEX)

PS.
I have tested join the AP in WEP mode with my laptop and it works with my laptop.
So I think the problem is on ESP32 rather than the AP configuration.

AxelLin, 2020-03-25:
I tested again today and it works.
Not sure if something wrong in my previous test or something already fixed in recent idf update.

closed 2020-03-25 by AxelLin

sagb2015, 2020-03-25:
@AxelLin 2e7ae750119253b0ccf63ee3e302663465595f85 should have fixed it. Thanks!