espressif/esp-idf #4943: BLE mesh + NimBLE:  build error when calling esp_ble_mesh_deinit()

alyf80, 2020-03-16:
Module or chip used: ESP32-WROOM-32
IDF version: v4.0-223-gfdbdf9a0e
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux

Hello,
When using the BLE mesh functionality with the NimBLE host, any attempt at building code that calls `esp_ble_mesh_deinit()` results in the following error:

```
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(proxy_client.c.obj):(.literal.bt_mesh_proxy_prov_client_deinit+0x0): undefined reference to `bt_mesh_gattc_conn_cb_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(proxy_client.c.obj): in function `bt_mesh_proxy_prov_client_deinit':
/mnt/data/work/wioo/box/build/../esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_client.c:994: undefined reference to `bt_mesh_gattc_conn_cb_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(proxy_server.c.obj):(.literal.bt_mesh_proxy_deinit+0x0): undefined reference to `bt_mesh_gatts_service_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(proxy_server.c.obj):(.literal.bt_mesh_proxy_deinit+0x4): undefined reference to `bt_mesh_gatts_conn_cb_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(proxy_server.c.obj): in function `bt_mesh_proxy_deinit':
/mnt/data/work/wioo/box/build/../esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_server.c:1443: undefined reference to `bt_mesh_gatts_service_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/data/work/wioo/box/build/../esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_server.c:1455: undefined reference to `bt_mesh_gatts_conn_cb_deregister'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(main.c.obj):(.literal.bt_mesh_deinit+0x8): undefined reference to `bt_mesh_gatt_deinit'
/mnt/data/work/esp/tools-4/xtensa-esp32-elf-8.2.0-esp-2019r2/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(main.c.obj): in function `bt_mesh_deinit':
/mnt/data/work/wioo/box/build/../esp-idf/components/bt/esp_ble_mesh/mesh_core/main.c:417: undefined reference to `bt_mesh_gatt_deinit'
collect2: error: ld returned 1 exit status
```
It looks like those functions are defined for BlueDroid (in `components/bt/esp_ble_mesh/mesh_core/bluedroid_host/mesh_bearer_adapt.c`), but are currently missing from its NimBLE counterpart.

Alvin1Zhang, 2020-03-17:
@alyf80 Thanks for reporting.

Campou, 2020-03-17:
Hi @alyf80 

This issue has been fixed internally, and the bugfix will be synchronized to Github soon.

Thanks.

Alvin1Zhang, 2020-05-25:
@alyf80 Thanks for reporting and really appreciate your patience for waiting for the fix. The fix has been available on GitHub. Feel free to reopen if any updates. Thanks.

closed 2020-05-25 by Alvin1Zhang