espressif/esp-idf #4945: Add Stream Buffers and Message Buffers

dereks, 2020-03-16:
This is a follow-up to a request filed by user @hchaudhary1 a couple of years ago. Basically, he requested that FreeRTOS's newer StreamBuffer and MessageBuffer APIs be added in #1436, and provided a pull request #1447. He then rejected his own PR saying the ESP-created Ringbuffer was already available as a substitute, but then later (oddly) commented that the Ringbuffer wasn't working for him and he went with the StreamBuffer anyway.

I request that his PR should be re-opened and accepted. The Ringbuffer is poorly designed for an RTOS and may even cause errors.  It should be deprecated and removed from ESP-ISP API, or at least, re-implemented as a thin wrappers around StreamBuffers.

FreeRTOS's StreamBuffers and MessageBuffers use `xTaskNotifyWait()` to block until data is finally ready for it. This is a kernel scheduler function. **This means that a Task blocked waiting for data from a StreamBuffer uses zero CPU cycles.**

The Ringbuffer API, in contrast, grabs a semaphore, **enters a critical section, and then busy loops** until data is available or until timeout. This can be seen in the function `prvReceiveGeneric()`:

```
//No item available for retrieval, adjust ticks and take the semaphore again
if (xTicksToWait != portMAX_DELAY) {
    xTicksRemaining = xTicksEnd - xTaskGetTickCount();
}
```

This is terrible for performance in an RTOS. Not only is the CPU busy looping instead of sleeping (killing significant battery savings that could be had in the IDLE Task), it does so in a critical section, causing irregular delays for any hardware interrupts if a Task happens to be waiting on a Ringbuffer. (This is why the FreeRTOS StreamBuffer uses `vTaskSuspendAll()`/`xTaskResumeAll()` to suspend SysTick context switches without affecting other hardware interrupts.)

So, for improved performance, reduced battery usage, and a common API with official FreeRTOS (and thus AWS IoT Core), I request that StreamBuffers be pulled in to ESP-IDF.

All that said, there's currently an open bug with StreamBuffers, but a fix is available:

https://github.com/aws/amazon-freertos/issues/1837

hchaudhary1, 2020-03-16:
They re-fixed the ring buffer API at a later point... Again...

Bottom line, been using ring buffer API for a while.. working fine

dereks, 2020-03-17:
The problem I mention above (busy looping over a critical section) exists in the master branch as of this writing.

Yes, Ringbuffer will work, but it will consume extra battery, will cause hardware interrupt jitter, and is not compatible with upstream FreeRTOS.

Alvin1Zhang, 2020-03-17:
@dereks Thanks for reporting and @hchaudhary1 for updates. We will look into. Thanks.

dereks, 2020-03-17:
Note that the StreamBuffer has macros called `sbSEND_COMPLETED` and `sbRECEIVE_COMPLETED` that can be customized. By default, they just wake up a blocked reader or writer Task using `xTaskNotify()`. However, for an SMP dual-core system this macro will probably need to be customized similar to the other changes documented at

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html

An example of an SMP version of this macro is provided by FreeRTOS. From their docs:

> It is possible for the application writer to change this default behaviour by providing their own implementation of sbSEND_COMPLETED() in FreeRTOSConfig.h. That is useful when a stream buffer is used to pass data between cores on a multicore processor. In that scenario, sbSEND_COMPLETED() can be implemented to generate an interrupt in the other CPU core, and the interrupt’s service routine can then use the xStreamBufferSendCompletedFromISR() API function to check, and if necessary unblock, a task that was waiting for the data. The FreeRTOS/Demo/Common/Minimal/MessageBufferAMP.c source file provides a heavily commented example of exactly that scenario.

- https://www.freertos.org/RTOS-stream-buffer-example.html

Alvin1Zhang, 2020-03-18:
@dereks Thanks for reporting, we have taken note of this request, however sorry that we can't promise any date when this will be implemented.

igrr, 2020-04-23:
@dereks There is now a preview branch available with FreeRTOS 10 port, please see https://www.esp32.com/viewtopic.php?f=10&t=15322 for details.

saloid, 2020-07-16:
@igrr Nice, is it going to be in nearest release?

igrr, 2020-07-16:
The next upcoming releases, v4.1 and v4.2 will not include FreeRTOS 10, yet. We have just started v4.3-dev and we plan to merge FreeRTOS 10 into master some time soon, so it will end up in v4.3 eventually.

projectgus, 2020-10-15:
Thanks everyone for being patient while we worked on this.

The master branch (v4.3-dev) has now updated with FreeRTOS 10 support, which will be part of the v4.3 final release.

I'm going to close this issue now as these features are available in master. If you encounter any problems or bugs with the new features, please open new issues with the details.

closed 2020-10-15 by projectgus