espressif/esp-idf #4948: SYSTEM_EVENT_STA_DISCONNECTED

umer-ilyas, 2020-03-17:
## Environment

- Development Kit:     custom
- Kit version  
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.0-174-gf9cb434ee
- Build System:         CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?:     No
- Power Supply:        Battery

## Problem Description

ESP is not connected to WIFI AP and disconnect reason is WIFI_REASON_AUTH_EXPIRE, i am using different routers to connect but its same error , if use my make a hotspot on my phone and place it very close (3 feet) to esp then it will able to connect. 

### Expected Behavior
wifi should connect 
### Actual Behavior
wifi is not connecting and giving the reason WIFI_REASON_AUTH_EXPIRE

## Debug Logs

```
wifi mode 1
I (751051) wifi: clear blacklist
I (751061) wifi: Set ps type: 0

I (751061) wifi: Start wifi connect
I (751071) wifi: connect status 0 -> 0
I (751071) wifi: connect chan=0
I (751071) wifi: connect status 0 -> 1
I (751071) wifi: filter: set rx policy=3
I (751201) wifi: filter: set rx policy=4
I (751201) wifi: handoff_cb: status=0
I (751201) wifi: best bss has set.
I (751201) wifi: ap found, mac=34:1e:6b:34:27:a4
I (751201) wifi: bssid=34:1e:6b:34:27:a4, LR=0
I (751211) wifi: new_bss=0x3ffbc8c8, cur_bss=0x0, new_chan=<2,0>, cur_chan=2
I (751211) wifi: filter: set rx policy=5
I (751221) wifi: new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (751221) wifi: connect_op: status=0, auth=5, cipher=3
I (751231) wifi: auth mode is not none
I (751231) wifi: connect_bss: auth=1, reconnect=0
I (751241) wifi: state: init -> auth (b0)
I (751241) wifi: start 1s AUTH timer
I (752241) wifi: auth timeout
I (752241) wifi: state: auth -> init (200)
I (752241) wifi: connect status 1 -> 4
I (752241) wifi: stop beacon/connect timer
I (752251) wifi: reason: auth expire(2)
I (752251) wifi: add bssid 34:1e:6b:34:27:a4 to blacklist, cnt=0
I (752261) wifi: stop CSA timer
I (752261) wifi: remove 34:1e:6b:34:27:a4 from rc list
I (752261) wifi: new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (752271) wifi: filter: set rx policy=8
I (752271) wifi: Send disconnect event, reason=2, AP number=0
I (752281) WIFI: SYSTEM_EVENT_STA_DISCONNECTED
I (752291) WIFI: disconnect reason: 2
```

## Other items if possible

umer-ilyas, 2020-03-18:
it seems like a wifi range issue, if I move the device near to AP it connects.

but most probably induced by latest software updates, because my setup is pretty much same.

some times wifi do connect but GOT_IP event never triggered 

```
I (1006296) wifi: Set ps type: 0

I (1006296) wifi: Start wifi connect
I (1006306) wifi: connect status 4 -> 0
I (1006306) wifi: connect chan=0
I (1006306) wifi: connect status 0 -> 1
I (1006316) wifi: filter: set rx policy=3
I (1006436) wifi: filter: set rx policy=4
I (1006436) wifi: handoff_cb: status=0
I (1006436) wifi: best bss has set.
I (1006436) wifi: ap found, mac=34:1e:6b:34:27:a4
I (1006436) wifi: bssid=34:1e:6b:34:27:a4, LR=0
I (1006446) wifi: new_bss=0x3ffbc8c0, cur_bss=0x0, new_chan=<2,0>, cur_chan=2
I (1006446) wifi: filter: set rx policy=5
I (1006456) wifi: new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (1006456) wifi: connect_op: status=0, auth=5, cipher=3 
I (1006466) wifi: auth mode is not none
I (1006466) wifi: connect_bss: auth=1, reconnect=0
I (1006476) wifi: state: init -> auth (b0)
I (1006476) wifi: start 1s AUTH timer
I (1006496) wifi: recv auth: seq=2, status=0
I (1006496) wifi: state: auth -> assoc (0)
I (1006496) wifi: restart connect 1s timer for assoc
I (1006516) wifi: recv assoc: type=0x10
I (1006516) wifi: filter: set rx policy=6
I (1006516) wifi: state: assoc -> run (10)
I (1006516) wifi: start 10s connect timer for 4 way handshake
I (1006526) wifi: sta recv dup seq=16 tid=17, discard
I (1006526) wifi: sta recv dup seq=16 tid=17, discard
I (1006536) wifi: sta recv dup seq=32 tid=17, discard
I (1006536) wifi: sta recv dup seq=32 tid=17, discard
I (1006776) wifi: wpa_psk start
I (1006786) wifi: wpa_psk handle succeed
I (1006786) wifi: sta recv dup seq=0 tid=7, discard
I (1006786) wifi: sta recv dup seq=0 tid=7, discard
I (1006786) wifi: sta recv dup seq=0 tid=7, discard
I (1006786) wifi: sta recv dup seq=0 tid=7, discard
I (1006796) wifi: sta recv dup seq=0 tid=7, discard
I (1006816) wifi: wpa_psk start
I (1006826) wifi: wpa_psk handle succeed
I (1006826) wifi: sta recv dup seq=16 tid=7, discard
I (1006826) wifi: sta recv dup seq=16 tid=7, discard
I (1006826) wifi: connected with Mission Control NTL, aid = 14, channel 2, BW20, bssid = 34:1e:6b:34:27:a4
I (1006836) wifi: security type: 4, phy: bgn, rssi: -61
I (1006836) wifi: remove all except 34:1e:6b:34:27:a4 from rc list
I (1006846) wifi: clear blacklist
I (1006846) wifi: filter: set rx policy=7
I (1006856) wifi: pm start, type: 0

I (1006856) wifi: Send sta connected event
I (1006856) wifi: connect status 1 -> 5
I (1006866) wifi: AP's beacon interval = 102400 us, DTIM period = 1
wifi connected 
I (1007026) wifi: sta recv dup seq=0 tid=0, discard
I (1007036) wifi: sta recv dup seq=0 tid=0, discard
I (1007036) wifi: sta recv dup seq=0 tid=0, discard
I (1007036) wifi: sta recv dup seq=0 tid=0, discard
I (1007036) wifi: sta recv dup seq=0 tid=0, discard
I (1007036) wifi: sta recv dup seq=0 tid=0, discard
I (1007046) wifi: sta recv dup seq=0 tid=0, discard
I (1008426) wifi: sta recv dup seq=32 tid=0, discard
I (1008436) wifi: sta recv dup seq=32 tid=0, discard
I (1008436) wifi: sta recv dup seq=32 tid=0, discard
[... 219 lines cut ...]
I (1055426) wifi: sta recv dup seq=2992 tid=0, discard
I (1055446) wifi: sta recv dup seq=3024 tid=0, discard
I (1055446) wifi: sta recv dup seq=3024 tid=0, discard
I (1055456) wifi: sta recv dup seq=3024 tid=0, discard
I (1055466) wifi: sta recv dup seq=3040 tid=0, discard
I (1055466) wifi: sta recv dup seq=3040 tid=0, discard
I (1056336) wifi: sta recv dup seq=3056 tid=0, discard
I (1056346) wifi: sta recv dup seq=3072 tid=0, discard
I (1056346) wifi: sta recv dup seq=3072 tid=0, discard
I (1056346) wifi: sta recv dup seq=3072 tid=0, discard
I (1056346) wifi: sta recv dup seq=3072 tid=0, discard
I (1056356) wifi: sta recv dup seq=3072 tid=0, discard
I (1056356) wifi: sta recv dup seq=3072 tid=0, discard
I (1056366) wifi: sta recv dup seq=3088 tid=0, discard
I (1056376) wifi: sta recv dup seq=3104 tid=0, discard
I (1056396) wifi: sta recv dup seq=3104 tid=0, discard
I (1056406) wifi: sta recv dup seq=3104 tid=0, discard
I (1059306) wifi: sta recv dup seq=3120 tid=0, discard
I (1059306) wifi: sta recv dup seq=3120 tid=0, discard
I (1059316) wifi: sta recv dup seq=3120 tid=0, discard
I (1059316) wifi: sta recv dup seq=3120 tid=0, discard
I (1059316) wifi: sta recv dup seq=3120 tid=0, discard
I (1059326) wifi: sta recv dup seq=3120 tid=0, discard

```

liuzfesp, 2020-03-22:
HI @umer-ilyas, 

`but most probably induced by latest software updates` that seems strange because our QA test didn't notify any difference between v4.0 and previous releases. Could you let me know the IDF commit ID that WiFi connection works well? 

Moreover, how about changing a different ESP32 device and do a similar test?

liuzfesp, 2020-05-08:
Hi @umer-ilyas any update about this issue?

umer-ilyas, 2020-05-08:
it semms it was due to to netif updates 
i had to call  ` esp_netif_create_default_wifi_sta();` and now it is not reproducing any more

closed 2020-05-08 by umer-ilyas