espressif/esp-idf #4949: Secure Boot, Secure Flash, Sign image, FAILED

filzek, 2020-03-17:
Hi, as os latest master, we cant build the image anymore if we enable the follow options in menuconfig.

[image]

v4.2-dev-710-g85e66d811
espsecure.py v3.0-dev
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump.exe: signature_verification_key.bin.S.obj: file format not recognized

Something is not accurate and running correct anymore.

Please advise two things:

1) How to solve the problem
2) How to move from a just DEV Flash Encrypt mode to a FULL Security mode in idf.py

thanks and awaiting solution.

supreetd21, 2020-03-18:
Assuming that the submodules are in sync on the latest master (git submodule update --init --recursive ) and you have generated the apt ECDSA signing key (python espsecure.py generate_signing_key -v 1 secure_boot_sign.pem), I could not reproduce your issue. Can you please provide some more information (Sdkconfig file, Build logs).

Which build system are you using (Make/CMake)?

Also, could you please elaborate on question 2?
The way to enable flash encryption release mode(full security).
You can set the release mode through menuconfig. I can see you have enabled it in the above screenshot.
(https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html#release-mode)

If you have already flashed a firmware with development mode and selected some insecure options, you can manually program(espefuse.py) the corresponding eFuses to ensure that you have the requisite security.

projectgus, 2020-03-19:
Hi @filzek ,

Same as @supreetd21, I also couldn't reproduce this with a new install on Windows.

Is it possible that the `signature_verification_key.bin.S.obj` file was corrupted somehow after it was generated in the build directory? Can try `idf.py fullclean` and then rebuild to rule this out.

One other thing I noticed:

> v4.2-dev-710-g85e66d811

`85e66d811` is not an ESP-IDF commit from our repo, in fact the latest version in our repo right now is `v4.2-dev-701-g0ae960f2fe` - 9 commits behind. Is it possible some change has been added to your copy of IDF that broke secure boot in the build system?

> The way to enable flash encryption release mode(full security).

In addition to Supreet's answer which I agree with, we don't provide a way to upgrade one device from Development to Release mode. It is possible to do this by manually burning the `download_dis_encrypt`/`download_dis_decrypt` efuses mentioned in the linked docs,  but we would generally recommend keeping one device as the "Development device" only and using a whole new device as your "Release mode" device, to avoid any possibility of accidentally shipping a device which is not configured securely.

filzek, 2020-03-23:
Ok, i have removed all new and old files, have downloaded new installer and new esp-idf, not things got back to track, but I need to understant something about enable secure boot.

Our testing device does not have secure boot, but does have flash encryption enabled.

How to I enable now Secure Boot on it, we tryed on one board and it has die, so, we are afraid to try it again and kill another esp32, so, please advise before I continue.

projectgus, 2020-03-23:
Hi @filzek ,

Glad that the Secure Boot build is working now.

> Our testing device does not have secure boot, but does have flash encryption enabled.
>
> How to I enable now Secure Boot on it, we tryed on one board and it has die, so, we are afraid to try it again and kill another esp32, so, please advise before I continue.

Can you please give more details, including:

* Are the test devices using Flash Encryption configured in Development mode or Release mode?
* How exactly did you flash the board to enable Secure Boot?
* What does it do exactly when it "dies"?
* What is the output of `espefuse.py -p PORT summary` on the dead board?

filzek, 2020-03-24:
Same problem

Cant make it work, more than 20 esp32 throw in trash!

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

Did exact what all papers ask to do, have burned the flash, and nothing works.

Why this is so terrible difficult to make it to work? losing hope on this....

filzek, 2020-03-24:
@projectgus summary

espefuse.py v3.0-dev
Connecting....
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1079 R/W (0x13)

Identity fuses:
MAC                    Factory MAC Address
  = c8:2b:96:ac:35:b8 (CRC 0x31 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
UART_DOWNLOAD_DIS      Disable UART download mode (ESP32 rev3 only)      = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 1 R/W (0x1)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 1 R/W (0x1)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 1 R/W (0x1)
BLK1                   Flash encryption key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK2                   Secure boot key
  = 7e 9a 10 8b 33 d9 ff df 7a 09 e4 30 36 6a 8e 39 66 0a e6 09 55 9f d6 07 34 dc 0a 87 9a be f8 22 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 128 R/W (0x80)
RD_DIS                 Efuse read disablemask                            = 1 R/W (0x1)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 52 R/W (0x34)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

projectgus, 2020-03-24:
Hi @filzek,

The efuse summary you posted here are unusual, but I think I can see what has happened. Most likely these ESP32s can still be recovered, and do not need to be thrown away:

* This ESP32 does not have flash encryption enabled (`FLASH_CRYPT_CNT` == 0)
* However the other efuses required for flash encryption look to have already been burned.
* This ESP32 has not had secure boot enabled (`ABS_DONE_0` == 0)
* A flash encryption key has been generated and burned to efuse BLK1, and read/write protected
* A secure boot key has been generated and burned to efuse BLK2, but has not been read or write protected (meaning it is not protected against readout by an attacker).

My guess is that the bootloader has self-encrypted on the first boot, but some problem was detected so the process was aborted before flash encryption was enabled. This leaves the flash contents unreadable as it's already been encrypted, but you can reflash to start the encryption process again.

* How are you flashing each ESP32 now? It should be possible to flash them in a way where you can watch the log output from the first boot, and see what the problem is. (If using esptool.py to flash the bootloader, then make sure to pass `--after no_reset` when flashing and then run a serial monitor program afterward to capture the first boot's log output.)

The only thing I can't explain is the unprotected Secure Boot key in BLK2. Did you burn this manually with espefuse.py?

filzek, 2020-03-24:
used the latest idf, clean install of it using esp-idf-tools-setup-2.3.exe, new  machine.

Have enable the secure option in the idf.py menuconfig for secure boot and flash encryption in DEV mode.

proceed with the idf.py build

it has ask to manually flash the bootloader and so I did.

later I have runned the latest line telling to do the flash of the rest.

just it and result was that.

projectgus, 2020-03-25:
Hi @filzek ,

Thanks for the extra details

To confirm: You never ran any step to manually flash the Secure Boot key to the chip in this case?

Can you please try the suggestion from above to try and capture the log output from first boot? You can re-flash the ESP32 that produced the "summary" output above and then boot it, don't need to try a new one:

> It should be possible to flash them in a way where you can watch the log output from the first boot, and see what the problem is. (If using esptool.py to flash the bootloader, then make sure to pass [...]

filzek, 2020-03-25:
No, I did it automatic using the idf.py command propose by the idf.py build

I am not sure what is the command I need to run, can you give the full command to run?

projectgus, 2020-03-25:
Hi @filzek,

`idf.py build` should prints a command line to run `esptool.py` with all the correct arguments. (If Secure Boot V1 is on then it prints two command lines at different points, one to flash the app and one to flash the bootloader - because the bootloader can be flashed once only.)

As a shortcut, try running the following and then capture the log output:

```
idf.py flash bootloader-flash monitor
```

(Do this to one of the ESP32s which is already not working.)

Angus

filzek, 2020-03-25:
==============================================================================
Bootloader built. Secure boot enabled, so bootloader not flashed automatically.
One-time flash command is:
        C:\Users\filze\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe D:/Dropbox/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 D:/Dropbox/esp-idf/Dev/PRODUCAO/build/bootloader/bootloader.bin

C:\Users\filze\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x9000 build\partition_table\partition-table.bin 0x10000 build\ota_data_initial.bin 0x20000 build\Switch.bin

there is the two provided command after idf.py build

now running

idf.py flash bootloader-flash monitor

`D:\Dropbox\esp-idf\Dev\PRODUCAO>idf.py flash bootloader-flash monitor
Executing action: flash
Choosing default port b'COM5' (use '-p PORT' option to set a specific serial port)
Running ninja in directory d:\dropbox\esp-idf\dev\producao\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D D:\Dropbox\esp-idf\Dev\PRODUCAO\bu...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0xa000,24K,
otadata,data,ota,0x10000,8K,
phy_init,data,phy,0x12000,4K,
nvs_keys,data,nvs_keys,0x13000,0x6300,
factory_nvs,data,nvs,0x19300,0x6300,
ota_0,app,ota_0,0x20000,1664K,encrypted
ota_1,app,ota_1,0x1c0000,1664K,encrypted
storage,data,fat,0x360000,576K,encrypted
*******************************************************************************
[2/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] cmd.exe /C "cd /D D:\Dropbox\esp-idf\components\espt...:/Dropbox/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM5 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x9000 partition_table/partition-table.bin 0x10000 ota_data_initial.bin 0x20000 Switch.bin
esptool.py v3.0-dev
Serial port COM5
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:ac:35:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3140 bytes to 247...
Writing at 0x00009000... (100 %)
Wrote 3140 bytes (247 compressed) at 0x00009000 in 0.0 seconds (effective 1877.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x00010000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x00010000 in 0.0 seconds (effective 7973.3 kbit/s)...
Hash of data verified.
Compressed 1507316 bytes to 884415...
Writing at 0x00020000... (1 %)
Writing at 0x00024000... (3 %)
Writing at 0x00028000... (5 %)
Writing at 0x0002c000... (7 %)
Writing at 0x00030000... (9 %)
Writing at 0x00034000... (11 %)
Writing at 0x00038000... (12 %)
Writing at 0x0003c000... (14 %)
Writing at 0x00040000... (16 %)
Writing at 0x00044000... (18 %)
Writing at 0x00048000... (20 %)
Writing at 0x0004c000... (22 %)
[... 72 lines cut ...]
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:ac:35:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 24688 bytes to 16553...
Writing at 0x00001000... (50 %)
Writing at 0x00005000... (100 %)
Wrote 24688 bytes (16553 compressed) at 0x00001000 in 0.4 seconds (effective 488.8 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
Executing action: monitor
Running idf_monitor in directory d:\dropbox\esp-idf\dev\producao
Executing "C:\Users\filze\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe D:\Dropbox\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf- d:\dropbox\esp-idf\dev\producao\build\Switch.elf -m 'C:\Users\filze\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe' 'D:\Dropbox\esp-idf\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
--- idf_monitor on \\.\COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
`

filzek, 2020-03-25:
It is only blinking in the boot

[image]

filzek, 2020-03-25:
I did reset the board, and it does:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

only

filzek, 2020-03-25:
python ..\..\components\esptool_py\esptool\espefuse.py -p com5 summary
espefuse.py v3.0-dev
Connecting........_
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 128 R/W (0x80)
RD_DIS                 Efuse read disablemask                            = 1 R/W (0x1)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 52 R/W (0x34)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 1 R/W (0x1)
UART_DOWNLOAD_DIS      Disable UART download mode (ESP32 rev3 only)      = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 1 R/W (0x1)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 1 R/W (0x1)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 1 R/W (0x1)
BLK1                   Flash encryption key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK2                   Secure boot key
  = 7e 9a 10 8b 33 d9 ff df 7a 09 e4 30 36 6a 8e 39 66 0a e6 09 55 9f d6 07 34 dc 0a 87 9a be f8 22 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1079 R/W (0x13)

Identity fuses:
MAC                    Factory MAC Address
  = c8:2b:96:ac:35:b8 (CRC 0x31 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V)

projectgus, 2020-03-30:
Hi @filzek ,

Sorry for the delay in getting back to you. I'm not sure what is going with your boards, the new efuse summary shows that (compared to the previous efuse summary) the FLASH_CRYPT_CNT efuse has now been burned to value 1, which means that flash encryption is enabled. 

This has happened since the [first efuse capture](https://github.com/espressif/esp-idf/issues/4949#issuecomment-602992809), that shows the same board does not have FLASH_CRYPT_CNT burned at that time.

Unfortunately it seems like whenever this happened, the log output wasn't captured so it's not possible to see what went wrong to cause it to fail again.

Is it possible that the board is being reset part way through the first boot? The ESP32 needs to be left alone during the first boot (ie after the first reset after flashing) to complete the initial self-encryption and setup of the security features. This can take a couple of minutes in some cases.

This ESP32 probably still isn't bricked, but you'll need to burn the `FLASH_CRYPT_CNT` efuse manually `espefuse.py -p PORT burn_efuse FLASH_CRYPT_CNT` to disable flash encryption again (setting it to value 0x3). *You can do only do this a limited number of times so suggest not doing it until we figure out what keeps going wrong here and why you're not seeing any log output from the first boot*.

Would you be able to send me your `sdkconfig` file, `build/bootloader/bootloader.bin` and `build/bootloader/bootloader.elf` file, please? Can send privately to angus at espressif dot com.

projectgus, 2020-07-03:
Hi @filzek ,

I believe my colleagues followed up with you about this internally. Are you still experiencing any issue?

Angus

closed 2020-08-06 by projectgus