espressif/esp-idf #4951: Pico D4 core dump not working after changing in the sdkconfig file on PlatformIO

vijayprakash1, 2020-03-18:
Hello Everyone,

I deployed a few(100) devices on the field using PICO d4, but in some, I am facing issues related to module halting at an unknown location. Even CONFIG_ESP32_PANIC_PRINT_REBOOT=y is set.
After pressing the reset devices starts working fine.

To know the exact issue where my device is halting I am trying to implement core dump but it seems not working.

I wrote a small code to just implement the core dump 

PS: I am using PLATFORMIO latest version

#include <Arduino.h>

void setup() {

  Serial.begin(115200);
  Serial.println("Hello");
  delay(1000);

}

void loop() {

  Serial.println(100/0);

}

## Debug Logs

ets Jun  8 2016 00:22:57
[SDK_Config.txt](https://github.com/espressif/esp-idf/files/4347259/SDK_Config.txt)

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Hello
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0c03  PS      : 0x00060f30  A0      : 0x800d1d74  A1      : 0x3ffb1f90
A2      : 0x3ffc00ec  A3      : 0x3ffb0060  A4      : 0x00000020  A5      : 0x80000020
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb1f60
A10     : 0x3ffc00ec  A11     : 0x00000064  A12     : 0x0000000a  A13     : 0x00000003
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace: 0x400d0c03:0x3ffb1f90 0x400d1d71:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Hello
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0c03  PS      : 0x00060f30  A0      : 0x800d1d74  A1      : 0x3ffb1f90
A2      : 0x3ffc00ec  A3      : 0x3ffb0060  A4      : 0x00000020  A5      : 0x80000020
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb1f60
A10     : 0x3ffc00ec  A11     : 0x00000064  A12     : 0x0000000a  A13     : 0x00000003
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace: 0x400d0c03:0x3ffb1f90 0x400d1d71:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Rebooting...

[SDK_Config.txt](https://github.com/espressif/esp-idf/files/4347262/SDK_Config.txt)

Also, I am attaching my sdkconfig file setting.

igrr, 2020-03-18:
@vijayprakash1 I don't think you can enable core dump when using Arduino, as all the IDF components are pre-compiled there. You can however use Arduino as IDF component (without platformio), in which case you should be able to change sdkconfig. See https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md.

Alvin1Zhang, 2020-04-07:
@vijayprakash1 Thanks for reporting, any feedback on this issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-05-21:
Will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-21 by Alvin1Zhang