espressif/esp-idf #4955: Failing to install esp-idf-tools-setup-2.3.exe in Windows 10

Eric-AMIG, 2020-03-18:
Install directories:

ESP-IDF:
C:\Users\Edo\esp\esp-idf

ESP-IDF Tools:
C:\Users\Edo\.espressif

Using Python 3.7:
      C:\Users\Edo\AppData\Local\Programs\Python\Python37\python.exe

Using Git 2.22.0.windows.1:
      C:\Program Files\Git\cmd\git.exe

Will download ESP-IDF v4.0 into:
      C:\Users\Edo\esp\esp-idf

IDF tools directory (IDF_TOOLS_PATH):
      C:\Users\Edo\.espressif

- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version: Not sure
- Module or chip used:  [ESP32-WROOM-32]
- Operating System:     [Windows 10]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: Trying to configure VS Code but cannot install ESP-IDF, so I haven't gone deeper into configuring VS Code's Terminal and environment.
- Power Supply:         [USB]

## Problem Description

I have been trying to install ESP-IDF along with the ESP-IDF Tools by running the installer esp-idf-tools-setup-2.1.exe (and both newer and older versions of the installer including 2.3 and 1.2). I select the existing version of Python 3.7 for the installation, along with the existing version of Git 2.22. After I select the option of downloading ESP-IDF (having tried with both 4.0 and 3.3.1 versions). 

After a while installing I get the following error: Installation has failed with exit code 1.

I have also gotten an exit code 2 in different attempts.

I have no spaces in the paths for ESP-IDF or the tools directories. Is this a bug in the installer?

Will really appreciate help on solving this issue.

## Debug Logs

Here I attach an extract from the error log:

```
2020-03-18 12:25:10.778   --------
2020-03-18 12:25:10.778   Running command: C:\Program Files\Git\cmd\git.exe -C C:\Users\Edo\esp\esp-idf reset --hard
                          Checking out files: 100% (4424/4424), done
                          HEAD is now at 143d26aa4 Merge branch 'bugfix/doc_package_versions_v3.3' into 'release/v3.3
                          
2020-03-18 12:25:10.778   --------
2020-03-18 12:25:10.782   Resetting the submodules: C:\Program Files\Git\cmd\git.exe -C C:\Users\Edo\esp\esp-idf reset --hard
2020-03-18 12:25:15.640   Done, exit code=0
2020-03-18 12:25:15.640   --------
2020-03-18 12:25:15.640   Running command: C:\Program Files\Git\cmd\git.exe -C C:\Users\Edo\esp\esp-idf submodule foreach git reset --hard
                          Entering 'components/asio/asio
                          HEAD is now at 55efc179 Merge branch 'idf' into 'master
                          Entering 'components/aws_iot/aws-iot-device-sdk-embedded-C
                          HEAD is now at 2991832 Merge tag 'v3.0.1
                          Entering 'components/bt/lib
                          HEAD is now at bd605c2 Fix assert in ld_fm.c due to no free element from(5aed448
                          Entering 'components/coap/libcoap
                          HEAD is now at 6468887 .gitignore: do not ignore custom files in m
                          Entering 'components/esp32/lib
                          HEAD is now at f50b25d Fix WiFi scan and connect bugs when coexist with BT/BLE(ac331d7
                          Entering 'components/esptool_py/esptool
                          HEAD is now at de30f21 Version 2.
                          Entering 'components/expat/expat
                          HEAD is now at a7bc26b Set release date for 2.2.
                          Entering 'components/json/cJSON
                          HEAD is now at 3c89356 Merge pull request #355 from DaveGamble/release-1.7.1
                          Entering 'components/libsodium/libsodium
                          HEAD is now at 70170c28 Inden
                          Entering 'components/lwip/lwip
                          HEAD is now at 9b9d126b dhcp_timeout: change the timeout type u16 to u3
                          Entering 'components/mbedtls/mbedtls
                          HEAD is now at f5f2e5926 Merge branch 'fix/warnings_in_NimBLE_build_failure' into 'mbedtls-2.16.2-idf
                          Entering 'components/micro-ecc/micro-ecc
                          HEAD is now at d037ec8 Add note that uECC_VLI_NATIVE_LITTLE_ENDIAN affects key compatiblit
                          Entering 'components/mqtt/esp-mqtt
                          HEAD is now at e3b013e Extended error structure to be used in mqtt library and include mqtt specific errors. Corrected typos in comments and log message
                          Entering 'components/nghttp/nghttp2
                          HEAD is now at 3bcc416e Update manual page
                          Entering 'components/nimble/nimble
                          HEAD is now at 8326807c Save the correct key on peer_sec
                          Entering 'components/protobuf-c/protobuf-c
                          HEAD is now at dac1a65 Merge pull request #283 from protobuf-c/edmonds/changelog/1.3.
                          Entering 'components/spiffs/spiffs
                          HEAD is now at f5e26c4 fixes #16
                          Entering 'components/unity/unity
                          HEAD is now at 7d2bf62 Merge pull request #363 from Deltrix/patch-
                          Entering 'examples/build_system/cmake/import_lib/main/lib/tinyxml2
                          HEAD is now at 7e8e249 Merge pull request #707 from SwiftEngineering/issue_70
                          
2020-03-18 12:25:15.640   --------
2020-03-18 12:25:15.644   idf_tools.py exists in IDF directory
2020-03-18 12:25:15.644   idf_tools.py command: C:\Users\Edo\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Edo\esp\esp-idf\tools\idf_tools.py
2020-03-18 12:25:15.644   Installing tools:C:\Users\Edo\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Edo\esp\esp-idf\tools\idf_tools.py install
2020-03-18 12:25:15.959   Done, exit code=1
2020-03-18 12:25:15.959   --------
2020-03-18 12:25:15.959   Running command: C:\Users\Edo\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Edo\esp\esp-idf\tools\idf_tools.py install
                          Traceback (most recent call last):
                            File "C:\Users\Edo\esp\esp-idf\tools\idf_tools.py", line 1349, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\Edo\esp\esp-idf\tools\idf_tools.py", line 1345, in main
                              action_func(args)
                            File "C:\Users\Edo\esp\esp-idf\tools\idf_tools.py", line 1097, in action_install
                              tools_info = load_tools_info()
                            File "C:\Users\Edo\esp\esp-idf\tools\idf_tools.py", line 796, in load_tools_info
                              with open(tool_versions_file_name, 'r') as f:
                          FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Program Files\\Espressif\\ESP-IDF Tools\\tools/tools.json'
                          
2020-03-18 12:25:15.959   --------
2020-03-18 12:25:15.959   Message box (OK):
                          Installation has failed with exit code 1
2020-03-18 12:27:20.424   User chose OK.
2020-03-18 12:27:20.431   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-03-18 12:27:22.593   User chose Yes.
2020-03-18 12:27:22.609   CurStepChanged raised an exception.
```

igrr, 2020-03-19:
Could you please check your environment variables and see if there is an entry for IDF_PATH set to `C:\\Program Files\\Espressif\\ESP-IDF Tools`? If there is one, please remove it and try again.

Alvin1Zhang, 2020-08-18:
@Eric-AMIG Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Jigar3690, 2020-10-29:
[image]

[Setup Log 2020-10-28 #002.txt](https://github.com/espressif/esp-idf/files/5455642/Setup.Log.2020-10-28.002.txt)

I had everything working, since last many days.
I got it messed up due to my last merge to my develop branch.

So I uninstalled IDF tools and tried to re-install. but had this error.

[image]

In Path definitions I have:
[image]

Let me know if any more details required.

igrr, 2020-10-29:
@Jigar3690 Please open a separate issue, this isn't the same problem as of @Eric-AMIG. Please also mention the version of ESP-IDF located in C:\users\jigar\Desktop\esp-mdf\esp-idf.

grlind61, 2020-12-16:
i have tried multiple configurations of python 3.7 and newer 3.9 i still cannot install the ide.. any suggestions?  i have used the master release etc, 4,0, 4.1 etc no difference

Installation has failed with exit code 1

[Setup Log 2020-12-15 #010.txt](https://github.com/espressif/esp-idf/files/5699583/Setup.Log.2020-12-15.010.txt)

bharath5673, 2021-01-08:
Facing same error ... plz help :(((

Installation has failed with exit code 1

grlind61, 2021-01-08:
What I did to solve the issue was to 
1. remove all python installs - installed python 3.6
2. the enviromental string was way to long to so I install python at the C:\python\Python36, 
3. also make sure the python path has legal --no crazy issues mine installed with a "space" and the is not legal in windows paths --- set the enviroment paths at the system startup- I set them manually editing them for windows 10
4. rebooted machine tested python install and enviromental variables.
5. install the esp-idf tools for visual code.

georgik, 2021-04-13:
@Eric-AMIG @grlind61 @bharath5673 Thank you for reporting the problems and we're sorry for the late response.

Meanwhile, we updated the Windows Tools installer so that it contains Embedded Python which solves many problems with the collision of system Python or Anaconda distribution. We also built missing Python Wheels for Windows and the installer can load them from https://dl.espressif.com/pypi/.

We also added several checks and corrections to the installer based on GitHub issues.

Please, download latest version of the installer from https://dl.espressif.com/dl/esp-idf/

Should you find a problem, please let us know.

closed 2021-04-16 by Alvin1Zhang